theyosh/TerrariumPI #847: [BUG]: Error after trying to add relais to enclosure

1ucket, 2023-10-16:
**Setup:**
 - TerrariumPI version: 4.7.0
 - Install: Manual
 - Raspberry PI: 4

**Describe the bug**
I pulled after #846 was fixed. I added the relais (which worked). Now I wanted to add it to the corresponding area, which resulted in an error:
`Error updating area Area lights Basking Spot links in timer mode, part of Enclosure Terrarium with 3 areas. fromisoformat: argument must be str`

Now it won't start up anymore, with the following error:
`2023-10-16 20:17:17,052 - INFO    - terrariumEngine       - Loading existing enclosures from database.
Traceback (most recent call last):
  File "terrariumPI.py", line 19, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 158, in __init__
    self.__load_existing_enclosures()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 1197, in __load_existing_enclosures
    list(enclosure.areas),
  File "/home/pi/TerrariumPI/terrariumEnclosure.py", line 24, in __init__
    self.load_areas(areas)
  File "/home/pi/TerrariumPI/terrariumEnclosure.py", line 65, in load_areas
    self.add(terrariumArea(area.id, self, area.type, area.name, area.mode, area_setup))
  File "/home/pi/TerrariumPI/terrariumArea.py", line 86, in __init__
    self.load_setup(setup)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 1002, in load_setup
    super().load_setup(data)
  File "/home/pi/TerrariumPI/terrariumArea.py", line 342, in load_setup
    self._time_table()
  File "/home/pi/TerrariumPI/terrariumArea.py", line 257, in _time_table
    begin = datetime.time.fromisoformat(self.setup[period]["begin"])
TypeError: fromisoformat: argument must be str`

**To Reproduce / Installation steps**
Edit enclosure, possibly has to do with the relay.

**Expected behavior**
No error, TerrariumPI starts properly.

theyosh, 2023-10-16:
Updated the code. Do a `git pull` and restart in debug mode:  https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

And see if it continues. If so, could then try to post the area setup? Not sure how many areas you have, but there is an area with a timer mode that has not the correct begin and/or end time.

1ucket, 2023-10-16:
The old database wouldn't start up. Tried to set up everything from scratch again. I could successfully add the relais and test them, and add an enclosure. Unfortunately I ran into multiple issues:

- When I try to add an area to the first created enclosure I got one of the two error messages:
`Area could not be added. Attribute Enclosure.id is required`
`Area could not be added. Value type for attribute Enclosure.id must be str. Got: `
- When trying to edit the terrarium a greyed out, empty settings window opens; the loading icon doesn't disappear
None of the things throw any errors in the console.

I then created a second terrarium. I could add an area to that one. But upon reloading/ editing TerrariumPI crashed:
```
  fname = self.search(name, self.lookup)
2023-10-16 23:21:53,196 - ERROR   - terrariumArea         - Error loading begin time: 8 - fromisoformat: argument must be str
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumArea.py", line 259, in _time_table
    begin = datetime.time.fromisoformat(self.setup[period]["begin"])
TypeError: fromisoformat: argument must be str
2023-10-16 23:21:53,199 - ERROR   - terrariumArea         - Error loading begin time: 20 - fromisoformat: argument must be str
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumArea.py", line 265, in _time_table
    end = datetime.time.fromisoformat(self.setup[period]["end"])
TypeError: fromisoformat: argument must be str
2023-10-16 23:21:53,200 - WARNING - terrariumArea         - Error generating timer periods. Either begin or end time is incorrect.
```

theyosh, 2023-10-16:
Crap crap crap... some forms logic... damm.. trying to fix it now...

theyosh, 2023-10-16:
Took a bit more time than wanted... but I think I have fixed it for now. Do a `git pull` and reload the browser. Make sure holding shift key or clearing the browser cache.

I hope this will fix it for now. Else we have to wait about the next evening. Getting late here...

1ucket, 2023-10-16:
Thanks, will test it tomorrow morning. Already turned everything off for the day. Have a good night!

theyosh, 2023-10-16:
> > Crap crap crap... some forms logic... damm.. trying to fix it now...
> 
> Man, I love you for being so dedicated! Thanks for being so quick in resolving those issues!

Doing it for the animals... :)

1ucket, 2023-10-17:
Needed to delete the database for enclosures to appear again, but then it worked. Now everything's up again and our beloved beardie gets his UV/ heat on time and not with a sudden burst but smoothly. Thank you very much!

closed 2023-10-17 by 1ucket