theyosh/TerrariumPI #848: [FEATURE]: Set Display Range for Sensor Graph

threedom, 2023-10-17:
**Describe the solution you'd like**
When viewing the graphs for sensors (e.g. Temperature) the graph shows a large range of temperatures whereas the relevant temperature range only takes up a very narrow portion of the graph. This is sometimes caused by missing data points during a reboot or when setting wide limits for temperatures (e.g. I have a very low lower limit for one of my temp sensors to avoid the low alarm all together because even when I set the area to "ignore" the low alarm, dependent areas still trigger.

It would be great if I could set the range of temperature that the graph displays (e.g. if my temps are usually between 85-95 F, to set the visible range of the graph to 80-100 F). This would avoid the very narrow view that is created when data is missing.

**Additional context**
Screenshot of the current graph and how big of a range it occasionally shows. 
[image]

theyosh, 2023-10-30:
Hmm, this is intended. Because this will 'signal' a bad sensor.

And normally you would use the limit low and high value for this. But that could trigger an error on this sensor, when the measured values are outside the limit low and high values.

Also, adding even more values to configure would also make it more complex. As there are already limit low/high as alarm low/high values. I am a not a fan of having even more values to enter...

theyosh, 2023-11-04:
So I made a different solution. In the settings screen there is now an option to limit the graph values.

When toggled on, the graph will limited to +/-20% of the alarm min and max values. But this is for all sensor graphs.

So you can test this by updating to the latest code with `git pull` and rerun the installer `sudo ./install.sh`. There are some new python libraries that needs to be installed.
Restart TP4 and you can enable the graph limiting in the settings screen.

theyosh, 2023-11-13:
Is this new graph option sufficient for you? Can you give an update on this?

theyosh, 2023-12-10:
Closing due to no reponse

closed 2023-12-10 by theyosh