theyosh/TerrariumPI #849: [BUG]: Dimmed lamp stays at 1% after ramp down.

1ucket, 2023-10-18:
**Setup:**
 - TerrariumPI version: 4.7.4
 - Install: Manual
 - Raspberry PI: 4

**Describe the bug**
The lamp is scheduled to turn off. In the relay page it's at 0%, but it's slightly glowing. If I turn it down to 0% manually it goes off, but afterwards will dim to 100% and then back to 1% over the next half hour.

**To Reproduce / Installation steps**
Add dimmable relay to an area with timer and ramp down. Observe.
[image]

**Expected behavior**
The lamp shuts off completely.

**Screenshots**
[image]
[image]

theyosh, 2023-10-30:
Hmm sorry for the late response, but does it help when you edit the relay and use a frequency of 7500 at the calibration part?

1ucket, 2023-10-30:
> Hmm sorry for the late response, but does it help when you edit the relay and use a frequency of 7500 at the calibration part?

I set that value. Will report back later (will be dimmed down in 3 hours)

1ucket, 2023-10-30:
Unfortunately it didn't fix the issue, the lamp is still slightly on. :/

theyosh, 2023-11-01:
Hmm, to bad.

Does it change when you do a restart? As the frequency is changed, it could be that it works after a restart.

Also, what happens when you manually turn the dimmer to zero?

theyosh, 2023-11-04:
And maybe playing with the offset value of the dimmer. Try adding -1. Not sure what will happen, but maybe it helps.

1ucket, 2023-11-14:
Sorry for not responding for a week. I'll try the possible solutions tomorrow and will give feedback afterwards.

theyosh, 2024-01-06:
Closing due to no response

closed 2024-01-06 by theyosh