theyosh/TerrariumPI #851: [BUG]: Can't get mail notifications to work

duenni, 2023-10-21:
**Setup:**
 - TerrariumPI version: 4.0.7
 - Install: Manual
 - Raspberry PI: 3B

**Describe the bug**
I can't get mail notifications to work and have several issues with it.

**To Reproduce / Installation steps**

1. Add a mail notification service and try to edit it afterwards.
No matter what field I try to edit I always get a 404 error. I am logged in as admin.
[image]

2. When adding a notification message and saving it with the mail service enabled I get the folowing:
[image]
I can save the notification with the mail service unticked. After editing this new notification and enabling the mail service I get a 404 again.
[image]
Editing a notification throws a slightly different error sometimes:
[image]

**Expected behavior**
Mail notification entry should be editable.
A notification should be added without error.

theyosh, 2023-10-23:
Hmm, found this also somewhere else. So I think I have fixed it. Do a git pull and restart. It should than work again,

duenni, 2023-10-24:
Works like a charm now. 👍 
Thanks for fixing it so fast!

closed 2023-10-24 by duenni