theyosh/TerrariumPI #855: Access to terrariumPi

jezzalencko, 2023-10-30:
Hi @theyosh, my terraiumpi has been running fine for a while, but have just tried accessing it on the network and [pi]:8090 is unreachable. I’ve tried restarting to no avail. I’ve also confirmed it exists on the network and has taken an IP.
Any other ideas I can try.?

theyosh, 2023-10-30:
Manual debugging: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

jezzalencko, 2023-10-31:
Hi,

I have tried this manual access and note I am getting this consistent error:

2023-10-31 21:18:55,422 - WARNING - terrariumEngine       - Could not take
a new measurement from sensor am2320 temperature named 'Air Temperature
(AM2320)' at address '5c'. Tried for 0.00 seconds. Skipping this update.

2023-10-31 21:18:56,902 - WARNING - terrariumArea         - All sensors for
area Heating named Door Heat at period low are in an error state. Relays
will not power on! Please check you sensors hardware!

Seems to be the AM2320 perhaps?

How do I escape out of this loop?

theyosh, 2023-10-31:
Is there not more? I would guess that running in debug/manual mode it should continue. It can take some time due to the retries.

Could you start in in manual mode, let it run for 5 - 10 minutes, and then try to paste the log in a file and upload the file here. I need some more debug messages to see if it does not continue.

jezzalencko, 2023-11-01:
Hi,
Yes it continually loops so only included the repeated parts. Here is a
bigger screen dump in a file after about 5 mins.

Kind Regards,
Jeremy.

theyosh, 2023-11-01:
No attachment...

jezzalencko, 2023-11-01:
[terrarium_log.txt](https://github.com/theyosh/TerrariumPI/files/13224026/terrarium_log.txt)

Strange - I did attach it to the email.

theyosh, 2023-11-01:
Could it be that there are 2 TerrariumPI processes running: Because my software cannot bind on port number 8090, and therefore you do not have a web interface

> 2023-11-01 10:39:59,000 - INFO    - terrariumWebserver    - Running webserver at 192.168.20.10:8090
Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 175, in __init__
    self.webserver.start()
  File "/home/pi/TerrariumPI/terrariumWebserver.py", line 306, in start
    quiet=True)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 767, in run
    run(self, **kwargs)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle.py", line 3175, in run
    server.run(app)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/bottle_websocket/server.py", line 17, in run
    server.serve_forever()
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/baseserver.py", line 398, in serve_forever
    self.start()
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/baseserver.py", line 336, in start
    self.init_socket()
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/pywsgi.py", line 1545, in init_socket
    StreamServer.init_socket(self)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/server.py", line 180, in init_socket
    self.socket = self.get_listener(self.address, self.backlog, self.family)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/server.py", line 192, in get_listener
    return _tcp_listener(address, backlog=backlog, reuse_addr=cls.reuse_addr, family=family)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/server.py", line 288, in _tcp_listener
    sock.bind(address)
  File "/home/pi/TerrariumPI/venv/lib/python3.7/site-packages/gevent/_socketcommon.py", line 563, in bind
    return self._sock.bind(address)
OSError: [Errno 99] Cannot assign requested address: ('192.168.20.10', 8090)

So make sure TerrariumPI is stopped before running it manually: https://theyosh.github.io/TerrariumPI/faq/how-to-start-stop-restart-and-disable-enable-terrariumpi/

theyosh, 2023-11-01:
or wait, you have entered a hardcode IP in the settings. If the Pi is on a different IP now, it will also not work. So make sure your PI is at IP `192.168.20.10`

jezzalencko, 2023-11-01:
Ah ok. But how do I do this from terminal?

I can't get back into the UI to be able to change IP address back. Can I
change rPi address from in terminal?

theyosh, 2023-11-01:
Added a FAQ which explains it: https://theyosh.github.io/TerrariumPI/faq/reset-webgui-address/

theyosh, 2023-11-03:
Is the FAQ clear enough? Did it fix your issue for the web interface? Your sensor is a different issue. That looks broken or poorly connected.

jezzalencko, 2023-11-05:
Thanks yes, I can at least access the web UI now.

Now to try to resolve the sensor issue...

theyosh, 2023-11-11:
Closing to solved the initial issue.

closed 2023-11-11 by theyosh