theyosh/TerrariumPI #856: [FEATURE]: Smooth daylight saving transition

1ucket, 2023-10-30:
**Is your feature request related to a problem? Please describe.**
Our bearded dragon is reaaally confused why the lamps stay on for an hour longer now that it was switched to summer time on Sunday.
[image]

**Describe the solution you'd like**
There could be a (if possible configurable) transition period of e.g. a month, where the lamps turn on/ off gradually earlier/ later.

cmonicob, 2023-12-17:
I think a good implementation for the correct functioning of the terrarium would be to disable the DST in the logic part, as the animal doesn't care about the clock.
I don't know if it would be possible to just keep the DST on the frontend part.

theyosh, 2023-12-27:
I have thought about this, but it is rather hard to make. Because this will introduce all kind of logic. Because DST is based on time zones. And shifting back and forth during the months before DST is switching is tricky.

So if you do not want any DST stuff, you could put your Pi in UTC timezone. But I do not know what will happen in the GUI with the dates and times.

And this is only happening twice a year, of which I think even animals can adjust to. So for now, I will not add extra logic for this.

theyosh, 2024-01-03:
So, I suggested to set your timezone to UTC. I would not do that anymore. As I use the timezone now for finding the correct Merros IOT server. We use the EU or US servers, based on the chosen timezone. So if your timezone does not start with `Europe/` TP4 will make a connection to a US server. And I do not know what will happen when your relays or sensors are connected to the EU servers.

theyosh, 2024-01-06:
For now, I am closing this issue, as I think it is to much programming and testing for an event that only happens twice a year. And it is also hard to test, as I have to simulate daylight saving actions from the system...

So, sorry, but this is something I will not add.

closed 2024-01-06 by theyosh