theyosh/TerrariumPI #861: [BUG]: Water Tank setup in Enclosure. No access to GUI now and Service seems to bounce.

AaronNolet, 2023-12-05:
Issue: Setup Water Tank in Enclosure and now cannot access GUI and Service seems to Bounce...

Debug Log of issue:

2023-12-05 11:07:33,860 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.8.0 on a Raspberry Pi 4 Model B Rev 1.1 ...
2023-12-05 11:07:33,908 - INFO    - terrariumEngine       - Loaded 30 settings in 0.05 seconds.
2023-12-05 11:07:33,935 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Sarnia,CAD&appid=d621ffa5cb6f1627031ea37fa611937a
2023-12-05 11:07:40,158 - INFO    - weather.openweathermap_org_weather - Using Openweathermap One Call API 2.5
2023-12-05 11:07:41,141 - INFO    - weather.openweathermap_org_weather - Loaded new historical weather data (48 measurements) from 2023-12-02 14:00:00 till 2023-12-04 13:00:00 in 0.98 seconds.
2023-12-05 11:07:41,142 - INFO    - weather               - Loaded new weather data in 7.203 seconds.
2023-12-05 11:07:42,193 - ERROR   - terrariumNotification - Error loading display OLED: Unable to load display OLED SSD1327 (I2C) at address '3c' (16x2): I2C device not found on address: 0x3C
2023-12-05 11:07:43,080 - INFO    - terrariumEngine       - Loading existing sensors from database.
2023-12-05 11:07:43,898 - WARNING - terrariumEngine       - Measurement for sensor 1wire temperature named 'Basking' at address '28-3c01d6072701' of 33.81F is outside valid range 60.00F to 140.00F during startup in 0.81 seconds. Will be updated in the next round.
2023-12-05 11:07:44,730 - WARNING - terrariumEngine       - Measurement for sensor 1wire temperature named 'House' at address '28-3c01d607dfa6' of 24.69F is outside valid range 60.00F to 95.00F during startup in 0.83 seconds. Will be updated in the next round.
2023-12-05 11:07:44,810 - INFO    - terrariumEngine       - Loaded sensor hc-sr04 distance named 'Water Table Distance' at address '36,35' with value 6.35cm in 0.08 seconds.
2023-12-05 11:07:45,348 - INFO    - terrariumEngine       - Loaded sensor am2302 humidity named 'Humidity' at address '38' with value 28.10% in 0.54 seconds.
2023-12-05 11:07:45,349 - INFO    - terrariumEngine       - Scanning for new sensors ...
2023-12-05 11:07:56,097 - INFO    - terrariumEngine       - Loaded 4 sensors in 13.02 seconds.
2023-12-05 11:07:56,098 - INFO    - terrariumEngine       - Loading existing relays from database.
2023-12-05 11:07:56,482 - INFO    - terrariumEngine       - Loaded relay gpio relay named 'Lights' at address '40' value 0.00 in 0.36 seconds.
2023-12-05 11:07:56,484 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2023-12-05 11:08:07,550 - INFO    - terrariumEngine       - Loaded 1 relays in 11.45 seconds.
2023-12-05 11:08:07,552 - INFO    - terrariumEngine       - Loading existing buttons from database.
2023-12-05 11:08:07,565 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.01 seconds.
2023-12-05 11:08:07,565 - INFO    - terrariumEngine       - Loading existing webcams from database.
2023-12-05 11:08:11,169 - INFO    - terrariumEngine       - Loaded rpicam webcam 'Eddy Von Prickles' at address 'rpicam' in 3.48 seconds.
2023-12-05 11:08:11,170 - INFO    - terrariumEngine       - Loaded 1 webcams in 3.60 seconds.
2023-12-05 11:08:11,171 - INFO    - terrariumEngine       - Loading existing enclosures from database.
2023-12-05 11:08:11,183 - INFO    - terrariumArea         - Toggle the relays for area Lighting named Main part day to state on.
2023-12-05 11:08:11,215 - INFO    - terrariumArea         - Set the relay Lights to 100.0 with a delay of 0.00 minutes
2023-12-05 11:08:11,216 - INFO    - terrariumArea         - Updated area Lighting named Main in timer mode at enclosure EVP's Hut in 0.03 seconds.
2023-12-05 11:08:11,216 - INFO    - terrariumArea         - Adjusting the sensors based on day/night difference. Changing by 27.5 going from night to day
2023-12-05 11:08:11,233 - INFO    - terrariumArea         - Updated area Heating named Basking in sensors mode at enclosure EVP's Hut in 0.02 seconds.
Traceback (most recent call last):
  File "/home/aaron/TerrariumPI/terrariumPI.py", line 19, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/aaron/TerrariumPI/terrariumEngine.py", line 158, in __init__
    self.__load_existing_enclosures()
  File "/home/aaron/TerrariumPI/terrariumEngine.py", line 1220, in __load_existing_enclosures
    new_enclosure.update()
  File "/home/aaron/TerrariumPI/terrariumEnclosure.py", line 109, in update
    area_states[area_id] = self.areas[area_id].update(read_only)
  File "/home/aaron/TerrariumPI/terrariumArea.py", line 686, in update
    self.state["sensors"] = self.current_value(self.setup["sensors"])
  File "/home/aaron/TerrariumPI/terrariumArea.py", line 1376, in current_value
    volume_per_distance = self.setup["watertank_volume"] / (
ZeroDivisionError: float division by zero

theyosh, 2023-12-05:
Ok, do a git pull and restart. I hope I found the bug and should be fixed now.

AaronNolet, 2023-12-05:
This has been resolved... Thank you. I Imagine it was to to the 0 height value... 👍

AaronNolet, 2023-12-05:
R

closed 2023-12-05 by AaronNolet