theyosh/TerrariumPI #864: [BUG]: Attempt to play mp3 playlist error log

AaronNoletPDIT, 2023-12-11:
**Setup:**
 - TerrariumPI version: 4.8.0
 - Install: Manual
 - Raspberry PI: 4

**Describe the bug**
Dec 11 16:24:15 raspberrypi python[2336]: 2023-12-11 16:24:15,454 - INFO    - terrariumArea         - Toggle the player for area Audio named Day period day to state on.
Dec 11 16:24:15 raspberrypi python[2336]: Exception in thread Thread-20:
Dec 11 16:24:15 raspberrypi python[2336]: Traceback (most recent call last):
Dec 11 16:24:15 raspberrypi python[2336]:   File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
Dec 11 16:24:15 raspberrypi python[2336]:     self.run()
Dec 11 16:24:15 raspberrypi python[2336]:   File "/usr/lib/python3.9/threading.py", line 892, in run
Dec 11 16:24:15 raspberrypi python[2336]:     self._target(*self._args, **self._kwargs)
Dec 11 16:24:15 raspberrypi python[2336]:   File "/home/aaron/TerrariumPI/terrariumAudio.py", line 80, in __run
Dec 11 16:24:15 raspberrypi python[2336]:     self.volume(playlist.get("volume", 80))
Dec 11 16:24:15 raspberrypi python[2336]:   File "/home/aaron/TerrariumPI/terrariumAudio.py", line 131, in volume
Dec 11 16:24:15 raspberrypi python[2336]:     terrariumAudio.volume(int(self.__hw), int(value))
Dec 11 16:24:15 raspberrypi python[2336]:   File "/home/aaron/TerrariumPI/terrariumAudio.py", line 54, in volume
Dec 11 16:24:15 raspberrypi python[2336]:     mixer.setvolume(int(value), alsaaudio.MIXER_CHANNEL_ALL)
Dec 11 16:24:15 raspberrypi python[2336]: alsaaudio.ALSAAudioError: Volume out of range

theyosh, 2023-12-13:
Hi, I changed the volume setting code and it should now work, or at least not crash anymore.

Do a `git pull` and restart TP4. It should work again.

AaronNoletPDIT, 2023-12-14:
This has resolved the issue. Thank you!!!

closed 2023-12-14 by AaronNoletPDIT