theyosh/TerrariumPI #866: [FEATURE]: Option to change wind speed to mph.

STGMavrick, 2023-12-20:
Currently under settings there are options for kph, m/s, and beaufort. Us yanks go by miles per hour. Openweathermap does show that the API parameter "wind" has an imperial flag.

Thanks for the consideration!

theyosh, 2023-12-27:
Hi I added the option to the settings. If you do a `git pull` and restart TP4 you can set your wind speed to miles per hour. But it was always al possible by setting the wind speed to kilometer per hour and distance to inch. This should also get you miles per hour.

So I guess you do not use inches for distance?

STGMavrick, 2023-12-27:
I do have it set on inches & kph. I honestly thought it was still showing in metric. m/h are the units next to the weather speed. I assumed this was meters per hour as our units for miles per hour is always displayed in mph.

theyosh, 2023-12-27:
Ah yes, I shall update to `mph` which is more clear

theyosh, 2023-12-27:
Do a `git pull` and restart. It should now show `mph` for wind speed

theyosh, 2024-01-03:
Did you try the new `mph` setting? I would like to close this issue

STGMavrick, 2024-01-03:
Hey Yosh - I apologize. Holidays were hectic. I did a git pull today and unfortunately the web interface isn't coming up. SSH login screen shows it's up and working. I've tried the setup twice + reboots. How can I troubleshoot what's causing the web interface to not come up?
[image]

theyosh, 2024-01-03:
I suggest to rerun the installer. If that does not help, try running in debug: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

STGMavrick, 2024-01-03:
In console mode, i get the following at the end of the debug log

[image]

theyosh, 2024-01-03:
Not sure why, but your host setting is wrong. https://theyosh.github.io/TerrariumPI/faq/reset-webgui-address/

STGMavrick, 2024-01-03:
Yeah, not sure either but that fixed it. Webgui works again. mph shows up fine! I appreciate your work & help. Close away!

closed 2024-01-03 by theyosh