theyosh/TerrariumPI #868: FYI: Meross may stop working after 1-1-2024

theyosh, 2023-12-27:
Received a mail from Meross about changing their API at 1-1-2024. I can't fix this. Lets hope it can be reversed engineered. More info hopefully at https://github.com/albertogeniola/MerossIot/issues/340

theyosh, 2023-12-27:
After updating the Meross library, there is a new connection parameter. This is the location to connect to which is split to us, eu, and asia.

For now, we try to detect the content where the Pi is on, to connect to the right Meross API. If not in the EU, you will be send to the US server.

I hope this will solve the issue with the old API which is closing this year.

For every body: do a `git pull` and rerun the installer. Than restart TP4.

DaddyBoard, 2023-12-30:
Just got this email too. Does anyone know if the dev of this current version is aware and is confident we can keep using this to toggle wifi plugs? This will break some serious automations for me that I put HOURS into :(

albertogeniola, 2023-12-30:
Hi! 
You can upgrade to the latest MerossIot api which handles that correctly.

theyosh, 2023-12-30:
Ok, than we should be save with the release of TP4 version 4.8.1 which is using the new API with the new connection urls. So that means upgrade your TP4 today or tomorrow.

Thanks @albertogeniola

theyosh, 2024-01-03:
Confirmed, Meross is nicely working with the latest python library.

closed 2024-01-03 by theyosh