theyosh/TerrariumPI #869: I am very new to this - be gentle. Docker-Compose issue

ThunderClap86, 2023-12-28:
I have never used anything like this before. Used the manual install steps because I could not figure out the docker-compose. Went through the steps and made the hello-world run, but now that I am getting a 4.8.1 update notification in the upper right. I cannot make the process work. I apparently did not put the docker-compose.yml file in the correct location. I followed the upgrade steps of [this page](https://github.com/theyosh/TerrariumPI/releases/tag/4.8.1) for both the docker and the manual install. 
```
pi@lizard:~ $ docker compose pull
-bash: docker: command not found
```
and using the manual steps produces this 
```
pi@lizard:~ $ sudo service terrariumpi stop
pi@lizard:~ $ git pull
fatal: not a git repository (or any of the parent directories): .git
```
am I missing something obvious? Anyone have a direction for me to start setting this up?

I am waiting on some sensors and a relay board to arrive later this week, was just hoping to have the Pi working and ready before attempting the hookup of the sensors and relay. Plan on using sparkfun qwiic devices if anyone has done that yet.

theyosh, 2023-12-28:
It looks like you do not have docker installed. So that means a manual installation. In order to update that, you need to go the TerrariumPI folder which is mostly at: `/home/pi/TerrariumPI`. run the command: `cd /home/pi/TerrariumPI` and then try `git pull`.

If that works, run the command: `sudo ./install.sh`. A reboot at the end is NOT needed.

Then start TP4 again with: `sudo service terrariumpi start`

ThunderClap86, 2023-12-28:
Thanks for the reply, the 4.8.1 appears to be working with the directory change command. Ill update when finished and close this issue once confirmed. 

I've since performed the uninstall for docker; but when I ran it before following the instructions, I got to the "hello world" section and was able to receive a reply from the container that said (paraphrased) if you're getting this message then you've correctly set up docker. ` docker-compose --version` would output a version number. But attempting to run anything else would result in file not found or permission errors like `/usr/local/bin/docker-compose: Permission denied`, even when running through `sudo`. I ended up getting frustrated enough to look into the filesystem through WinSCP to make sure the path existed. I edited the .yaml file in VScode and made sure the extension was saved and exported correctly. 

It became easier to manually do the install after that. I ended up doing 3 different TF cards of installs to troubleshoot the 32 bit and 64 bit OS lite versions just to see if I was doing something else wrong.

ThunderClap86, 2023-12-28:
Upon reboot there is a new error with no webpage at .local:8090 and the SSH is giving me this:
`~ $ sudo service terrariumpi start
Job for terrariumpi.service failed because the control process exited with error code.
See "systemctl status terrariumpi.service" and "journalctl -xe" for details.`

theyosh, 2023-12-28:
Hmm, that is not so good :(. But lets see if we can fix this.

First could you run the command: `sudo apt -y install libopenblas-dev`

When that is finished ``sudo service terrariumpi restart` and see if it will start. If you still get the same error, the next step is to run it in debug mode: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/ and copy and post the output of the console here.

ThunderClap86, 2023-12-28:
I attempted to perform the sudo ./install.sh again in case it was incomplete the first time. When that finished it asked for the reboot and I selected <no>. Performed the first command you listed and this is the output.  I performed a `sudo reboot now` and I received the same message again. 
```
lizard:~ $ sudo apt -y install libopenblas-dev
E: Type 'sudo' is not known on line 1 in source list /etc/apt/sources.list.d/docker.list
E: The list of sources could not be read.
E: Type 'sudo' is not known on line 1 in source list /etc/apt/sources.list.d/docker.list
E: The list of sources could not be read.
```
Just for giggles I attempted to run the `sudo service terrariumpi restart` and received the same output as previously posted. 
####################
DEBUG OUTPUT
```
pi@lizard:~ $ sudo service terrariumpi stop
pi@lizard:~ $ cd /home/pi/TerrariumPI/
pi@lizard:~/TerrariumPI $ source venv/bin/activate
(venv) pi@lizard:~/TerrariumPI $ python terrariumPI.py
OpenCV bindings requires "numpy" package.
Install it via command:
    pip install numpy
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/numpy/core/__init__.py", line 24, in <module>
    from . import multiarray
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/numpy/core/multiarray.py", line 10, in <module>
    from . import overrides
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/numpy/core/overrides.py", line 8, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libopenblas.so.0: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/numpy/__init__.py", line 130, in <module>
    from numpy.__config__ import show as show_config
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/numpy/__config__.py", line 4, in <module>
    from numpy.core._multiarray_umath import (
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/numpy/core/__init__.py", line 50, in <module>
    raise ImportError(msg)
ImportError:

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.9 from "/home/pi/TerrariumPI/venv/bin/python"
  * The NumPy version is: "1.26.2"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: libopenblas.so.0: cannot open shared object file: No such file or directory

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 16, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 2, in <module>
    from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
  File "/home/pi/TerrariumPI/hardware/webcam/__init__.py", line 26, in <module>
    import cv2
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/cv2/__init__.py", line 11, in <module>
    import numpy
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/numpy/__init__.py", line 135, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
```

theyosh, 2023-12-28:
Somehow, some docker leftover is causing issues. Could you run: `sudo rm /etc/apt/sources.list.d/docker.list`

and than `sudo apt -y install libopenblas-dev` again?

ThunderClap86, 2023-12-28:
Just so you know. I am still waiting on sensor and relay hardware. I am NOT too proud to just start over without saving any configurations. The only thing I had connected to this pi3 or pi4 was a DSI raspicam, wouldn't hurt my feelings at all to just start over. Hopefully I am not causing you too much trouble. I wanted the 4.8.1 upgrade because you had some of the updates that I was really wanting haha. 

Thank you so much for taking any of your time to help me sort this.

ThunderClap86, 2023-12-28:
OKAY! WOO! looks like the missing `libopenblas-dev` and the remnants of the docker install is what was blocking it. We are back up and running with your nicely designed webpage frontend. ISSUE SOLVED. 

Is there anything else you'd like me to pull/output for you if it helps any? Before I close the issue?

theyosh, 2023-12-28:
Nope, that is it. Somehow the package `libopenblas-dev` is needed. I did not experience this at home, but looks a Pi4 related thing.

ThunderClap86, 2023-12-28:
Thank you again, I'll have to find the "Buy me a Coffee" link. This has been a fun experience so far, and cannot wait for the hardware to get here.

closed 2023-12-28 by ThunderClap86