theyosh/TerrariumPI #870: New fresh install fails to start terrariumpi service

worzel910, 2023-12-28:
**Setup:**
 - TerrariumPI version: 4.6.3
 - Install: [Manual] 
 - Raspberry PI: 4b
**Describe the bug**
Clean install of Pi OS 32 bit (Legacy) Lite  followed be install of terrarium via git.

After installation terrariumpi service fails to start. 

**To Reproduce / Installation steps**
Clean install of Pi os lite then a install of terrariumpi. 

**Expected behavior**
Service to start after boot

worzel910, 2023-12-28:
After looking through the debug output discovered the libopenblas-dev package  is needed.
After install of the package terrariumpi.service now starts.
Needs to be added as a dependency ?

theyosh, 2023-12-28:
Please add some more debug information: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

theyosh, 2023-12-28:
ah that was quick :P

Hmm, OS package updates are hard to test. That takes about an hour... So I am always hoping that they stay to work.
I will add the package to the installation list

theyosh, 2023-12-28:
Added package `libopenblas-dev` to the installer script

worzel910, 2023-12-28:
> ah that was quick :P
> 
> Hmm, OS package updates are hard to test. That takes about an hour... So I am always hoping that they stay to work. I will add the package to the installation list

Nice to be an easy fix though!

theyosh, 2023-12-28:
But the strange thing is that I do not need it on my existing systems. So that is a bit strange... but I will just leave it as a dependency to install. But it should be solved now.

worzel910, 2023-12-28:
Oddly, didn't need it for the desktop image. Issue arose when I went to the lite version.

closed 2023-12-29 by worzel910