theyosh/TerrariumPI #872: Qwiic Quad Relay unable to set up or activate within TerrariumPi

ThunderClap86, 2023-12-31:
Sorry, me again. 

Hoping to have some support on how to activate the Sparkfun Qwiic Quad Relay. I have performed `i2cdetect -y 1` and received a return for 6d (quad relay board) and 77 (BME280). I was able to correctly configure the BME sensor within the GUI and it is providing a readout. The qwiic [relay board](https://www.sparkfun.com/products/16566) is being seen by the Pi, but the "Scan Relays" function is showing no relays, and using the "New Relay" menu item for manual addition I am using the following settings:

> Hardware: Qwiic Relay
> Address: 1,0x6d - 1,6d - 6d - 0x6d - 0x6d, 0x01 (attempted something different each time)
> Name: Relay
> Wattage: 1
> Water Flow: 0

I have tried several variations for the address and I have read throug the setup guide quite a few ways. 
```
2023-12-31 15:56:09,567 - ERROR   - hardware.relay        - Error getting new data from relay Qwiic Relay relay named 'Relay1' at address '0x6d'. Error: unknown
2023-12-31 15:56:09,563 - ERROR   - hardware.relay        - Error getting new data from relay Qwiic Relay relay named 'Relay1' at address '0x6d'. Error: invalid literal for int() with base 10: '0x6d'
######### OR #########
2023-12-31 15:57:39,593 - ERROR   - hardware.relay        - Error getting new data from relay Qwiic Relay relay named 'Relay1' at address '1,0x6d'. Error: [Errno 22] Invalid argument

```

I also have a [DockerPi 4 Channel Relay HAT](https://www.seeedstudio.com/DockerPi-4-Channel-Relay-p-4096.html) that I understand is on the unsupported list, but same thing. The relay is seen by the Pi, and I can control the on/off through terminal but the TerrariumPi is not seeing it.

ThunderClap86, 2024-01-02:
Update on the Qwiic... 1,0x6d wouldn't work. Tried 1,0x6D (capital letter) and it worked. 

Also if anyone reads this, I read on another closed previous issue the Qwiic board says it can be powered by 12v. Don't, it stopped working just like their issue. Now powered directly by 5v with a jumper and it's back switching again.

closed 2024-01-02 by ThunderClap86

theyosh, 2024-01-03:
Hmm, I think the casing 1,0x6D or 1,0x6d should both work. So I get the feeling the 5v jumper was the fix. I have updated the documentation with this 5volt option.

But nice you got it working.