theyosh/TerrariumPI #874: BUG: Service Start Failure (new Install)

AaronNoletPDIT, 2024-01-05:
Setup:
 - TerrariumPI version: 4.8.1
 - Install: Manual
 - Raspberry PI: 4
 - OS: Raspberry PI OS (Legacy, 32-bit) Lite

**Describe the bug**
Service Fails to start

**Debug Data**

Jan 05 16:32:08 TerraPI python[2898]: 2024-01-05 16:32:08,653 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.8.1 on a Raspberry Pi 4 Model B Rev 1.1 ...
Jan 05 16:32:09 TerraPI python[2898]: 2024-01-05 16:32:08,747 - INFO    - terrariumEngine       - Loaded 30 settings in 0.09 seconds.
Jan 05 16:32:11 TerraPI python[2898]: 2024-01-05 16:32:11,055 - INFO    - terrariumEngine       - Loading existing sensors from database.
Jan 05 16:32:11 TerraPI python[2898]: 2024-01-05 16:32:11,069 - INFO    - terrariumEngine       - Scanning for new sensors ...
Jan 05 16:32:12 TerraPI python[2898]: 2024-01-05 16:32:12,760 - INFO    - terrariumEngine       - Added new new sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-3c01d6072701' to database with value 27.94C in 0.86 seconds.
Jan 05 16:32:14 TerraPI python[2898]: 2024-01-05 16:32:14,395 - INFO    - terrariumEngine       - Added new new sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-3c01d607580b' to database with value 23.56C in 0.80 seconds.
Jan 05 16:32:16 TerraPI python[2898]: 2024-01-05 16:32:16,028 - INFO    - terrariumEngine       - Added new new sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-3c01d60777cc' to database with value 27.12C in 0.81 seconds.
Jan 05 16:32:17 TerraPI python[2898]: 2024-01-05 16:32:17,661 - INFO    - terrariumEngine       - Added new new sensor 1wire temperature named '1-Wire sensor measuring temperature' at address '28-3c01d607dfa6' to database with value 26.88C in 0.81 seconds.
Jan 05 16:32:26 TerraPI python[2898]: 2024-01-05 16:32:26,831 - INFO    - terrariumEngine       - Loaded 4 sensors in 15.78 seconds.
Jan 05 16:32:26 TerraPI python[2898]: 2024-01-05 16:32:26,833 - INFO    - terrariumEngine       - Loading existing relays from database.
Jan 05 16:32:26 TerraPI python[2898]: 2024-01-05 16:32:26,847 - INFO    - terrariumEngine       - Scanning for new relays ...
Jan 05 16:32:38 TerraPI python[2898]: Traceback (most recent call last):
Jan 05 16:32:38 TerraPI python[2898]:   File "/home/aaron/TerrariumPI/terrariumPI.py", line 19, in <module>
Jan 05 16:32:38 TerraPI python[2898]:     terrariumEngine = terrariumEngine(__version__)
Jan 05 16:32:38 TerraPI python[2898]:   File "/home/aaron/TerrariumPI/terrariumEngine.py", line 140, in __init__
Jan 05 16:32:38 TerraPI python[2898]:     self.scan_new_relays()
Jan 05 16:32:38 TerraPI python[2898]:   File "/home/aaron/TerrariumPI/terrariumEngine.py", line 845, in scan_new_relays
Jan 05 16:32:38 TerraPI python[2898]:     for relay in terrariumRelay.scan_relays(callback=self.callback_relay):
Jan 05 16:32:38 TerraPI python[2898]:   File "/home/aaron/TerrariumPI/hardware/relay/__init__.py", line 307, in scan_relays
Jan 05 16:32:38 TerraPI python[2898]:     for relay in relay_device._scan_relays(callback, **kwargs):
Jan 05 16:32:38 TerraPI python[2898]:   File "/home/aaron/TerrariumPI/hardware/relay/voltcraft_sem6000_relay.py", line 58, in _scan_relays
Jan 05 16:32:38 TerraPI python[2898]:     device[1],
Jan 05 16:32:38 TerraPI python[2898]: IndexError: list index out of range

theyosh, 2024-01-05:
My fault. Do a new `git pull` and restart. Should work again.

ThunderClap86, 2024-01-05:
Just had a similar failure to start and tried this:
> pi@lizard:~ $ cd ./TerrariumPI
> pi@lizard:~/TerrariumPI $ git pull
> hint: Pulling without specifying how to reconcile divergent branches is
> hint: discouraged. You can squelch this message by running one of the following
> hint: commands sometime before your next pull:
> hint:
> hint:   git config pull.rebase false  # merge (the default strategy)
> hint:   git config pull.rebase true   # rebase
> hint:   git config pull.ff only       # fast-forward only
> hint:
> hint: You can replace "git config" with "git config --global" to set a default
> hint: preference for all repositories. You can also pass --rebase, --no-rebase,
> hint: or --ff-only on the command line to override the configured default per
> hint: invocation.
> Already up to date.

Debut output: shows me missing the lib3relind module? 
```
(venv) pi@lizard:~/TerrariumPI $ python terrariumPI.py
2024-01-05 16:48:12,900 - INFO    - terrariumEngine       - Starting up Terrariu                                                                                                                                                             mPI 4.8.1 on a Raspberry Pi 4 Model B Rev 1.4 ...
2024-01-05 16:48:12,944 - INFO    - terrariumEngine       - Loaded 30 settings i                                                                                                                                                             n 0.04 seconds.
2024-01-05 16:48:12,968 - INFO    - terrariumEngine       - Loading weather data                                                                                                                                                              from source https://api.openweathermap.org/data/2.5/weather?q=cabot&appid=d7482                                                                                                                                                             2d216853d837a441741ecfa666c
2024-01-05 16:48:12,972 - ERROR   - terrariumEngine       - Loading weather exce                                                                                                                                                             ption: Weather url 'https://api.openweathermap.org/data/2.5/weather?q=cabot&appi                                                                                                                                                             d=d74822d216853d837a441741ecfa666c' is not valid! Please check your source
2024-01-05 16:48:13,413 - INFO    - terrariumEngine       - Loading existing sen                                                                                                                                                             sors from database.
2024-01-05 16:48:13,451 - WARNING - terrariumEngine       - Measurement for sens                                                                                                                                                             or bme280 temperature named 'Temp' at address '77' of 21.16F is outside valid ra                                                                                                                                                             nge 40.00F to 100.00F during startup in 0.03 seconds. Will be updated in the nex                                                                                                                                                             t round.
2024-01-05 16:48:13,463 - INFO    - terrariumEngine       - Loaded sensor bme280                                                                                                                                                              humidity named 'HUM' at address '77' with value 24.71% in 0.01 seconds.
2024-01-05 16:48:13,465 - INFO    - terrariumEngine       - Loaded sensor bme280                                                                                                                                                              pressure named 'Press' at address '77' with value 1000.12hPa in 0.00 seconds.
2024-01-05 16:48:13,467 - INFO    - terrariumEngine       - Scanning for new sen                                                                                                                                                             sors ...
2024-01-05 16:48:22,582 - INFO    - terrariumEngine       - Loaded 3 sensors in 9.17 seconds.
2024-01-05 16:48:22,584 - INFO    - terrariumEngine       - Loading existing relays from database.
2024-01-05 16:48:22,598 - INFO    - terrariumEngine       - Scanning for new relays ...
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 19, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 140, in __init__
    self.scan_new_relays()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 845, in scan_new_relays
    for relay in terrariumRelay.scan_relays(callback=self.callback_relay):
  File "/home/pi/TerrariumPI/hardware/relay/__init__.py", line 304, in scan_relays
    for hardware_type, relay_device in terrariumRelay.available_hardware.items():
  File "/home/pi/TerrariumPI/terrariumUtils.py", line 47, in __get__
    return classmethod(self.fget).__get__(None, owner)()
  File "/home/pi/TerrariumPI/hardware/relay/__init__.py", line 63, in available_hardware
[... cut ...]
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/home/pi/TerrariumPI/hardware/relay/8relay-stack_relay.py", line 30, in <module>
    from lib8relind import set as relay8SetV4
ModuleNotFoundError: No module named 'lib8relind'
```

theyosh, 2024-01-05:
Sorry, I was also changing some code. Sorry.

So do another `git pull` and rerun the installer `sudo ./install.sh`. A reboot is NOT needed. After reinstalling I hope it should work again.

ThunderClap86, 2024-01-05:
> Sorry, I was also changing some code. Sorry.
> 
> So do another `git pull` and rerun the installer `sudo ./install.sh`. A reboot is NOT needed. After reinstalling I hope it should work again.

That IS my luck with working on getting my script relay going. Thanks.

theyosh, 2024-01-06:
@AaronNoletPDIT is your setup running again?

AaronNoletPDIT, 2024-01-06:
Will check later today I hope

AaronNoletPDIT, 2024-01-07:
Seems to be good now... Thanks!

closed 2024-01-07 by AaronNoletPDIT