theyosh/TerrariumPI #875: VEML6075 UVA and UVB light sensor alternatives ?

nikkisstuff, 2024-01-06:
Hi I am having a hard time finding a veml6075 that is not outrageously priced are there any suitable alternatives?

theyosh, 2024-01-06:
Hmm, not at the moment. I found some UV Index sensors which could be supported. But I could not find a different sensor for UVA and UVB light.

One alternative that could measure UV index is: https://wiki.seeedstudio.com/Grove-Sunlight_Sensor/#si1145---play-with-raspberry-pi

Or https://www.adafruit.com/product/1777

But for all sensors I need to add new code. So even when there is a different sensor, I need to make some code. So if you find a working sensor with some python example code, I can mostly add them to TerrariumPI

nikkisstuff, 2024-01-06:
Thanks. I have found this one as another possibility [](https://www.adafruit.com/product/1918)

theyosh, 2024-01-06:
> Thanks. I have found this one as another possibility (https://www.adafruit.com/product/1918)

That is an analog sensor. Does not work with a Raspberry Pi directly. This needs an analog to digital converter. :(

nikkisstuff, 2024-01-07:
After searching for a few hours for the best alterative i found this Its not low cost but they are available and run on i2c and cover uva uvb and uvc https://www.sparkfun.com/products/23517

theyosh, 2024-01-07:
Hmm, sorry to say, but I can't find python code for that sensor. Only Arduino :( So than you need to to use an Arduino and hook that to the Pi. It can be done, but that will cost some programming. And I can't help with that part, as I do not use Arduino boards.

Sorry to say, but I think there is only one solution... pay for expensive sensors :( ...

nikkisstuff, 2024-01-07:
I converted this sketch using chatgpt  https://github.com/kriswiner/AS7331/tree/main/AS7331_DataReady_Ladybug

I don't know enough about programming to make the suggested changes

This Python code includes dummy classes and functions for the STM32, I2Cdev, and AS7331 classes since these classes are not available in Python. You will need to replace these placeholders with the actual implementations based on your requirements and available libraries in Python.

`import time
import smbus2

# AS7331 definitions
AS7331_intPin = 8  # interrupt pin definitions

# Specify sensor parameters
mmode = "AS7331_CONT_MODE"  # choices are modes are CONT, CMD, SYNS, SYND
cclk = "AS7331_1024"  # choices are 1.024, 2.048, 4.096, or 8.192 MHz
sb = 0x01  # standby enabled 0x01 (to save power), standby disabled 0x00
breakTime = 40  # sample time == 8 us x breakTime (0 - 255, or 0 - 2040 us range), CONT or SYNX modes

gain = 8  # ADCGain = 2^(11-gain), by 2s, 1 - 2048 range, 0 < gain = 11 max, default 10
time_val = 9  # 2^time in ms, so 0x07 is 2^6 = 64 ms, 0 < time = 15 max, default 6

# sensitivities at 1.024 MHz clock
lsbA = 304.69 / (2 ** (11 - gain)) / (2 ** time_val / 1024.0) / 1000.0  # uW/cm^2
lsbB = 398.44 / (2 ** (11 - gain)) / (2 ** time_val / 1024.0) / 1000.0
lsbC = 191.41 / (2 ** (11 - gain)) / (2 ** time_val / 1024.0) / 1000.0

# Logic flags to keep track of device states
tempData = 0
UVAData = 0
UVBData = 0
UVCData = 0
allData = [0, 0, 0, 0]
temp_C = 0
AS7331_Ready_flag = False
status = 0

# Dummy class for RTC functions
class RTC:
    @staticmethod
    def getDay():
        return 1  # Placeholder value

    @staticmethod
    def getMonth():
        return 1  # Placeholder value

    @staticmethod
    def getYear():
        return 2023  # Placeholder value

    @staticmethod
    def getHours():
        return 0  # Placeholder value

    @staticmethod
    def getMinutes():
        return 0  # Placeholder value

    @staticmethod
    def getSeconds():
        return 0  # Placeholder value

# Dummy class for STM32 functions
class STM32:
    @staticmethod
    def getVREF():
        return 3.3  # Placeholder value

    @staticmethod
    def getTemperature():
        return 25.0  # Placeholder value

    @staticmethod
    def sleep():
        time.sleep(1)  # Placeholder sleep time

# Dummy class for I2Cdev functions
class I2Cdev:
    def __init__(self, bus):
        self.bus = bus

    def read_byte_data(self, addr, reg):
        # Placeholder read_byte_data function
        pass

    def write_byte_data(self, addr, reg, value):
        # Placeholder write_byte_data function
        pass

# Dummy class for AS7331 functions
class AS7331:
    def __init__(self, i2cdev):
        self.i2cdev = i2cdev

    def powerUp(self):
        # Placeholder powerUp function
        pass

    def reset(self):
        # Placeholder reset function
        pass

    def getChipID(self):
        # Placeholder getChipID function
        return 0x21  # Placeholder value

    def setConfigurationMode(self):
        # Placeholder setConfigurationMode function
        pass

    def init(self, mmode, cclk, sb, breakTime, gain, time):
        # Placeholder init function
        pass

    def setMeasurementMode(self):
        # Placeholder setMeasurementMode function
        pass

    def getStatus(self):
        # Placeholder getStatus function
        return 0  # Placeholder value

    def readAllData(self, data):
        # Placeholder readAllData function
        pass

    def readUVAData(self):
        # Placeholder readUVAData function
        return 0  # Placeholder value

    def readUVBData(self):
        # Placeholder readUVBData function
        return 0  # Placeholder value

    def readUVCData(self):
        # Placeholder readUVCData function
        return 0  # Placeholder value

    def readTempData(self):
        # Placeholder readTempData function
        return 0  # Placeholder value

    def oneShot(self):
        # Placeholder oneShot function
        pass

# Constants
myLed = 13

# Variables
seconds, minutes, hours, day, month, year = 0, 0, 0, 0, 0, 0
Seconds, Minutes, Hours, Day, Month, Year = 0, 0, 0, 0, 0, 0
alarmFlag = False

VDDA, VBAT, VBUS, Temperature = 0, 0, 0, 0

# Instantiate objects
I2C_BUS = smbus2.SMBus(1)
i2c_0 = I2Cdev(I2C_BUS)
AS7331 = AS7331(i2c_0)

# Setup
SerialDebug = True
AS7331_Ready_flag = False

# Main loop
while True:
    # Placeholder for the main loop
    pass
`

nikkisstuff, 2024-01-07:
I have found this https://github.com/pimoroni/veml6075-python

nikkisstuff, 2024-01-08:
I found this https://www.waveshare.com/wiki/Template:AS7341_Spectral_Color_Sensor-RaspberryPi-Guides

nikkisstuff, 2024-01-08:
I've found this https://ams.com/o/download-server/document-download/download/15639123

theyosh, 2024-01-08:
> I've found this https://ams.com/o/download-server/document-download/download/15639123

Found this hardware device :( https://www.futureelectronics.com/fr/p/development-tools--dev-tools-sensors/as7331-evk-ams-osram-5169244 and https://uk.farnell.com/ams-osram-group/as7331-evk/evaluation-kit-spectral-uv-sensor/dp/4134313 which is way more expensive...

This is the only device I can find. The python code can be used, but only with that expensive development board... (I guess)

nikkisstuff, 2024-01-08:
I think its probably worth waiting to see if the veml6075 ever comes back into production or a updated version is released. For now i will continue to use my uv check cards

ThunderClap86, 2024-01-10:
> Hmm, sorry to say, but I can't find python code for that sensor. Only Arduino :( So than you need to to use an Arduino and hook that to the Pi. It can be done, but that will cost some programming. [...]

[Is this the python you'd need?](https://github.com/kriswiner/AS7331)

theyosh, 2024-01-10:
> > Hmm, sorry to say, but I can't find python code for that sensor. Only Arduino :( So than you need to to use an Arduino and hook that to the Pi. It can be done, but that will cost some programming. And I can't help with that part, as I do not use Arduino boards.
> > Sorry to say, but I think there is only one solution... pay for expensive sensors :( ...
> 
> [Is this the code you'd need?](https://github.com/kriswiner/AS7331)

Unfortunately, that is not python code. That is Arduino code. So that can't be used directly.

closed 2024-01-11 by nikkisstuff

ThunderClap86, 2024-01-12:
> Unfortunately, that is not python code. That is Arduino code. So that can't be used directly.

Is there a way we can chat off of this thread? I ran the C++ from that github through a "converter" and it spit out what looks a lot more like Python. I have no way of knowing if it is right or not and don't want to clog up this sensor subject with what could be nothing.

Griizly, 2024-01-16:
Hi! I'm not a terrariumPi user but I've been working a bit with the VEML6075 trying to make my own SolarMeter to check the UVI of my arcadia ramps.

I would like to warn you **not** to use it in terrarium :) measurements are not reliable
Why not? 
because the VEML6075 is a **solar** UV sensor and doesn't work with our UVB lamps or ramps because most of them don't emit IR (infrared) unlike the sun which emits IR and the VEML6075 needs visible AND infrared light to be able to calculate UVI.

In this issue https://github.com/theyosh/TerrariumPI/issues/210#issuecomment-904892482 you can see that someone has the sensor and in the dashboard he only gets a UVI of 0.178 which is way too low, value is not accurate...

I tested this sensor and by placing it 30cm away from my brand new arcadia t5 12% desert 120cm I only get 1 UVI with the sensor whereas the SolarMeter 6.5 gives me just over 8 UVI, which is a good value for this lamp.

So avoid using the VEML6075, the UVI values are unreliable.

theyosh, 2024-01-18:
@Griizly that explains a lot. I have the same 'problem' with my UV values. They are really low, while the sensor is pretty close to a Solar Raptor UV light. During development I test it with sunlight behind a window, and that works. Now I know why.

But than also, the SolarMeter you mention, is way more expensive. Which makes clear that the rather cheap VEML6075 sensors can never be that good or accurate.

So, you could use them when you want to measure sunlight uv outside. But for inside, they are no good.

I will update the documentation about this. Thanks for the information.

JoergLindner, 2025-05-09:
Hi,
first of all the Solarmeter 6.5R does not measure UVI Erythem ISO 17166, which cover not only the UVB area, as the Solarmeter 6.5R does. Solarmeter 6.5R try to follow with the sensor the D3 Synthese Curve, which they basically not really do below 300nm. The D3 synthes curve whould drop much more, compare to the Sensor response curve. This means the UVI might be shown higher as it is. In the other hand on real world ther is mor or less non UV below 300nm, as it is blocked by the Ozon layer, which of course than are not calculated in embedded Sesnors, which use may be UVA and VIS for UVI Index.  
UVI based on ISO 17166  (Erythema Action Curve ) will go to 400nm, means cover full UVA/B Spectrum + a bit in the VIS area.  Common cheap Sensors calculates always the UVI based on Erythema, which is somehow complicated as they use the visible Data and use a relation between Visible measured Data and UVA/B. This you will not find in any Datasheet. 
Besides that the 6.5R use a SIC Sensor and all others a GaN Sensors. OK GaN Sensor works too, but they are more temperature sensitiv and have a higher Temperature Coefficient. Any to get a similar reading as a Solarmeter 6.5R the reponse curve of the Sensor must be similiar, which means you need a optical filter to do this + a diffusor would help too.
Than of course, you still just measure a pseudo UVI, as this is basically a 'reptile' measure area without any norm. 
Just to the cost. If you like to by one single SIC Sensor with nearly the same response curve as the 6.5R unit it will cost at least 180EUR (if you find it) just for the sensor. If you buy 1000pcs, it will drop to about 50EUR and we talking here about an analog output.
So really nothing which make sense, if you are not in the Sensor business, 

Last but not least: Even if there (and it will be soon) a unit available with remote Sensor, which could be installed in a terrarium, don't do it. UVB have much mor energy as UVA or VIS. This accelerates the aging of the sensor, and a replacement is not cheap, as you can imagine.

Regards
Jörg Lindner