theyosh/TerrariumPI #878: Unable to delete unused displays

nikkisstuff, 2024-01-11:
I have been trying a few of my oled/lcd  displays and the delete function is not working. I am getting this error:

The service 'oled' could not be deleted!
Error: Error deleting notification service with id Notification service display oled. 'NoneType' object has no attribute 'stop'

nikkisstuff, 2024-01-11:
I fixed it buy disabling the display then rebooting and then it would allow me to delete it

closed 2024-01-11 by nikkisstuff

manolof, 2024-03-13:
@theyosh I have the same issue:
[image]

theyosh, 2024-03-13:
Updated the code. With a `git pull` and a restart should this be fixed