theyosh/TerrariumPI #880: [BUG?]: Button could not be added: No access to /dev/mem. Try running as root!

Swiftnesses, 2024-01-13:
Hello :)

Latest version (clean install) - I'm getting

`Button could not be added: No access to /dev/mem. Try running as root!` when trying to add a magnetic door sensor.

All other aspects work fine, inc webcam, relays.

I've checked permissions and all seems OK.

theyosh, 2024-01-19:
Did you rebooted the Pi after clean install? It need at least one reboot after a clean install.

Swiftnesses, 2024-01-19:
I sure did!

theyosh, 2024-01-20:
What did you enter as GPIO address? I can just add a button without any problems

theyosh, 2024-01-22:
Keep in mind the address should be the physical number on the Raspberry PI board.

Swiftnesses, 2024-01-27:
This is just a new install @theyosh, I'm using GPIO14 for the magnetic sensor (I enter 14 like before).

theyosh, 2024-01-27:
entering [14](https://pinout.xyz/pinout/ground) is NOT possible. That is a ground pin. So if you have connected on GPIO14 according to https://pinout.xyz/ you have to enter [8](https://pinout.xyz/pinout/pin8_gpio14/) as address

Swiftnesses, 2024-01-27:
Let me check, I simply pulled the values from my old db, so everything was copied across.

Swiftnesses, 2024-01-27:
Feels like user error, I'm just baffled how this worked before! Or at least that value was in the old db (working).

Swiftnesses, 2024-01-27:
Sorry, it's on GPIO24, so I entered 18, not 14 :-/

It doesn't accept that.

Swiftnesses, 2024-01-27:
I fixed it with:

```
sudo chown root.gpio /dev/gpiomem
sudo chmod g+rw /dev/gpiomem
```

theyosh, 2024-01-27:
Are you on a Pi zero?

Swiftnesses, 2024-01-28:
Pi 3 +

theyosh, 2024-01-28:
Ok, but it is al working now? Still strange that you had to change some rights..

But this issue can now be closed?

closed 2024-01-29 by Swiftnesses