theyosh/TerrariumPI #882: [FEATURE]: Save in DB Telegram started user

cmonicob, 2024-01-16:
Save start Telegram users in database

When we restart Terrarium PI we lose all the started users, it would be interesting to be able to save somehow the IDs of the started chats to be able to restore the notifications, either by a manual restart or by system failure, and to know that it has had some error.

theyosh, 2024-01-16:
Hmm, that is a nice feature. And I think it can be done. But we need to update the database so that we can store a state per notification service.

During startup, we can read that out, and use the last known chat ids to start communicating again. And there is a broadcast function, which we can use to send out a broadcast message about the reconnect of TP.

Let me think a bit more about this.

theyosh, 2024-01-20:
So, I just updated the code which should make it possible to reconnect to existing chats after a restart.

Do a `git pull` and restart TP. You have to start a new chat in Telegram with the `/start` command. With that command your chat id is stored in the database. When you then restart TP, you should see a message in Telegram when it is reconnected.

I hope this works, as I cannot test it. Can you test this?

cmonicob, 2024-01-21:
Error in load state

```
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 19, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 116, in __init__
    self.notification.load_services()
  File "/home/pi/TerrariumPI/terrariumNotification.py", line 263, in load_services
    setup['state'] = copy.deepcopy(service["state"])
KeyError: 'state'
```

theyosh, 2024-01-21:
Ok, do a `git pull` and restart. I hope it should start again.

theyosh, 2024-01-22:
Did the last commit fixed the startup issue?

cmonicob, 2024-01-23:
Now the system starts correctly, but the notifications are failing.
I have been looking for where the error came from and it seems that at the moment of the /start of the conversation and saving the new chat id in status it encrypts again both the token and the allowed list.
This means that when the system recovers after a reboot, the token is no longer correct.

also an error occurs in the gui when saving the Telegram service due to status

theyosh, 2024-01-23:
> Now the system starts correctly, but the notifications are failing.

No error messages when running in debug mode?

> I have been looking for where the error came from and it seems that at the moment of the /start of the conversation and saving the new chat id in status it encrypts again both the token and the allowed list.
This means that when the system recovers after a reboot, the token is no longer correct.

They are then double encrypted, and encrypt over and over again... yeah, that is not good. That needs some more debugging

> also an error occurs in the gui when saving the Telegram service due to status

I will look into this tonight.

theyosh, 2024-01-23:
So I made a rather big change with the encryption, so I hope all is working for you.

I expect that you have to update the Telegram service in the gui. That should now also be fixed. Saving will store the correct data in the database.

Now, I do not know what will happen, when Telegram is working again. I think you will have to start a new chat.
And as far I know, a Telegram bot can never start a chat. So when it reconnects, Telegram sees this, and even when we load the old chat ids, I guess they will not allow that.
Else you have to search on how to reconnect a bot to existing chats... I cannot test it..

And I get the warnings: 
```
PTBUserWarning: If 'per_message=False', 'CallbackQueryHandler' will not be tracked for every message. Read this FAQ entry to learn more about the per_* settings: https://github.com/python-telegram-bot/python-telegram-bot/wiki/Frequently-Asked-Questions#what-do-the-per_-settings-in-conversationhandler-do.
  fallbacks=[CommandHandler("webcam", self.webcamSelect)],
```
Not sure what to do with it... do you have a clue?

So with a `git pull` and a restart, I hope it is all working again, and you can edit the Telegram settings. Which will make the Telegram bot work again in a new chat... If that works, we are at least at stable base again.

cmonicob, 2024-01-25:
The new implementation is correct, now it does not fail to start the conversation.

the ConversationHandler per_message=False is set to False, to ensure that in future updates of the library it does not change it can be added in the ConversationHandlers
````
ConversationHandler(
    entry_points=[CommandHandler("webcam", self.webcamSelect)],
    states={0: [CallbackQueryHandler(self.webcam)]},
    fallbacks=[CommandHandler("webcam", self.webcamSelect)],
    per_message=False,
)
````

To stop the warning from appearing, you could add the following import:
````
from warnings import filterwarnings
from telegram.warnings import PTBUserWarning
````

and in the Telegram load_setup the following function:
````
filterwarnings(action="ignore", message=r".*CallbackQueryHandler", category=PTBUserWarning)
````

theyosh, 2024-01-25:
But do we want the option `per_message=False` ? I can't judge very well the implications in a chat when this is added.

cmonicob, 2024-01-28:
I have tried to set it to true and the answer is that there is no conversation with the bot, I will review the issue a little more to see if it is convenient to change the programming to set it to True mode, but from what I understand if you do not have to be in a Telegram group it is not necessary.

In my case, I only have private communication with the bot established.

theyosh, 2024-02-02:
The goal is only private communication. As we have a list of allowed users, I do not see it working in a open/free chat with other users. They can than only listen... Could be nice. But the communications part is made with the idea that it is always private.

theyosh, 2024-02-12:
Any new updates about the `per_message=False,` setting?

cmonicob, 2024-02-22:
For now, it is to keep it false or not explicitly set it false.
When I have a moment, I want to rebuild the bot so that it can be True and have much more control of the chats.

I have detected that when restarting the system it sends the image of the terrarium pi avatar, could it be removed?

theyosh, 2024-02-27:
> I have detected that when restarting the system it sends the image of the terrarium pi avatar, could it be removed?

I thought this would be a nice feature ;) It is a broadcast message, so you know that the Pi has restarted. I can imagine it is a bit annoying when you are testing. But in normal usage, you restart/reboot the Pi only once a month? I mostly 3 times a year.

So I thought this is a nice feature. If you changed the image, you should get that image. And it seems to work, as you get it on your device :)

So I would like to keep the broadcast message during startup.

theyosh, 2024-03-08:
But can we close this issue? As the saving does work, and restarting is also working. When you have code updates, just create a PR and I will at it.

theyosh, 2024-03-14:
Closing for now. As the saving and restarting conversation is working

closed 2024-03-14 by theyosh