theyosh/TerrariumPI #883: notification Message to display sensors on 12c lcd ?

nikkisstuff, 2024-01-18:
I am struggling to configure notification Message to display sensors on 12c lcd. How do i do it?

Thanks Nikki

nikkisstuff, 2024-01-19:
Can anyone show me a example so i can learn how to do it? I am trying to show the current temperature's

nikkisstuff, 2024-01-20:
I have sort off got it working how Can I split my 3 sensors I am using. Do I just have to duplicate it twice more?

${name:}
${value:.2f}C

theyosh, 2024-01-20:
Hi, I think you example message is ok. And you only have to do this once. And then depending on the chosen notification trigger this text will appear on the display. But you cannot combine multiple sensors in 1 screen. So every sensor will output the same text, after each other.

But I want to say that the notification code and logic is not that great. Also the supported displays are pretty small, so not much text to put onto. So most people make their on code to readout TP and make their own message for the display.

nikkisstuff, 2024-01-20:
Thanks. I have been testing a few different messages and like you said the screen is not much use at the moment. I think I will use red amber and green leds as a system status indicator. 

Thanks Nikki

closed 2024-01-22 by theyosh