theyosh/TerrariumPI #884: Refused Connection

Gwigga, 2024-01-22:
I have installed, re-installed, and read almost everything, but still cannot find a definitive answer to the refused connection issue i am facing.
I have follow the instructions to a t. I can SSH to the Raspberry pi but cannot get the Dash to load

[image]

theyosh, 2024-01-22:
Could you try to debug: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/ and send the outcome of the startup?

Gwigga, 2024-01-22:
Traceback (most recent call last):
  File "/home/robin/TerrariumPI/terrariumPI.py", line 16, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/robin/TerrariumPI/terrariumEngine.py", line 2, in <module>
    from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
  File "/home/robin/TerrariumPI/hardware/webcam/__init__.py", line 2, in <module>
    import terrariumLogging
  File "/home/robin/TerrariumPI/terrariumLogging.py", line 92, in <module>
    logging.config.fileConfig("logging.cfg")
  File "/usr/lib/python3.9/logging/config.py", line 79, in fileConfig
    handlers = _install_handlers(cp, formatters)
  File "/usr/lib/python3.9/logging/config.py", line 145, in _install_handlers
    h = klass(*args, **kwargs)
  File "/usr/lib/python3.9/logging/handlers.py", line 208, in __init__
    BaseRotatingHandler.__init__(self, filename, 'a', encoding=encoding,
  File "/usr/lib/python3.9/logging/handlers.py", line 58, in __init__
    logging.FileHandler.__init__(self, filename, mode=mode,
  File "/usr/lib/python3.9/logging/__init__.py", line 1142, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.9/logging/__init__.py", line 1171, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding,
PermissionError: [Errno 13] Permission denied: '/home/robin/TerrariumPI/log/terrariumpi.log'

theyosh, 2024-01-22:
Ok, I think the following command should fix it: `sudo chown pi.pi /dev/shm/terrariumpi.*`

After that it should run. You can try it in debug first, and when it continues, just pres `Ctrl+C` to stop it and start the service normal again.

Gwigga, 2024-01-22:
sudo chown pi.pi /dev/shm/terrariumpi.*
chown: invalid user: ‘pi.pi’

theyosh, 2024-01-22:
sudo chown pi /dev/shm/terrariumpi.*

does work? If so, that should be enough also

Gwigga, 2024-01-22:
Traceback (most recent call last):
  File "/home/robin/TerrariumPI/terrariumPI.py", line 16, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/robin/TerrariumPI/terrariumEngine.py", line 2, in <module>
    from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
  File "/home/robin/TerrariumPI/hardware/webcam/__init__.py", line 2, in <module>
    import terrariumLogging
  File "/home/robin/TerrariumPI/terrariumLogging.py", line 92, in <module>
    logging.config.fileConfig("logging.cfg")
  File "/usr/lib/python3.9/logging/config.py", line 79, in fileConfig
    handlers = _install_handlers(cp, formatters)
  File "/usr/lib/python3.9/logging/config.py", line 145, in _install_handlers
    h = klass(*args, **kwargs)
  File "/usr/lib/python3.9/logging/handlers.py", line 208, in __init__
    BaseRotatingHandler.__init__(self, filename, 'a', encoding=encoding,
  File "/usr/lib/python3.9/logging/handlers.py", line 58, in __init__
    logging.FileHandler.__init__(self, filename, mode=mode,
  File "/usr/lib/python3.9/logging/__init__.py", line 1142, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.9/logging/__init__.py", line 1171, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding,
PermissionError: [Errno 13] Permission denied: '/home/robin/TerrariumPI/log/terrariumpi.debug.log'
(venv) robin@raspberrypi:~/TerrariumPI $

theyosh, 2024-01-22:
sudo chown pi /home/robin/TerrariumPI/log/terrariumpi.debug.log

I guess you have run it as root somehow...

Gwigga, 2024-01-22:
Traceback (most recent call last):
  File "/home/robin/TerrariumPI/terrariumPI.py", line 19, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/robin/TerrariumPI/terrariumEngine.py", line 98, in __init__
    init_db(self.version)
  File "/home/robin/TerrariumPI/terrariumDatabase.py", line 41, in init
    with backend.lock():
  File "/usr/lib/python3.9/contextlib.py", line 117, in __enter__
    return next(self.gen)
  File "/home/robin/TerrariumPI/venv/lib/python3.9/site-packages/yoyo/backends/base.py", line 324, in lock
    self._insert_lock_row(pid, timeout)
  File "/home/robin/TerrariumPI/venv/lib/python3.9/site-packages/yoyo/backends/base.py", line 355, in _insert_lock_row
    raise exceptions.LockTimeout(
yoyo.exceptions.LockTimeout: Database locked (run yoyo break-lock to remove this lock)

Gwigga, 2024-01-22:
sure no problem

theyosh, 2024-01-22:
again, somehow the database rights are incorrect... 

`sudo rm /home/pi/TerrariumPI/data/terrariumpi.*`
`sudo chown pi /home/pi/TerrariumPI/data/`

and try again....

Gwigga, 2024-01-22:
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 16, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 2, in <module>
    from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
  File "/home/pi/TerrariumPI/hardware/webcam/__init__.py", line 2, in <module>
    import terrariumLogging
  File "/home/pi/TerrariumPI/terrariumLogging.py", line 92, in <module>
    logging.config.fileConfig("logging.cfg")
  File "/usr/lib/python3.9/logging/config.py", line 79, in fileConfig
    handlers = _install_handlers(cp, formatters)
  File "/usr/lib/python3.9/logging/config.py", line 145, in _install_handlers
    h = klass(*args, **kwargs)
  File "/usr/lib/python3.9/logging/handlers.py", line 208, in __init__
    BaseRotatingHandler.__init__(self, filename, 'a', encoding=encoding,
  File "/usr/lib/python3.9/logging/handlers.py", line 58, in __init__
    logging.FileHandler.__init__(self, filename, mode=mode,
  File "/usr/lib/python3.9/logging/__init__.py", line 1142, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.9/logging/__init__.py", line 1171, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding,
PermissionError: [Errno 13] Permission denied: '/home/pi/TerrariumPI/log/terrariumpi.log'
(venv) pi@raspberrypi:~/TerrariumPI $

theyosh, 2024-01-22:
sudo chown pi /dev/shm/terrariumpi.*

Gwigga, 2024-01-22:
venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2024-01-22 20:32:23,188 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.8.2 on a Raspberry Pi Zero 2 W Rev 1.0 ...
2024-01-22 20:32:23,270 - INFO    - terrariumEngine       - Loaded 30 settings in 0.08 seconds.
fatal: detected dubious ownership in repository at '/home/pi/TerrariumPI'
To add an exception for this directory, call:

        git config --global --add safe.directory /home/pi/TerrariumPI
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 19, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 111, in __init__
    self.load_settings()
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 343, in load_settings
    gitversion = str(terrariumUtils.get_script_data("git rev-parse HEAD").decode("utf-8")).strip()
  File "/home/pi/TerrariumPI/terrariumUtils.py", line 441, in get_script_data
    raise ex
  File "/home/pi/TerrariumPI/terrariumUtils.py", line 433, in get_script_data
    data = subprocess.check_output(
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/subprocess.py", line 404, in check_output
    raise CalledProcessError(retcode, process.args, output=output)
subprocess.CalledProcessError: Command 'git rev-parse HEAD' returned non-zero exit status 128.
^CKeyboardInterrupt
2024-01-22T20:32:30Z
Exception ignored in: <module 'threading' from '/usr/lib/python3.9/threading.py'>
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/monkey.py", line 911, in _shutdown
    orig_shutdown()
  File "/usr/lib/python3.9/threading.py", line 1428, in _shutdown
    lock.acquire()
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/thread.py", line 121, in acquire
    acquired = BoundedSemaphore.acquire(self, blocking, timeout)
  File "src/gevent/_semaphore.py", line 180, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_semaphore.py", line 249, in gevent._gevent_c_semaphore.Semaphore.acquire
  File "src/gevent/_abstract_linkable.py", line 521, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait
  File "src/gevent/_abstract_linkable.py", line 487, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 490, in gevent._gevent_c_abstract_linkable.AbstractLinkable._wait_core
  File "src/gevent/_abstract_linkable.py", line 442, in gevent._gevent_c_abstract_linkable.AbstractLinkable._AbstractLinkable__wait_to_be_notified
  File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
KeyboardInterrupt:
(venv) pi@raspberrypi:~/TerrariumPI $

Gwigga, 2024-01-22:
looks like im going

theyosh, 2024-01-22:
Have you followed: https://theyosh.github.io/TerrariumPI/install/ ? I get the feeling you have not installed the Raspberry Lite LEGACY OS version. I needs to a LITE image at least, and the LEGACY is known to work.

You have a newer image, as you have Python 3.9, where I work with 3.7 version.

Again, it looks like the rights are not correct. Again.
>     git config --global --add safe.directory /home/pi/TerrariumPI
This should not happen when everything is installed as user Pi. Somehow, you have again mixed up the rights... please pay attention to the installation steps... https://theyosh.github.io/TerrariumPI/install/#manual

Gwigga, 2024-01-22:
2023-12-05-raspios-bullseye-armhf-lite.img is the image I have used
[image]
and followed the instructions, literally copying and pasting the commands in :( im not completely inept

theyosh, 2024-01-22:
Hmm, seems like the legacy image is upgraded.... 

That means I have to recheck the installer... because I have not seen those problems with a new installation. But that takes more time, and hope some more information in a few days.

Sorry... did not expect this strange issues...

Gwigga, 2024-01-22:
I can try an earlier version, whihc of the below do you think would work
[image]

theyosh, 2024-01-22:
I guess this one should work: https://downloads.raspberrypi.com/raspios_oldstable_armhf/images/raspios_oldstable_armhf-2023-05-03/  The .xz file is the image

RobinM-MB, 2024-01-23:
I will try this one tonight and confirm what happens

Gwigga, 2024-01-23:
> I guess this one should work:
> 
> Update: this is the correct image: https://downloads.raspberrypi.com/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2023-05-03/
> 
> The 2023-05-03 folder

This one wouldn't even install git

and the 2023-10-10 folder just gives the same as the latest

RobinM-MB, 2024-01-24:
Just an update, I have tried this on a zerow2 and a B+ all with the same out come, either git install doesn't work as it can't get the repos for the old versions, or if i do get TerrariumPI installed i get the same issues as before

theyosh, 2024-01-24:
Hmm, apparently there was a startup issue. So that has been fixed. 

I tested with this image: https://downloads.raspberrypi.com/raspios_oldstable_lite_armhf/images/raspios_oldstable_lite_armhf-2023-05-03/2023-05-03-raspios-buster-armhf-lite.img.xz You can chose an image from disk with RPI imager tool https://www.raspberrypi.com/software/

Also I have changed the manual installation documentation at bit https://theyosh.github.io/TerrariumPI/install/#manual It involves how to install the git command.

I have tested with a custom username, so user `pi` is not needed.

And it is working. Only bluetooth scanning is somehow not working right now. But that could be a hardware issue with my Pi.

So download the old lite image, and follow the manual installation. It works over here now.

theyosh, 2024-01-24:
Installation works also on the bullseye (current legacy image) on a Raspberry Pi 3.

RobinM-MB, 2024-01-24:
Clearly i am inept as, its just giving me the same issue :( Copied and pasted all of the commands from the install page (updated ones)

venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
Traceback (most recent call last):
  File "terrariumPI.py", line 16, in <module>
    from terrariumEngine import terrariumEngine
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 2, in <module>
    from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
  File "/home/pi/TerrariumPI/hardware/webcam/__init__.py", line 2, in <module>
    import terrariumLogging
  File "/home/pi/TerrariumPI/terrariumLogging.py", line 92, in <module>
    logging.config.fileConfig("logging.cfg")
  File "/usr/lib/python3.7/logging/config.py", line 79, in fileConfig
    handlers = _install_handlers(cp, formatters)
  File "/usr/lib/python3.7/logging/config.py", line 145, in _install_handlers
    h = klass(*args, **kwargs)
  File "/usr/lib/python3.7/logging/handlers.py", line 199, in __init__
    BaseRotatingHandler.__init__(self, filename, 'a', encoding, delay)
  File "/usr/lib/python3.7/logging/handlers.py", line 54, in __init__
    logging.FileHandler.__init__(self, filename, mode, encoding, delay)
  File "/usr/lib/python3.7/logging/__init__.py", line 1092, in __init__
    StreamHandler.__init__(self, self._open())
  File "/usr/lib/python3.7/logging/__init__.py", line 1121, in _open
    return open(self.baseFilename, self.mode, encoding=self.encoding)
PermissionError: [Errno 13] Permission denied: '/home/pi/TerrariumPI/log/terrariumpi.log'

theyosh, 2024-01-24:
I guess still you have run the `git clone ... ` command as root user. That makes everything owning by root. And that give errors by running as normal user. So the installation has to be done over.

Remove the existing TerrariumPI folder. `rm -Rf /home/pi/TerrariumPI`
Now make sure that you are the user `pi` and not root when running the command `git clone --branch main --depth 1 https://github.com/theyosh/TerrariumPI.git`

And follow the installation as documented. If you still get the above error, just retry ones with a clean image. 

I have done this now twice, and it works. Else I am sorry for now, but I can't fix things I cannot reproduce. Also, this testing costs about an hour for each image. Which is a lot of time...

theyosh, 2024-01-24:
Also, Pi Zero is at own risk. I do not actively support those.

Gwigga, 2024-01-24:
Ok no problem

RobinM-MB, 2024-01-24:
> Also, Pi Zero is at own risk. I do not actively support those.

which ones do you support?

theyosh, 2024-01-24:
> > Also, Pi Zero is at own risk. I do not actively support those.
> 
> which ones do you support?

https://github.com/theyosh/TerrariumPI#hardware

RobinM-MB, 2024-01-24:
Re did it all as root 
[image]

RobinM-MB, 2024-01-24:
Just can't login now :D using admin/password lol

theyosh, 2024-01-24:
Ok, for a Pi Zero that is correct.. so do not change it!! :) 

Could you run 1 command for me at the console and paste the outcome here:

`cat /proc/cpuinfo`

I use that information for detecting a Pi Zero. And I guess yours is a bit different. With that info, I can fix the installer for a Pi Zero2 I hope

theyosh, 2024-01-24:
> Just can't login now :D using admin/password lol

I guess some encryption issue... but if you can first past the output of /proc/cpuinfo I can adjust the installer

RobinM-MB, 2024-01-24:
processor       : 0
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
model name      : ARMv7 Processor rev 4 (v7l)
BogoMIPS        : 38.40
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

Hardware        : BCM2835
Revision        : 902120
Serial          : 00000000488b94f9
Model           : Raspberry Pi Zero 2 W Rev 1.0

RobinM-MB, 2024-01-24:
I also don't think it runs unless i logically run TerrariumPI.py as soon as i close the terminal i lost the dashboard

theyosh, 2024-01-24:
Ok, the installer did correctly detect your Pi Zero. So that is ok

theyosh, 2024-01-24:
If you run it in debug, you cannot close the terminal. But it should run as a service...  but  again, I know the Zero has issues. And therefore with a Pi Zero you should have some technical knowledge. 

Also search in the closed issues for clues. https://github.com/theyosh/TerrariumPI/issues/823#issuecomment-1693236850

But from my side, without any clear errors, I cannot do any thing.

RobinM-MB, 2024-01-24:
So I don't think the service is actually starting when the i reboots,
I cannot run TerrariumPI.py unless i am within venv
`pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
Traceback (most recent call last):
  File "terrariumPI.py", line 4, in <module>
    from gevent import monkey
ImportError: No module named gevent`

`(venv) root@raspberrypi:/home/pi/TerrariumPI# python terrariumPI.py
2024-01-24 22:38:36,703 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.8.2 on a Raspberry Pi Zero 2 W Rev`

RobinM-MB, 2024-01-24:
> If you run it in debug, you cannot close the terminal. But it should run as a service... but again, I know the Zero has issues. And therefore with a Pi Zero you should have some technical knowledge.
> 
> Also search in the closed issues for clues. [#823 (comment)](https://github.com/theyosh/TerrariumPI/issues/823#issuecomment-1693236850)
> 
> But from my side, without any clear errors, I cannot do any thing.

Cool going to Order a Pi4 anyways

Gwigga, 2024-01-24:
Restarted it now and is running disconnected from terminal yay thankyou
Kind regards

Robin Morgan

Gwigga, 2024-01-25:
One last question

How easy is it to add another type of smart plug.

As we are Uk many of the ones configured arnt readily available.

But we do get Tapo P100 which are readily available and have a very easy api

Kind regards

Robin Morgan

Gwigga, 2024-01-25:
Don’t worry I will by some of the Meross ones from Amazon

Kind regards

Robin Morgan

theyosh, 2024-01-25:
I think the Tapo P100 can be supported. You need to know (I guess) you email and password at TP-Link and the IP of the Tapo P100 plug. So I guess only the IP is something you have to find of your plug. If you know that we can try to add this device.

I will need some testing from your side. So if you can test this for me with your setup at home, please open a new issue for this and I will try to add some support to it.

RobinM-MB, 2024-01-25:
Perfect I will open a new issue

Kind Regards
Robin Morgan | Digital Consultant | RPA Developer
24 Hill Street, St Helier, JE2 4UA
01534 761 464
***@***.*** | www.masonbreese.com
Image

theyosh, 2024-01-25:
Al ready created an issue and support for TAP0 P 100. https://github.com/theyosh/TerrariumPI/issues/888

theyosh, 2024-01-25:
Closing this issue as it has been resolved.

closed 2024-01-25 by theyosh