theyosh/TerrariumPI #889: [FEATURE]: Add a way to offset weather by months

manolof, 2024-02-09:
The weather feature is great, as I would like to simulate my Gecko's natural habitat sunrise and sunset hours. However, I realize that when it's summer in my Gecko's natural habitat (e.g. New Caledonia), it is winter over here in central Europe, and vice-versa.

Question 1:
Is there a way to offset the weather time by, let's say, months?
I understand that with the real-time weather API this would probably be impossible, but could we use another API, like historical weather, to achieve that?

Question 2:
Or plan B if someone wishes to not use the weather feature but implement their own yearly cycle, wouldn't it be nice to have a feature where we specify the Day duration for each month? Or something like that.
So for January, would be start 8:00 end 18:00, and so on..

Thanks for the great software by the way! I just bought you a coffee ;)

theyosh, 2024-02-12:
He @manolof thanks for the ☕.

Answer 1: Set your weather setup to New Caledonia. And create a lights area based on weather data. This will mimic the sun rise and set times based on New Caledonia. This means more daylight in your winter period, and less lights in the summer.
This will not make the terrarium turn the lights on during your night. If the sun rise at 06:30 at New Caledonia, the lights go on at 06:30 at your home.

But, my experience with this that it does not work for me. As the terrarium is in my living room, it will stay light during the summer. The lights are out, but the temperature is around 25 still and the sun is still shining outside.

Answer2: See 1. Use weather data as source.

manolof, 2024-02-12:
> This will not make the terrarium turn the lights on during your night. If the sun rise at 06:30 at New Caledonia, the lights go on at 06:30 at your home.

Thanks for clarifying - I was aware of that.
And yes, the time difference is not that big, but even then, I can adjust it in my area settings page - no problem.

I was mostly wondering if I can have the New Caledonian summer coincide with my local summer somehow.. Otherwise it makes handling the whole setup a bit difficult:
In the summer, indoor temperatures rise to about 28 degrees, and in the winter they drop to around 22. This is the complete opposite of the New Caledonian temperatures.

[image]

that means, that I would probably have to intervene throughout the year; to cool things down during my summer, and spend more electricity on trying to heat it up during my winter.

theyosh, 2024-02-12:
yeah, for heating it could turn out that you have to cool your house/room in the summer to get the winter temperature from the other side of the world.
I have looked into this, but then I need historical data, which is about 950 euro a year :(  https://openweathermap.org/price

So I just used my local weather for sun rise and set, with a min hours of 10 and max hours of 14 or 16 a day. So I do have some kind of seasons. Mine gecko is from Madagascar.

For temperature, I keep it al year the same. As the days become shorter, the temperature will also drop, so that is good enough for me. I only have 1 heating spot, and that is enough. And that is at least 25 C during the cold period.

theyosh, 2024-02-15:
But I think I cannot do much about this. As tor this we need some kind of historical data.. As the realtime data is always of now. Does not matter where on the planet. So that means I have to record for 1 year, and use that for shifting months.

And also, as you already mentioned, you have to cool your hose to get the colder winter temperatures in the summer.

So I would just go for a fixed average temperature for the hole year. And play with the amount of light hours. I use a minimum of 10 and max of 14/16 hours. And let it follow the local sun rise and set. It will give you different season effect, as it would in real life.

Therefore, I would like to close this issue. It is a nice feature to have, but not to do with the current free external data that I can use.

theyosh, 2024-02-15:
Sorry for this... I still appreciate that you bought some coffee. Thanks again.

manolof, 2024-02-16:
Yeah, my first thought was to use a fixed schedule and update it every month, but then you said something better; I will be using the local weather and will adjust the min/max durations to fit the Gecko's natural habitat.
Problem solved 😄 

Thanks again for your time.
The coffee wasn't for this topic, but for the great work already done.

closed 2024-02-16 by manolof