theyosh/TerrariumPI #890: [BUG]: Unprogrammed relays are activating randomly.

ThunderClap86, 2024-02-16:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.8.1 (and 4.8.0)
 - Install: [Manual]
 - Raspberry PI: 3B+ and 4
 - Qwiic Quad Relay module (i2C)

**Describe the bug**
During continued operation of the system, the program triggers relays as intended. I have Relay 1 (R1) configured to weather for main lights, triggers on/off normally. R2 is heater pad via sensors, on/off as-needed. R3 is a cooling ventilation fan, comes on and stays on until manually turned off (Not sure what setting to change). R4, unassigned becomes triggered randomly and unsure as to why. 

**Expected behavior**
I would expect R3 to turn off when the temperature cools below the sensor range specified. I would expect R4 to never activate because it’s unassigned. 

**Screenshots**
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

**Additional context**

theyosh, 2024-02-16:
Are you also able to upload the logfile in the folder `log`? As far as I know should al sensitive date being stripped from the logfile. So you could upload it here.
Else you can also mail it to github@theyosh.nl Then I can look in the logs to see if I can find something.

The code for this relay is dead simple, so there is not much I can change to it.

Do you have other I2C hardware in use?

theyosh, 2024-02-27:
I have not received a log file for further debugging. Else it will be hard to say what is causing this.. a log file could help a lot.

ThunderClap86, 2024-02-29:
I apologize for the lengthy delay, I am away from my Pi at the moment and do not have it on the web to pull the log. I will be back physically at the Pi again this weekend to pull the log and upload here.

ThunderClap86, 2024-04-05:
> Any updates?

Having trouble with this... can't get in to pull the logs. https://github.com/theyosh/TerrariumPI/assets/105729168/153bc640-1c62-4448-8d09-57970d1fcd7e

theyosh, 2024-04-05:
Strange. Does this also happen on a desktop browser? I guess the image was of a Android with Chrome?
And did you enter the credentials, or does the error occur before you can enter the credentials.

You can also download the logfile through ssh. It is located at `/home/pi/TerrariumPI/log/terrariumpi.log` WinSCP is a tool that can do it. Or FileZilla

theyosh, 2024-05-02:
Is this still happening? I do not have other people with login errors. Without a log file I can't do anything.

Is would like to close this issue if there are no updates anymore.

ThunderClap86, 2024-05-08:
Yes, there are issues with the login still. The relays are also still doing the activation despite being unprogrammed. My next step is to reinstall everything and start over from scratch. Perhaps this will solve it.

theyosh, 2024-05-08:
Have a look at https://theyosh.github.io/TerrariumPI/faq/which-files-to-backup/ and https://theyosh.github.io/TerrariumPI/faq/restoring-backup-files/ for backup and restoring your database and calendar.

ThunderClap86, 2024-05-08:
> Any updates?

Do I really want to back it all up and "import" the relays acting the way they are if it's somehow a config setting I messed up? Genuine question.

theyosh, 2024-05-08:
Good point. Than make at least a backup. So you can use a SQlite editor to read out the old values from the backup database when you reconfigure your setup.

closed 2024-05-11 by ThunderClap86