theyosh/TerrariumPI #892: [FEATURE]: Sort Order of the enclosures

ewiandr73, 2024-02-18:
**Is your feature request related to a problem? Please describe.**
The sort order of the of the enclosures when using numbers

current
1,10,11,2,3,4,5,6,7,8,9

**Describe the solution you'd like**
What would be nice.
1,2,3,4,5,6,7,8,9,10,11

**Additional context**
Alphabet sort works fine.
I currently have 11 enclosures running, with 6 more still to add. :)
36 relays and 19 sensors.

theyosh, 2024-02-27:
Hmm, I have to think of this. I do order based on names, and not numbers.

You could also use the numbers `01`, `02`, `03` etc. But I can imagine that does not look nice...

theyosh, 2024-03-08:
So found a solution. The sorting will detect a sort between names starting with a number. And will than sort based on the number value. This will result in 1,2,3 .. 10, 11 etc.

Do a `git pull` and refresh the browser. It should re-arrange the enclosures

ewiandr73, 2024-03-12:
Sorry for the delay, I just needed to do a git pull, right?

It seems to be in a weird order.

[image]

Andy

ewiandr73, 2024-03-12:
Must be the Browser (Edge) firefox is ok :)

Thanks

theyosh, 2024-03-12:
Git pull should be enough. But maybe reload the browser holding down the shift key. Or clear cache. Could be a browser cache issue.

theyosh, 2024-03-12:
> Must be the Browser (Edge) firefox is ok :)
> 
> Thanks

Oh, I have only tested it on Firefox ;) I can have a look at Chrome based browsers

ewiandr73, 2024-03-12:
All good, firefox is better anyway...

theyosh, 2024-03-12:
Ah, found out that my sorting code was not correct. Now it should also work on a Chromium based browser. A `git pull` is just enough to upgrade the GUI.

ewiandr73, 2024-03-12:
That fixed it.

Is there a way to have the Area's listed in the same order across all the enclosures, currently they appear in the order they are created.  Not important but my OCD is keeps kicking in...  :)

theyosh, 2024-03-14:
Yeah, but than I have my own sorting wish ;) As I would like to have the lights always first. As this is defining the day, and therefore influence the other areas. That is the area with the main lights set.

So I change the sorting order in such a way, that main lights area is always first, and than sorting based on numbers when the names starts with a number, and else natural language.

ewiandr73, 2024-03-14:
Ok, I didn't realise that, some of my enclosures don't have the main light toggled as I wasn't sure whether it was needed or not.
I'll toggle them all and play with the naming.

Thanks again

theyosh, 2024-03-14:
Only when you toggle the main lights on, you can use that as a day source in other areas.

But when you just work with timers, it is not needed. But I use weather sunrise and set, and that is every day shifting. And with that, my humidity period will also grow, because it can only spray at 'day light' or with the lights on.

But not using that, will result in your OCD sorting ;)