theyosh/TerrariumPI #895: [BUG]: Low/High alarm deviation values not showing

manolof, 2024-02-21:
**Setup:**
 - TerrariumPI version: 4.8.3
 - Install: Docker
 - Raspberry PI: 4B

**Describe the bug**
The Low/High alarm deviation values inside the Area Settings show empty values after the user saves them and reloads the page.

**To Reproduce / Installation steps**
[image]

**Expected behavior**
The saved values of Low/High alarm deviation should show and persist after the user closes the modal window or reloads the page.

theyosh, 2024-02-27:
Hmm, so some fields are not posted correctly or not loaded.... I have to look into this.

theyosh, 2024-03-08:
The form fields are working again. Thanks for noticing

theyosh, 2024-03-08:
Also build a new docker containers with all the fixes

manolof, 2024-03-13:
Thanks! seems to work now.

closed 2024-03-13 by manolof