theyosh/TerrariumPI #898: [BUG]: Kasa relay not shutting off

HappyFrogVivarium, 2024-02-28:
- TerrariumPI version: 4.8.2 and 4.8.3
 - RaspberryPI 4
 - Kasa Smart Plug

System worked fine until I change the relay Power on time from 25 to 10 seconds and the settle time from 14400 to 10800.  If I manually stop the relay it will come on again after the settle time but again it will not shut off after the 10 seconds and I need to manually turn the relay off in the relay tab again  

The relay should turn on at 0900, 1200 and 1500 for 10 seconds 

[image]

[image]

[terrariumpi_logfile.txt](https://github.com/theyosh/TerrariumPI/files/14440205/terrariumpi_logfile.txt)

theyosh, 2024-03-06:
Hmm, that is not good. I need to investigate this.

theyosh, 2024-03-07:
Hmm, it looks like the code was doing the correct thing:

```
024-02-28 15:00:03,981 - INFO    - terrariumEngine       - Starting a new update round with 3 sensors, 7 relays, 0 buttons and 0 webcams.
2024-02-28 15:00:03,999 - INFO    - terrariumEngine       - Updated sensor 1-Wire sensor temperature named 'Lower Temp' at address '28-0000000264c6' with new value 73.62F in 0.01 seconds.
2024-02-28 15:00:04,021 - INFO    - terrariumEngine       - Updated relay gpio relay named '450 Exhaust Fan' at address '15' with new value 0.00 in 0.01 seconds.
2024-02-28 15:00:04,141 - INFO    - terrariumEngine       - Updated sensor BME280 sensor humidity named 'Humidity' at address '76' with new value 100.00% in 0.04 seconds.
2024-02-28 15:00:04,178 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named 'Grow Tent Circulation Fan' at address '192.168.141.118' with new value 0.00 in 0.05 seconds.
2024-02-28 15:00:04,256 - INFO    - terrariumEngine       - Updated sensor BME280 sensor temperature named 'Upper Temp' at address '76' with new value 75.62F in 0.01 seconds.
2024-02-28 15:00:04,328 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named '450 MistKing' at address '192.168.141.230' with new value 0.00 in 0.05 seconds.
2024-02-28 15:00:04,490 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named 'Grow Tent Light' at address '192.168.141.234' with new value 100.00 in 0.06 seconds.
2024-02-28 15:00:04,601 - INFO    - terrariumEngine       - Updated relay gpio relay named '450 Moon Lights' at address '29' with new value 0.00 in 0.01 seconds.
2024-02-28 15:00:04,713 - INFO    - terrariumEngine       - Updated relay dc-dimmer dimmer named '450 Day Lights' at address '33' with new value 100.00 in 0.01 seconds.
2024-02-28 15:00:04,825 - INFO    - terrariumEngine       - Updated relay gpio relay named '450 Circulation Fan' at address '37' with new value 0.00 in 0.01 seconds.
2024-02-28 15:00:05,579 - INFO    - terrariumArea         - Updated area Lighting named Day Lights in timer mode at enclosure Leucs Den in 0.00 seconds.
2024-02-28 15:00:05,580 - INFO    - terrariumArea         - Updated area Lighting named Moon Lights in timer mode at enclosure Leucs Den in 0.00 seconds.
2024-02-28 15:00:05,580 - INFO    - terrariumArea         - Relays for Cooling named Exhaust Fan period low are not switched on because we have to wait for 440.0 more seconds of the total settle time of 4400.0 seconds.
2024-02-28 15:00:05,581 - INFO    - terrariumArea         - Updated area Cooling named Exhaust Fan in timer mode at enclosure Leucs Den in 0.00 seconds.
2024-02-28 15:00:05,581 - INFO    - terrariumArea         - Relays for Cooling named Circulation Fan period low are not switched on because we have to wait for 1030.0 more seconds of the total settle time of 4000.0 seconds.
2024-02-28 15:00:05,581 - INFO    - terrariumArea         - Updated area Cooling named Circulation Fan in timer mode at enclosure Leucs Den in 0.00 seconds.
2024-02-28 15:00:05,582 - INFO    - terrariumArea         - Relays for Humidity named Mist King period low are not switched on because we have to wait for 30.0 more seconds of the total settle time of 10800.0 seconds.
2024-02-28 15:00:05,582 - INFO    - terrariumArea         - Updated area Humidity named Mist King in timer mode at enclosure Leucs Den in 0.00 seconds.
2024-02-28 15:00:05,585 - INFO    - terrariumEngine       - Updated Enclosure Leucs Den with 5 areas in 0.01 seconds.
2024-02-28 15:00:05,585 - INFO    - terrariumArea         - Updated area Lighting named Grow tent light in timer mode at enclosure Grow Tent in 0.00 seconds.
2024-02-28 15:00:05,586 - INFO    - terrariumArea         - Relays for Cooling named Grow Tent Fan period low are not switched on because we have to wait for 151.0 more seconds of the total settle time of 3600.0 seconds.
2024-02-28 15:00:05,586 - INFO    - terrariumArea         - Updated area Cooling named Grow Tent Fan in timer mode at enclosure Grow Tent in 0.00 seconds.
2024-02-28 15:00:05,589 - INFO    - terrariumEngine       - Updated Enclosure Grow Tent with 2 areas in 0.00 seconds.
2024-02-28 15:00:05,685 - INFO    - terrariumEngine       - Engine update done in 1.70 seconds. Waiting for 28.30 seconds for the next round.
2024-02-28 15:00:33,984 - INFO    - terrariumEngine       - Starting a new update round with 3 sensors, 7 relays, 0 buttons and 0 webcams.
2024-02-28 15:00:34,803 - INFO    - terrariumEngine       - Updated sensor 1-Wire sensor temperature named 'Lower Temp' at address '28-0000000264c6' with new value 73.62F in 0.81 seconds.
2024-02-28 15:00:34,824 - INFO    - terrariumEngine       - Updated relay gpio relay named '450 Exhaust Fan' at address '15' with new value 0.00 in 0.01 seconds.
2024-02-28 15:00:34,944 - INFO    - terrariumEngine       - Updated sensor BME280 sensor humidity named 'Humidity' at address '76' with new value 100.00% in 0.04 seconds.
2024-02-28 15:00:34,987 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named 'Grow Tent Circulation Fan' at address '192.168.141.118' with new value 0.00 in 0.06 seconds.
2024-02-28 15:00:35,055 - INFO    - terrariumEngine       - Updated sensor BME280 sensor temperature named 'Upper Temp' at address '76' with new value 75.62F in 0.01 seconds.
2024-02-28 15:00:35,142 - INFO    - terrariumEngine       - Updated relay tplinkkasa relay named '450 MistKing' at address '192.168.141.230' with new value 0.00 in 0.05 seconds.
[... cut ...]
It went on at 15:00:36 and off at 15:00:52, but the next update round, it reads that the relay is still on. 

So that means that somehow the relay off command to kasa did not work... A restart of TP can fix that.
The reconnect code for Kasa is not the best. I have to rely on other libraries for that.

Also there was a bug in the settle time logic. I made a fix for that in #900

So try a `git pull` and restart TP4. And see if it is still happening. But it looks like a Kasa connection issue :( 
And no other Kasa schemas set in the Kasa app on the mobile phone?

theyosh, 2024-03-07:
Manual shutting down a relay does not affect the settle time. Maybe that should be changed. I have to think about that.

HappyFrogVivarium, 2024-03-07:
Thank you for looking into it!  After I changed the timer to 18 seconds it responded as it should but I then lost connection to another Kasa smart plug.  But I'll perform the pull, restart and get back to you.

theyosh, 2024-03-07:
I just really did a new fix.. so do another `git pull` to be sure you have the latest code.

theyosh, 2024-03-07:
hmm, it is getting late... so code had a bug. Do another `git pull` and it all should work again.

theyosh, 2024-03-08:
So for now, I think it should all work. When you test, make sure you have the latest code.

theyosh, 2024-03-14:
Any improvements?

HappyFrogVivarium, 2024-03-26:
Sorry,  updating to this new version and I will check it.

HappyFrogVivarium, 2024-03-27:
Unfortunately, it did not respond correctly with the power on time at 15 seconds. The Kasa relay turned on at 1300 but it did not shut off without having to do so manually.  The 20-second power on worked as it should @0900 this morning.  
[Uploading terrariumpi_logfile (3).txt…]()

theyosh, 2024-03-27:
Hmm, sorry that I did not see this earlier. But I think you should add the value of 1 to the 'On duration' field. And set the value of 180 (3 * 60) in the 'Off duration' field.
This create an operating window of 1 minute every 3 hours.

And set the 'Power on time' to 15, and the settle time to 60. These values are calculated withing the 1 minute on duration setting. So 45 seconds should be enough for settle time, but I would use a bit more.

This should work better.

You can play with the timer values here: https://theyosh.github.io/TerrariumPI/scenarios/#timer-wizard And see what kind of effect you get.

theyosh, 2024-03-27:
Lets see if that does the job....

theyosh, 2024-03-28:
And? Did the relay stay off?

theyosh, 2024-04-05:
Any updates? I guess that it is working, else there would be a response I guess.

HappyFrogVivarium, 2024-04-06:
Sorry! Did not work, 18 seconds seems to be the line.  I went through all my apps to see if there was something interfering with  Kasa such as the Kasa app itself or Amazon Alexa but found nothing. TerrariumPI is the only connected app that has a schedule or timers set with that relay.  I also updated to the current version that morning and set the timer as suggested which I should also add is spot on now! With a 6 hour delay it kicked on @ 15:31:18 then the command to switch off at 15:01:36 but the relay did not turn off I had to manually do so.  
[image]

[terrariumpi_logfile (6).txt](https://github.com/theyosh/TerrariumPI/files/14894799/terrariumpi_logfile.6.txt)

theyosh, 2024-04-10:
Found the bug! Now lets see if I can fix it. The problem is that the kasa switch is using caching data which has a timeout of, you guess...., 20 seconds :)

So there is a caching issue. I have to think about the solution for this.

theyosh, 2024-04-10:
That seems not that hard. I think I fixed it. Do a `git pull` and restart TP4. And then it should work the next time period it toggles on for 18 seconds.

theyosh, 2024-05-02:
Any update? Is the issue fixed? Can we close this issue?

theyosh, 2024-05-11:
Closing due to new release which should fix this.

closed 2024-05-11 by theyosh