theyosh/TerrariumPI #899: Not an issue just some help needed!

G4KCM, 2024-03-01:
Hi,

I have just got a small UPS that I want to install on the terrarium in order that I can shut TerrariumPi down gracefully when we have a power outage. The power network supply to our house runs through a lot of trees and during winter months we often suffer from power outages, the worst part is that these are often a number of very short on off events or even just a brown outs both of which are the worst kind for computer equipment. I have twice now had catastrophic failure of the TerrariumPi SD card (I do keep a backup of the db file).  So now it is time for a terrarium UPS! I already have two UPS one for my main server cabinet POE cameras etc and another for the incoming ADSL modem. 

 I manage the two existing UPS with NUT running on my main Home Assistant server.  This monitors the state of the UPS and shuts everything down gracefully when the UPS are reach 25%.

My question….  What is the best way to shut down TerrariumPi remotely?  I think the API??  but I don’t know what the API command needs to be in order to put TerrariumPI in a shutdown state ready to have the power switched off?  Can you help me??   I have read the API docs and remain unsure so would appreciate some words of wisdom😀

Thanks Clive

theyosh, 2024-03-01:
What are the options from the UPS side?

The best thing to do is shutdown the Pi. That would also shutdown TerrariumPI software. So if there is an 'official' way to shutdown a server, that would be the best.

But, it is also possible to use the TerrariumPI API. If you are able to 'post' to an url, you can use:
`http://[username]:[password]@[terrariumpi_ip]/api/action/shutdown`
If needs to be a post action else it will no work. And this will shutdown the Pi completely. 
There is no need to actual posting data.

But, if you can tell a bit more about how you use the UPS on other servers, maybe you can use that also for the Pi

G4KCM, 2024-03-01:
> What are the options from the UPS side?
>
> The best thing to do is shutdown the Pi. That would also shutdown TerrariumPI software. So if there is an 'official' way to shutdown a server, that would be [...]

Hi, Thank you so much you have actually completely answered my question!  What I couldn’t figure out was what I needed to ‘post’  but now I see.  It will be quite easy as Home Assistant which I use to monitor my UPSs has a [REST/POST command](https://www.home-assistant.io/integrations/rest_command/) so all I need to do is initiate a shutdown of TerrariumPi using the API when the UPS has been online for a few minutes as the UPS will be monitored by Home Assistant using [NUT](https://networkupstools.org/) then shut the UPS down afterwards so it all comes back on when the main supply is restored.

Thank you and as always much appreciated!

closed 2024-03-01 by G4KCM