theyosh/TerrariumPI #900: [BUG]: Relays toggle off and immediately on, despite of Settle Time

manolof, 2024-03-03:
**Setup:**
- TerrariumPI version: 4.8.3
- Install: Docker
- Raspberry PI: 4B

**Describe the bug**
The following settings are defined for a heating area:
 [image]

_Originally I had tried with 300 seconds, but lowered to 60 while debugging._

Then, when looking at the Relay's behavior, I see the following:

```
2024-03-01 17:18:18,720 - INFO    - terrariumArea         - Set the relay GPIO devices relay named 'Relay - Heat (2)' at address '22' to 100.0 with 0 seconds delay
2024-03-01 17:18:17,570 - INFO    - terrariumArea         - Set the relay GPIO devices relay named 'Relay - Heat (2)' at address '22' to 0.0 with 0 seconds delay
```

As you notice, the value changes back to 100 within a second after it became 0.

**Expected behavior**
The relay should wait for the duration of the Settle Time, before it becomes active again.

theyosh, 2024-03-06:
Hmm, that is not good. I need to investigate this.

theyosh, 2024-03-07:
Ah, this is some how correct.... The settle time is calculated from the moment the relay went on. Not off. So if the relay is on for 1 hour, and the settle time is less than 1 hour, it does not settle... :(

And that sounds not correct...

I made this settle time with water sprayers in mind. So spray for 1 minute, and settle for 5 at least to get a correct new humidity average. And that worked, as it looked.
Other people do the same thing. For example fans. On for 1 minute, settle for 10 etc.. So that worked always..

So I fixed the code, and now the settle time is based on the last power of action. That should make it work like it should, after all those years... :)

theyosh, 2024-03-07:
hmm, it is getting late... so code had a bug. Do another `git pull` and it all should work again.

manolof, 2024-03-08:
Thanks! Any chance you can update the docker image too? I am using docker

theyosh, 2024-03-08:
New docker images are uploaded. No version change. But I guess with a docker compose pull you will get a new image

manolof, 2024-03-08:
Thanks, seems to work now!

closed 2024-03-08 by manolof