theyosh/TerrariumPI #902: [BUG]: Missing Sensirion SHTxx Sensors?

shane64, 2024-03-11:
**Setup:**
 - TerrariumPI version: 4.8.3
 - Install: Manual
 - Raspberry PI: 2B Rev 1.1

**Describe the bug**
Updated from 4.6.1 to 4.8.3 (via SD card wipe and re-install from scratch without using a backup) and I no longer see the Sensirion SHTxx temperature/humidity sensors in the list when adding new sensors.  These were available in version 4.6.1.  Have they been deprecated?

**To Reproduce / Installation steps**
Clean install (no backups used) of 4.8.3 using manual method on Buster Lite.

**Expected behavior**
Able to select the Sensirion SHT3x sensor to use my temp/humidity sensor as before.

**Screenshots**
Screenshot of the list I see (it appears to be alphabetical).  I do no see those sensors anywhere else in the list.
[image]

**Additional context**
Just looking to see if these have been deprecated or if they can be turned on again?  I ran another "sudo ./install.sh" to see if that would add them but that didn't work.  I have reviewed the FAQ and release notes since version 4.6.1 but see no mention of their removal.

theyosh, 2024-03-11:
They should be available:
[image]

But I get the feeling not all libraries are loaded... So I changed some logging. Do a `git pull` and run in debug mode: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

When it starts in debug mode, you should see some warnings about not able to load some sensor files. Please post the log here so I can see what the error message is.

shane64, 2024-03-12:
Hello.

  Thank you for the quick reply.  I have pasted the log with the relevant entry here:

```
2024-03-11 19:49:09,134 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.8.3 on a Raspberry Pi 2 Model B Rev 1.1 ...
2024-03-11 19:49:09,201 - INFO    - terrariumEngine       - Loaded 30 settings in 0.06 seconds.
2024-03-11 19:49:09,947 - WARNING - hardware.sensor       - Error loading /home/pi/TerrariumPI/hardware/sensor/shtxx_sensor.py: invalid syntax (__init__.py, line 79)
2024-03-11 19:49:11,150 - INFO    - terrariumEngine       - Loading existing sensors from database.
```

  Please advise next steps.  

Thank you.

theyosh, 2024-03-12:
Ok, this is strange and tricky... Can you do the following:

1. `cd /home/pi/TerrariumPI/`
2. `source venv/bin/activate`
3. `python -V`

paste output here

shane64, 2024-03-12:
Output is as follows:

`Python 3.7.3`

Thank you.

theyosh, 2024-03-12:
Hmm, this is very strange... So could you try the following:

1. `cd /home/pi/TerrariumPI/hardware/sensors/`
2. `rm shtxx_sensor.py`
3. `wget https://raw.githubusercontent.com/theyosh/TerrariumPI/main/hardware/sensor/shtxx_sensor.py`

And then restart in debug mode. Could be that the `shtxx_sensor.py` was some how corrupt.

shane64, 2024-03-12:
Followed steps 1 through 3 and restarted in debug mode and receive exactly the same error as before.  The file size was identical between the shtxx_sensor.py that was there and the one that was downloaded.

Thank you.

theyosh, 2024-03-12:
Ok, next test:

1. `cd /home/pi/TerrariumPI/`
2. `source venv/bin/activate`
3. `pip install sensirion-i2c-sht sensirion-i2c-driver`

and paste the outcome here?

shane64, 2024-03-12:
Output is as follows:

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: sensirion-i2c-sht in ./venv/lib/python3.7/site-packages (0.4.0)
Requirement already satisfied: sensirion-i2c-driver in ./venv/lib/python3.7/site-packages (1.0.0)
```

The error remains.  Thank you.

theyosh, 2024-03-13:
Hmm, so lets try to downgrade some libraries:

1. `cd /home/pi/TerrariumPI/`
2. `source venv/bin/activate`
3. `pip install sensirion-i2c-driver==1.0.0 sensirion-i2c-sht==0.3.0 adafruit-circuitpython-sht31d==2.3.20`

And rerun in debug mode. These are the old libraries in version 4.6.1. Lets see what this does.

shane64, 2024-03-13:
Hello.  Command from line 3 above was run with the following results:

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: sensirion-i2c-driver==1.0.0 in ./venv/lib/python3.7/site-packages (1.0.0)
Collecting sensirion-i2c-sht==0.3.0
  Downloading https://www.piwheels.org/simple/sensirion-i2c-sht/sensirion_i2c_sht-0.3.0-py3-none-any.whl (28 kB)
Collecting adafruit-circuitpython-sht31d==2.3.20
  Downloading https://www.piwheels.org/simple/adafruit-circuitpython-sht31d/adafruit_circuitpython_sht31d-2.3.20-py3-none-any.whl (7.7 kB)
Requirement already satisfied: Adafruit-Blinka in ./venv/lib/python3.7/site-packages (from adafruit-circuitpython-sht31d==2.3.20) (8.34.0)
Requirement already satisfied: adafruit-circuitpython-busdevice in ./venv/lib/python3.7/site-packages (from adafruit-circuitpython-sht31d==2.3.20) (5.2.6)
Requirement already satisfied: typing-extensions~=4.0 in ./venv/lib/python3.7/site-packages (from adafruit-circuitpython-sht31d==2.3.20) (4.7.1)
Requirement already satisfied: adafruit-circuitpython-typing in ./venv/lib/python3.7/site-packages (from adafruit-circuitpython-sht31d==2.3.20) (1.10.2)
Requirement already satisfied: Adafruit-PlatformDetect>=3.53.0 in ./venv/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-sht31d==2.3.20) (3.62.0)
Requirement already satisfied: Adafruit-PureIO>=1.1.7 in ./venv/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-sht31d==2.3.20) (1.1.11)
Requirement already satisfied: pyftdi>=0.40.0 in ./venv/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-sht31d==2.3.20) (0.54.0)
Requirement already satisfied: RPi.GPIO in ./venv/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-sht31d==2.3.20) (0.7.1)
Requirement already satisfied: rpi-ws281x>=4.0.0 in ./venv/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-sht31d==2.3.20) (5.0.0)
Requirement already satisfied: sysv-ipc>=1.1.0 in ./venv/lib/python3.7/site-packages (from Adafruit-Blinka->adafruit-circuitpython-sht31d==2.3.20) (1.1.0)
Requirement already satisfied: adafruit-circuitpython-requests in ./venv/lib/python3.7/site-packages (from adafruit-circuitpython-typing->adafruit-circuitpython-sht31d==2.3.20) (3.0.1)
Requirement already satisfied: pyusb!=1.2.0,>=1.0.0 in ./venv/lib/python3.7/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-sht31d==2.3.20) (1.2.1)
Requirement already satisfied: pyserial>=3.0 in ./venv/lib/python3.7/site-packages (from pyftdi>=0.40.0->Adafruit-Blinka->adafruit-circuitpython-sht31d==2.3.20) (3.5)
Requirement already satisfied: Adafruit-Circuitpython-ConnectionManager in ./venv/lib/python3.7/site-packages (from adafruit-circuitpython-requests->adafruit-circuitpython-typing->adafruit-circuitpython-sht31d==2.3.20) (1.0.1)
Installing collected packages: sensirion-i2c-sht, adafruit-circuitpython-sht31d
  Attempting uninstall: sensirion-i2c-sht
    Found existing installation: sensirion-i2c-sht 0.4.0
    Uninstalling sensirion-i2c-sht-0.4.0:
      Successfully uninstalled sensirion-i2c-sht-0.4.0
  Attempting uninstall: adafruit-circuitpython-sht31d
    Found existing installation: adafruit-circuitpython-sht31d 2.3.24
    Uninstalling adafruit-circuitpython-sht31d-2.3.24:
      Successfully uninstalled adafruit-circuitpython-sht31d-2.3.24
Successfully installed adafruit-circuitpython-sht31d-2.3.20 sensirion-i2c-sht-0.3.0
```

The same error as before continues to occur when run in debug mode.  Thank you.

theyosh, 2024-03-13:
So I found the issue. An external library was causing this. So again, some command to execute:

1. `cd /home/pi/TerrariumPI/`
2. `source venv/bin/activate`
3. `pip install adafruit-circuitpython-typing==1.10.1`

This should fix it. I am currently working on the installer to fix for the future.

theyosh, 2024-03-14:
Did the last actions fixed your SHT sensors?

shane64, 2024-03-15:
Hello,

  My apologies for the delay.  That seems to have gotten rid of the error and I can now add the SHT3X sensors.  Thank you!  I assume that I should re-upgrade the sensirion-i2c-sht library to 0.4.0?

  Now that I have the sensor added, I cannot communicate with it, but suspect that it's a hardware issue that I still haven't managed to solve (though I've exhausted all the ideas that I know of to try and get it working).  It's wired up the same way it was before with the previous TerrariumPI version where it was working (and I can see it in i2cdetect -y) but I'm running out of things to try and figured I'd ask if there's any way that there's a software explanation that might solve it?  (It's an SHT30, and I've tried a second on in it's place thinking it had gone bad with the same results)

  Errors as follows:

```
2024-03-15 14:52:14,908 - WARNING - terrariumEngine       - Could not take a new measurement from sensor Sensirion SHT3X sensor humidity named 'Humidity' at address '0x44,0'. Tried for 0.00 seconds. Skipping this update.
2024-03-15 14:52:14,894 - WARNING - terrariumEngine       - Could not take a new measurement from sensor Sensirion SHT3X sensor temperature named 'Temperature' at address '0x44,0'. Tried for 1.39 seconds. Skipping this update.
2024-03-15 14:52:14,885 - ERROR   - hardware.sensor       - Error updating sensor Sensirion SHT3X sensor temperature named 'Temperature' at address '0x44,0'. Check your hardware! Error getting new data from sensor Sensirion SHT3X sensor temperature named 'Temperature' at address '0x44,0': I2C transceive failed: [Errno 121] Remote I/O error
```

  If not, I'll carry on hardware troubleshooting and recommend you close this issue.

Thank you very much for your help!

theyosh, 2024-03-16:
You address is incorrect. I2C bus start at 1. So use address `0x44`. Remove the `,0` part. I guess that should do the trick

shane64, 2024-03-16:
Thank you.  I've actually got it running on bus 0 specifically as I have two of these sensors, one on each bus (as their addresses are fixed and thus conflict).  Funny that I can't even get the second one to show up on bus 1 using i2cdetect (whereas the one I originally ran worked fine on bus 0).  As I say, this is likely a hardware error (read: user error) and I'll carry on troubleshooting.  Thank again.

theyosh, 2024-04-05:
Is there something I can do. The only thing I can think of is creating small test code like this: https://sensirion.github.io/python-i2c-driver/quickstart.html The example is using I2C bus 1 (at the end of the code).

If that also fails, you know you have a hardware failure.

Else I would like to close this issue. As I cannot help much more.

theyosh, 2024-05-02:
Where you able to fix this? Can I close this issue?

closed 2024-05-11 by theyosh