theyosh/TerrariumPI #907: [BUG]: Dimming not working as expected

cyber-shifter, 2024-04-04:
**Setup:**
 - TerrariumPI version: 4.8.4
 - Install: [Manual]
 - Raspberry PI: [4]

So I've started trying to consolidate my TP3 enclosures to TP4. I have need for one less of the original enclosures, so that freed up a Pi to build the new environment on. This is the same equipment that I had in #512 as far as the dimmer,etc. I did some reading about the new system, setup, etc. According to what I've read, the PID dimming should keep the value here somewhere in the middle between the high and low alarms. Instead though, the value goes up to the low alarm, and then turns off the heat completely. I do not see a minimum dim value anymore, so the panel just drops to 0% and builds up again. The warbling of the temperature would be ok if it was above the low alarm, but it's actually below, and as I understand it, this isn't how it is supposed to work. I've included screenshots, the settings I used were modeled after the setup guide.

[image]
[image]
[image]

theyosh, 2024-04-05:
Hi,

can you also post the graph of the sensor 'Fifolet temp'? Because that is the sensor that should control the dimmer PID logic.

theyosh, 2024-04-05:
And it possible upload the terrariumpi.log file? It should log the relay changes based on the sensor values. That can help me to analyze this. Your setup seems correct.

cyber-shifter, 2024-04-08:
Yeah, I just realized I did a dumb and didn't post the temperature graph. I will get the log as well.

cyber-shifter, 2024-04-08:
[image]
How much of the log do you want? Time-wise.

theyosh, 2024-04-08:
Logging: just 1 day, last logfile should be enough.

cyber-shifter, 2024-04-09:
It's a lot. I just grabbed whatever was in the "logs" section of the web interface instead of the log file itself. See below:

`2024-04-09 07:45:23,531 - INFO    - terrariumEngine       - Engine update done in 0.30 seconds. Waiting for 29.70 seconds for the next round.
2024-04-09 07:45:23,490 - INFO    - terrariumEngine       - Updated Enclosure Fifolet with 1 areas in 0.00 seconds.
2024-04-09 07:45:23,490 - INFO    - terrariumArea         - Updated area Heating named 'Fifolet Panel' in 'sensors' mode at enclosure Fifolet in 0.00 seconds.
2024-04-09 07:45:23,485 - INFO    - terrariumEngine       - Updated Enclosure Wadjet and Enki with 0 areas in 0.00 seconds.
2024-04-09 07:45:23,483 - INFO    - terrariumEngine       - Updated Enclosure Fatima and Odin with 0 areas in 0.00 seconds.
2024-04-09 07:45:23,287 - INFO    - terrariumEngine       - Updated relay irf520-dimmer dimmer named 'Fifolet Panel' at address '32' with new value 0.00 in 0.01 seconds.
2024-04-09 07:45:23,263 - INFO    - terrariumEngine       - Updated sensor MLX90614 IR Thermometer sensor temperature named 'Fifolet Temp' at address '5a' with new value 89.01F in 0.02 seconds.
2024-04-09 07:45:23,229 - INFO    - terrariumEngine       - Starting a new update round with 1 sensors, 1 relays, 0 buttons and 0 webcams.
2024-04-09 07:44:53,478 - INFO    - terrariumEngine       - Engine update done in 0.25 seconds. Waiting for 29.75 seconds for the next round.
2024-04-09 07:44:53,438 - INFO    - terrariumEngine       - Updated Enclosure Fifolet with 1 areas in 0.00 seconds.
2024-04-09 07:44:53,437 - INFO    - terrariumArea         - Updated area Heating named 'Fifolet Panel' in 'sensors' mode at enclosure Fifolet in 0.00 seconds.
2024-04-09 07:44:53,432 - INFO    - terrariumEngine       - Updated Enclosure Wadjet and Enki with 0 areas in 0.00 seconds.
2024-04-09 07:44:53,431 - INFO    - terrariumEngine       - Updated Enclosure Fatima and Odin with 0 areas in 0.00 seconds.
2024-04-09 07:44:53,263 - INFO    - terrariumEngine       - Updated relay irf520-dimmer dimmer named 'Fifolet Panel' at address '32' with new value 0.00 in 0.01 seconds.
2024-04-09 07:44:53,246 - INFO    - terrariumEngine       - Updated sensor MLX90614 IR Thermometer sensor temperature named 'Fifolet Temp' at address '5a' with new value 89.01F in 0.01 seconds.
2024-04-09 07:44:53,226 - INFO    - terrariumEngine       - Starting a new update round with 1 sensors, 1 relays, 0 buttons and 0 webcams.
2024-04-09 07:44:23,577 - INFO    - terrariumEngine       - Engine update done in 0.35 seconds. Waiting for 29.65 seconds for the next round.
2024-04-09 07:44:23,536 - INFO    - terrariumEngine       - Updated Enclosure Fifolet with 1 areas in 0.08 seconds.
2024-04-09 07:44:23,536 - INFO    - terrariumArea         - Updated area Heating named 'Fifolet Panel' in 'sensors' mode at enclosure Fifolet in 0.08 seconds.
2024-04-09 07:44:23,535 - INFO    - terrariumEngine       - Updated Enclosure Fifolet with 1 areas in 0.00 seconds.
2024-04-09 07:44:23,535 - INFO    - terrariumArea         - Updated area Heating named 'Fifolet Panel' in 'sensors' mode at enclosure Fifolet in 0.00 seconds.
2024-04-09 07:44:23,531 - INFO    - terrariumEngine       - Updated Enclosure Wadjet and Enki with 0 areas in 0.00 seconds.
2024-04-09 07:44:23,530 - INFO    - terrariumEngine       - Updated Enclosure Fatima and Odin with 0 areas in 0.00 seconds.
2024-04-09 07:44:23,460 - INFO    - terrariumArea         - Set the relay IRF520 Dimmer dimmer named 'Fifolet Panel' at address '32' to 0.0 with 0 seconds delay
2024-04-09 07:44:23,459 - INFO    - terrariumArea         - Toggle the relays for area Heating named 'Fifolet Panel' part low to state off.
2024-04-09 07:44:23,454 - INFO    - terrariumEngine       - Updated Enclosure Wadjet and Enki with 0 areas in 0.00 seconds.
2024-04-09 07:44:23,452 - INFO    - terrariumEngine       - Updated Enclosure Fatima and Odin with 0 areas in 0.00 seconds.
2024-04-09 07:44:23,283 - INFO    - terrariumEngine       - Updated relay irf520-dimmer dimmer named 'Fifolet Panel' at address '32' with new value 100.00 in 0.01 seconds.
2024-04-09 07:44:23,258 - INFO    - terrariumEngine       - Updated sensor MLX90614 IR Thermometer sensor temperature named 'Fifolet Temp' at address '5a' with new value 89.15F in 0.02 seconds.
2024-04-09 07:44:23,224 - INFO    - terrariumEngine       - Starting a new update round with 1 sensors, 1 relays, 0 buttons and 0 webcams.
2024-04-09 07:43:53,481 - INFO    - terrariumEngine       - Engine update done in 0.26 seconds. Waiting for 29.74 seconds for the next round.
2024-04-09 07:43:53,435 - INFO    - terrariumEngine       - Updated Enclosure Fifolet with 1 areas in 0.01 seconds.
2024-04-09 07:43:53,435 - INFO    - terrariumArea         - Updating IRF520 Dimmer dimmer named 'Fifolet Panel' at address '32' to value 100%. Current: 88.79F, target: 92.50F
2024-04-09 07:43:53,431 - INFO    - terrariumArea         - Updated area Heating named 'Fifolet Panel' in 'sensors' mode at enclosure Fifolet in 0.00 seconds.
2024-04-09 07:43:53,427 - INFO    - terrariumEngine       - Updated Enclosure Wadjet and Enki with 0 areas in 0.00 seconds.
2024-04-09 07:43:53,425 - INFO    - terrariumEngine       - Updated Enclosure Fatima and Odin with 0 areas in 0.00 seconds.
2024-04-09 07:43:53,258 - INFO    - terrariumEngine       - Updated relay irf520-dimmer dimmer named 'Fifolet Panel' at address '32' with new value 100.00 in 0.01 seconds.
[... cut ...]
2024-04-09 07:12:53,282 - INFO    - terrariumArea         - Updated area Heating named 'Fifolet Panel' in 'sensors' mode at enclosure Fifolet in 0.00 seconds.
2024-04-09 07:12:53,277 - INFO    - terrariumEngine       - Updated Enclosure Wadjet and Enki with 0 areas in 0.00 seconds.
2024-04-09 07:12:53,275 - INFO    - terrariumEngine       - Updated Enclosure Fatima and Odin with 0 areas in 0.00 seconds.
2024-04-09 07:12:53,109 - INFO    - terrariumEngine       - Updated relay irf520-dimmer dimmer named 'Fifolet Panel' at address '32' with new value 100.00 in 0.01 seconds.
 in 0.01 seconds.`

cyber-shifter, 2024-04-09:
The log file is pretty large. Rather than paste its contents here, I have attached the file.
[terrariumpi.2024-04-07.log](https://github.com/theyosh/TerrariumPI/files/14918812/terrariumpi.2024-04-07.log)

theyosh, 2024-04-09:
Is there any change you have toggled the 'ignore **high** alarm' on the '**High** alarm' tab? Wild guess

cyber-shifter, 2024-04-10:
As a matter of fact, yes.

theyosh, 2024-04-10:
Toggle that to off... I think that will make it work like expected.

cyber-shifter, 2024-04-11:
Ah ha! That did it! 
[image]

cyber-shifter, 2024-04-11:
Do you have recommendations on how to smooth out the power curve? This looks a lot better than the sawtooth I had before, but I'd like to have as smooth of a power curve as I can, unless you think it isn't going to create problems. I am not so experienced with such power and heating systems.

[image]

theyosh, 2024-04-11:
You could lower the settle time to about 25 seconds. So it will update the relay every update run of TP. Now you have 60 seconds, which will make the relay only change once a minute. So that can be lowered.

And that is it. And a stronger heater could also help, as the dimmer is most of the time at around 100% power. So when it gets colder, there is no spare power left.

cyber-shifter, 2024-04-11:
I need to adjust the alarm values because this works differently than it did in TP3. Plus there is a difference between the temperature that the sensor reads and the actual temperature of the substrate (this has to do with the fact that I haven't figured out how to adjust the emissivity setting on the IR sensor). So basically I do trial and error readings with a known good temp good, and what the system says the temp is, and adjust once I figure out the offset. The radiant panel is sized for the enclosures and the animals, I just need to adjust now that I have gotten the whole system working like it is supposed to. 

That being said, thank you so much for all your work on this, and for your help to me here. I greatly appreciate it.

theyosh, 2024-04-11:
You can add or subtract a value in the offset field in the calibration part of the sensor.

cyber-shifter, 2024-04-12:
Yeah, that's what I usually do. It's not perfect, but it functionally works enough to keep the environment steady. I'm going to try what you suggested to smooth out the power.

cyber-shifter, 2024-04-12:
The power seems to be smoothing out. But I remembered why the offset wasn't a perfect solution. The difference isn't linear, so I just have to kinda get to a compromise point that works. I'm going to look up some more information about this sensor. The temp gun I have that uses an IR sensor has the ability to change the emissivity, so I'm guessing there is a way to do it for these sensors. It may be something I can code in for the sensor.

theyosh, 2024-04-12:
I can close it also. Nice to know that it is working better. And I do understand that an offset is not working for you. 
So for now I close this issue.

closed 2024-04-12 by theyosh