theyosh/TerrariumPI #909: [FEATURE]: Obfuscate password attempts in logs

cyber-shifter, 2024-04-15:
I'm not sure if anyone has thought about this or not, but I noticed this when looking for some other information in the logs. When you fail to log in with an administrator account to TerrariumPi, it puts the "password" that was used for the attempt in plain text, in the logs. Obviously whether or not this represents sensitive information depends on your personal stance and experience. As someone who works in IT every day, this isn't something I would like my devices doing. But I'm also a bit paranoid about IoT devices. So I wanted to just start the conversation about it I guess.

theyosh, 2024-04-16:
Euh, it should NOT be logged. So I will have a look into that.

My goal is NOT to log sensitive data.

cyber-shifter, 2024-04-16:
Oh! Well! I'm glad I said something. I wasn't going to originally.

theyosh, 2024-04-16:
Done!. It now logs stars.

cyber-shifter, 2024-04-16:
Excellent!! My paranoid IT brain can rest easier.

closed 2024-04-16 by cyber-shifter