theyosh/TerrariumPI #910: [FEATURE]:PWM Fan Control

MeanGreen420, 2024-04-17:
Id like to be able to have a variable PWM fan control for enclosures so it can turn on at a slow rpm with its getting warm and speeds up as the temp increases. I was also wanting to use it to control the Terrarium Pi Equipment fan so it can ramp up if it starts getting to warm.

It would be nice to be able to setup a timer feature on it also so it could simulate a breeze.

Thanks

theyosh, 2024-04-22:
This is possible by creating a `cooling area`. https://theyosh.github.io/TerrariumPI/setup/#other-areas

Use `temperature` mode and select the sensors you want to use.

In your case, select the PWM dimmer at the `high alarm` tab.

And that should do it. The sensor(s) `alarm min` and 'alarm max` will control the temperature. As you are using a dimmer, it will try to keep the temperature at the middle of the min and max alarm values. The fan should toggle on when the high alarm is triggered.

But it can happen that the fan will not spin up with low voltages... That is always a bit tricky.

MeanGreen420, 2024-04-22:
Ok, I thought that would work too just haven't had a chance to test it yet.

Thanks!

closed 2024-04-22 by MeanGreen420