theyosh/TerrariumPI #911: [BUG]: Lights turn on and off unexpectedly, outside schedule

manolof, 2024-04-19:
**Setup:**
- TerrariumPI version: 4.8.3
- Install: Docker
- Raspberry PI: 4B

**Describe the bug**
I have noticed that my main lights turn on every day, exactly one hour after the scheduled turn off time.
I am using Weather mode for my main lights. So if due to weather mode my lights are turning off at 20:20, then suddenly they turn on and off again at 21:20.

In addition, I noticed in the enclosure's area that the timestamps of 2 different areas (Main Lights, and Sunrise/Sunset Lights) have different values for sunrise/sunset, and only one of them corresponds to the correct value according to the Weather page in Dashboard. See screenshots.

**Screenshots**
Area Settings:
[image]
"Light" Relay:
[image]

Area:
[image]

Weather page:
[image]

theyosh, 2024-05-02:
Sorry for the late reply. But this is strange. Is this happening every evening?

The difference between your light and sunrise - sunset areas could be a day difference calculation. One area is one day ahead. After 24 hours, I expect that they should be the same.

I have to look into this, as I do not have this issue, while I have almost the same setup with the lights based on weather data. And also with a delay (only 30 min). So we have almost the same setup.

I have an idea, but that will take some time to test.

manolof, 2024-05-03:
It hasn't happened for a few days now.

I noticed something today: when I went into the Main Lights settings, and changed the "Maximum hours" from 14 to 15, both times displayed on the main Enclosures list showed the same - they got aligned!
Makes me wonder if this could be the root problem for the issue I had with the light that turns on one hour after light cut-off...

theyosh, 2024-05-05:
You conclusion about the maximum hours is correct. This looks the best to me. The same goes for the min amount of hours. That will also recalculate from the middle of the day (sun rise - set period).

But the problem here is I think that when the day is over, a new day is calculated. And for me that means 6 min more light every day going to summer period. 3 min earlier and 3 minutes longer. And that 3 minutes longer, is probably the problem. That will trigger the lights to go on, just after the lights went off. I should probably fix the new day calculation code. But that is already complex.

So for now, I have created a settle time of 15 minutes. And that should fix this issue for now.

In order to test this, you need to switch to the `bookworm` branch. And rerun the installer. I hope you know how to switch branches.

theyosh, 2024-05-10:
Oh crap, this is a docker install. That needs a new build/release. But I will release a new version this weekend with this fix.

theyosh, 2024-05-11:
New docker images are build. The version numbering has changed a bit https://theyosh.github.io/TerrariumPI/posts/2024/05/11/release-terrariumpi-v-4-9-0/#docker

theyosh, 2024-06-08:
Did the docker updates solved this issue?

manolof, 2024-06-12:
Sorry for taking long to respond. I just upgraded the system. I will close the issue for now and reopen if issue persists. Thanks!

closed 2024-06-12 by manolof

manolof, 2024-06-12:
As I was upgrading the system, I got errors due to the new builds. This is from docker logs:
[image]

```
$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

I first tried with the `latest` tag, as it was before. The docker wouldn't start. Then I started trying specific version, like 4.9.1-bookwork-java, 4.9.0-buster, but none worked.
I even tried `sudo apt-get install netcat-traditional` due to that command in run.sh failing (in my logs).
It only worked and started again when I downgraded to 4.8.4

theyosh, 2024-06-12:
Euh, this was indeed a different bug. I have fixed the Docker images. They should work again. Would be great if you can run the `:latest` image again