theyosh/TerrariumPI #916: [BUG]: No MQTT Connection after v4.9.0 upgrade

G4KCM, 2024-05-14:
Hi,
Just upgraded this afternoon to v4.9.0 on my Raspberry Pi Bullseye based system using manual install. After upgrade there is no MQTT reporting, it was working fine immediately prior to upgrade and all other devices are connecting to and using my MQTT server fine.  Tried a reboot but still no MQTT connection.  All else seems Ok.
Rgds - Clive

Log copied below...

2024-05-14 23:48:01,695 - ERROR   - terrariumNotification - Could not send message 30sec Relay State to MQTT to topic terrariumpi/relay/update/6f88282fad8d92ed91bb42b91b960cb7 as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:48:01,693 - INFO    - terrariumRelay        - Updated relay gpio relay named 'Relay 5 - Humidifier' at address '36' with new value 0.00 in 0.01 seconds.
2024-05-14 23:48:01,667 - INFO    - terrariumSensor       - Updated sensor SI7021 sensor temperature named 'Temperature Mid Cage Left' at address '0x40,3' with new value 21.37C in 0.01 seconds.
2024-05-14 23:48:01,666 - ERROR   - terrariumNotification - Could not send message 30 Sec Sensor to MQTT to topic terrariumpi/sensor/update/ba3149076452551bddffde799e629af8 as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:48:01,585 - ERROR   - terrariumNotification - Could not send message 30sec Relay State to MQTT to topic terrariumpi/relay/update/f06f3b8adbe61b8398f4b8ec9fd99e2c as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:48:01,582 - INFO    - terrariumRelay        - Updated relay gpio relay named 'Relay 6 - Fan' at address '35' with new value 0.00 in 0.01 seconds.
2024-05-14 23:48:01,555 - INFO    - terrariumSensor       - Updated sensor SI7021 sensor humidity named 'Humidity Mid Cage Left' at address '0x40,3' with new value 72.43% in 0.11 seconds.
2024-05-14 23:48:01,554 - ERROR   - terrariumNotification - Could not send message 30 Sec Sensor to MQTT to topic terrariumpi/sensor/update/3a19a045a80676b5c482af2925bb77b0 as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:48:01,474 - ERROR   - terrariumNotification - Could not send message 30sec Relay State to MQTT to topic terrariumpi/relay/update/dc7461cbd9ae484501f20ddef35e2a5c as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:48:01,472 - INFO    - terrariumRelay        - Updated relay gpio relay named 'Relay 4 - Floor Heater' at address '33' with new value 0.00 in 0.01 seconds.
2024-05-14 23:48:01,364 - ERROR   - terrariumNotification - Could not send message 30sec Relay State to MQTT to topic terrariumpi/relay/update/bc7722e9727bccf48bcb88afc7b2c15e as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:48:01,361 - INFO    - terrariumRelay        - Updated relay gpio relay named 'Relay 3 - Infra Red Light' at address '32' with new value 0.00 in 0.01 seconds.
2024-05-14 23:48:01,340 - INFO    - terrariumSensor       - Updated sensor SI7021 sensor temperature named 'Temperature Mid Cage Centre' at address '0x40,1' with new value 22.06C in 0.01 seconds.
2024-05-14 23:48:01,339 - ERROR   - terrariumNotification - Could not send message 30 Sec Sensor to MQTT to topic terrariumpi/sensor/update/cf9f6ef782e0951e8d489a9386aaa48b as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:48:01,253 - ERROR   - terrariumNotification - Could not send message 30sec Relay State to MQTT to topic terrariumpi/relay/update/0c040bb1d6e0aa8b53306296a2a4998c as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:48:01,251 - INFO    - terrariumRelay        - Updated relay gpio relay named 'Relay 2 - Basking Light' at address '31' with new value 0.00 in 0.01 seconds.
2024-05-14 23:48:01,228 - INFO    - terrariumSensor       - Updated sensor SI7021 sensor humidity named 'Humidity Mid Cage Centre' at address '0x40,1' with new value 69.98% in 0.13 seconds.
2024-05-14 23:48:01,226 - ERROR   - terrariumNotification - Could not send message 30 Sec Sensor to MQTT to topic terrariumpi/sensor/update/3854995fe2989eb03e69c9b62e0b8f0e as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:48:01,142 - ERROR   - terrariumNotification - Could not send message 30sec Relay State to MQTT to topic terrariumpi/relay/update/f403bea98501a5ee72f7767a41be13f1 as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:48:01,137 - INFO    - terrariumRelay        - Updated relay gpio relay named 'Relay 1 - Main Lights' at address '29' with new value 0.00 in 0.02 seconds.
2024-05-14 23:48:01,066 - INFO    - terrariumEngine       - Starting a new update round with 12 sensors, 8 relays, 0 buttons and 0 webcams.
2024-05-14 23:47:34,667 - INFO    - terrariumEngine       - Engine update done in 3.60 seconds. Waiting for 26.40 seconds for the next round.
2024-05-14 23:47:34,622 - INFO    - terrariumEngine       - Message of the Day generated in 0.12 seconds
2024-05-14 23:47:34,502 - INFO    - terrariumEnclosure    - Updated Enclosure Marley's Cage with 7 areas in 0.05 seconds.
2024-05-14 23:47:34,497 - INFO    - terrariumArea         - Updated area Cooling named 'Fan' in 'sensors' mode at enclosure Marley's Cage in 0.01 seconds.
2024-05-14 23:47:34,491 - INFO    - terrariumArea         - Updated area Heating named 'Infra Red Heater' in 'sensors' mode at enclosure Marley's Cage in 0.01 seconds.
2024-05-14 23:47:34,482 - INFO    - terrariumArea         - Updated area Humidity named 'Cage Humidity' in 'sensors' mode at enclosure Marley's Cage in 0.01 seconds.
2024-05-14 23:47:34,474 - INFO    - terrariumArea         - Updated area Heating named 'Basking Light' in 'sensors' mode at enclosure Marley's Cage in 0.01 seconds.
2024-05-14 23:47:34,473 - INFO    - terrariumArea         - Relays for Heating named 'Basking Light' period low are not switched on because the lights are off while on is requested.
2024-05-14 23:47:34,465 - INFO    - terrariumArea         - Updated area Water tank named 'Humidifier Tank Level' in 'sensors' mode at enclosure Marley's Cage in 0.00 seconds.
2024-05-14 23:47:34,461 - INFO    - terrariumArea         - Updated area Heating named 'Floor  Heater' in 'sensors' mode at enclosure Marley's Cage in 0.01 seconds.
2024-05-14 23:47:34,454 - INFO    - terrariumArea         - Updated area Lighting named 'Main Light' in 'timer' mode at enclosure Marley's Cage in 0.00 seconds.
2024-05-14 23:47:34,320 - INFO    - terrariumSensor       - Updated sensor 1-Wire sensor temperature named '1-Wire sensor Mid Cage Centre Temperature' at address '28-03172522afff' with new value 22.19C in 0.82 seconds.
2024-05-14 23:47:34,318 - ERROR   - terrariumNotification - Could not send message 30 Sec Sensor to MQTT to topic terrariumpi/sensor/update/51a5cc965ac22e8c477501cc90524623 as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:47:33,393 - INFO    - terrariumSensor       - Updated sensor 1-Wire sensor temperature named '1-Wire sensor Mid Cage Left Temperature' at address '28-0214815d42ff' with new value 21.94C in 0.81 seconds.
2024-05-14 23:47:33,390 - ERROR   - terrariumNotification - Could not send message 30 Sec Sensor to MQTT to topic terrariumpi/sensor/update/cd0edc30a0f07eb781f628511962dc3d as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:47:32,473 - INFO    - terrariumSensor       - Updated sensor 1-Wire sensor temperature named '1-Wire sensor Mid Cage Right Temperature' at address '28-021480f109ff' with new value 22.06C in 0.01 seconds.
2024-05-14 23:47:32,471 - ERROR   - terrariumNotification - Could not send message 30 Sec Sensor to MQTT to topic terrariumpi/sensor/update/cd77738d74b7556c8c1f2d17f760d33e as we are not connected to the MQTT broker at address: 192.168.1.22:1883
2024-05-14 23:47:32,360 - INFO    - terrariumSensor       - Updated sensor 1-Wire sensor temperature named '1-Wire sensor Floor Temperature 2' at address '28-021480e0efff' with new value 22.00C in 0.01 seconds.
2024-05-14 23:47:32,358 - ERROR   - terrariumNotification - Could not send message 30 Sec Sensor to MQTT to topic terrariumpi/sensor/update/23d3374c615266bdb572064630bbf20b as we are not connected to the MQTT broker at address: 192.168.1.22:1883
[... cut ...]
2024-05-14 23:46:04,458 - INFO    - terrariumArea         - Updated area Heating named 'Basking Light' in 'sensors' mode at enclosure Marley's Cage in 0.01 seconds.
2024-05-14 23:46:04,457 - INFO    - terrariumArea         - Relays for Heating named 'Basking Light' period low are not switched on because the lights are off while on is requested.
2024-05-14 23:46:04,449 - INFO    - terrariumArea         - Updated area Water tank named 'Humidifier Tank Level' in 'sensors' mode at enclosure Marley's Cage in 0.00

theyosh, 2024-05-15:
There should be warnings when TP starts up about not able to login into the MQTT server. So grep for `to MQTT Broker` in the logfile. Maybe there is some more information when TP starts up.

Also, when you edit the notification settings, it is all still correct values?

G4KCM, 2024-05-15:
> There should be warnings when TP starts up about not able to login into the MQTT server. So grep for `to MQTT Broker` in the logfile. Maybe there is some more information when TP starts up.
> 
> Also, when you edit the notification settings, it is all still correct values?

All is Ok in the configuration file here is the part of the log where the new TP starts, looks like something is not correct?

May 14 16:19:58 terrariumpi python[496]: 2024-05-14 16:19:58,648 - INFO    - OpenWeatherMap        - Using Openweathermap One Call API 2.5
May 14 16:19:58 terrariumpi systemd[1]: systemd-fsckd.service: Succeeded.
May 14 16:19:59 terrariumpi python[496]: 2024-05-14 16:19:59,847 - INFO    - OpenWeatherMap        - Loaded new historical weather data (40 measurements) from 2024-05-13 12:00:00 till 2024-05-15 03:00:00 in 1.20 seconds.
May 14 16:19:59 terrariumpi python[496]: 2024-05-14 16:19:59,849 - INFO    - terrariumWeather      - Loaded weather data from source https://api.openweathermap.org/data/2.5/weather?q=Noumea,NC&appid=144d3f645a94f10491acb972422f3036 in 1.74 seconds
May 14 16:19:59 terrariumpi python[496]: Exception ignored in: <function Client.__del__ at 0xf64aeb68>
May 14 16:19:59 terrariumpi python[496]: Traceback (most recent call last):
May 14 16:19:59 terrariumpi python[496]:   File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/paho/mqtt/client.py", line 874, in __del__
May 14 16:19:59 terrariumpi python[496]:     self._reset_sockets()
May 14 16:19:59 terrariumpi python[496]:   File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/paho/mqtt/client.py", line 1133, in _reset_sockets
May 14 16:19:59 terrariumpi python[496]:     self._sock_close()
May 14 16:19:59 terrariumpi python[496]:   File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/paho/mqtt/client.py", line 1119, in _sock_close
May 14 16:19:59 terrariumpi python[496]:     if not self._sock:
May 14 16:19:59 terrariumpi python[496]: AttributeError: 'Client' object has no attribute '_sock'
May 14 16:19:59 terrariumpi python[496]: 2024-05-14 16:19:59,945 - WARNING - terrariumNotification - Failed connecting to MQTT Broker at address: 192.168.1.22:1883: __init__() missing 1 required positional argument: 'callback_api_version'
May 14 16:20:01 terrariumpi python[496]: 2024-05-14 16:20:01,631 - ERROR   - terrariumNotification - Could not send message Starting up TerrariumPI 4.9.0 on a Raspberry Pi 4 Model B Rev 1.5 ... to topic terrariumpi/system/broadcast as we are not connected to the MQTT broker at address: 192.168.1.22:1883
May 14 16:20:02 terrariumpi python[496]: 2024-05-14 16:20:02,200 - INFO    - terrariumEngine       - Start loading total power and water usage
May 14 16:20:09 terrariumpi systemd[1]: systemd-hostnamed.service: Succeeded.
May 14 16:20:14 terrariumpi python[496]: 2024-05-14 16:20:14,664 - INFO    - terrariumEngine       - Loaded total power and water usage in 12.46 seconds.
May 14 16:20:14 terrariumpi python[496]: 2024-05-14 16:20:14,664 - INFO    - terrariumEngine       - Loading existing sensors from database.

For reference here is startup prior to upgrade to 4.9.0...
May 14 13:18:06 terrariumpi systemd[1]: Started watchdog daemon.
May 14 13:18:06 terrariumpi systemd[1]: Startup finished in 3.490s (kernel) + 1min 19.411s (userspace) = 1min 22.902s.
May 14 13:18:06 terrariumpi python[495]: 2024-05-14 13:18:06,977 - INFO    - terrariumEngine       - Loaded 30 settings in 0.18 seconds.
May 14 13:18:07 terrariumpi python[495]: 2024-05-14 13:18:07,365 - INFO    - terrariumEngine       - Loading weather data from source https://api.openweathermap.org/data/2.5/weather?q=Noumea,NC&appid=144d3f645a94f10491acb972422f3036
May 14 13:18:09 terrariumpi python[495]: 2024-05-14 13:18:09,338 - INFO    - weather.openweathermap_org_weather - Using Openweathermap One Call API 2.5
May 14 13:18:10 terrariumpi python[495]: 2024-05-14 13:18:10,178 - INFO    - weather.openweathermap_org_weather - Loaded new historical weather data (48 measurements) from 2024-05-12 12:00:00 till 2024-05-14 11:00:00 in 0.84 seconds.
May 14 13:18:10 terrariumpi python[495]: 2024-05-14 13:18:10,180 - INFO    - weather               - Loaded new weather data in 2.811 seconds.
May 14 13:18:11 terrariumpi python[495]: 2024-05-14 13:18:11,148 - INFO    - terrariumNotification - Connecting to MQTT Broker at address: 192.168.1.22:1883 ...
May 14 13:18:11 terrariumpi python[495]: 2024-05-14 13:18:11,407 - INFO    - terrariumNotification - Logged in to MQTT Broker at address: 192.168.1.22:1883.
May 14 13:18:15 terrariumpi python[495]: 2024-05-14 13:18:15,687 - INFO    - terrariumEngine       - Loading existing sensors from database.
May 14 13:18:15 terrariumpi python[495]: 2024-05-14 13:18:15,817 - INFO    - terrariumEngine       - Loaded sensor SI7021 sensor humidity named 'Humidity Mid Cage Centre' at address '0x40,1' with value 69.49% in 0.12 seconds.
May 14 13:18:15 terrariumpi python[495]: 2024-05-14 13:18:15,822 - INFO    - terrariumEngine       - Loaded sensor SI7021 sensor temperature named 'Temperature Mid Cage Centre' at address '0x40,1' with value 21.37C in 0.00 seconds.
May 14 13:18:15 terrariumpi python[495]: 2024-05-14 13:18:15,926 - INFO    - terrariumEngine       - Loaded sensor SI7021 sensor humidity named 'Humidity Mid Cage Left' at address '0x40,3' with value 69.98% in 0.10 seconds.
May 14 13:18:15 terrariumpi python[495]: 20

Rgds - Clive

G4KCM, 2024-05-15:
> There should be warnings when TP starts up about not able to login into the MQTT server. So grep for `to MQTT Broker` in the logfile. Maybe there is some more information when TP starts up.
> 
> Also, when you edit the notification settings, it is all still correct values?

Looks like it might be something to do with this..?
paho-mqtt 2.0, released on Feb, 10th, now requires an additional argument to mqtt.Client(), see their [migration notes](https://github.com/eclipse/paho.mqtt.python/blob/v2.0.0/docs/migrations.rst).

theyosh, 2024-05-15:
If you add the following code `mqtt.CallbackAPIVersion.VERSION1` on line in the file `terrariumNotification.py` https://github.com/theyosh/TerrariumPI/blob/96440499212b7010c50d8698e04de7f53d824a3b/terrariumNotification.py#L2327 so that the line will become:

`self.connection = mqtt.Client(mqtt.CallbackAPIVersion.VERSION1, client_id=f"TerrariumPI {self.setup['version']}")`

And restart TerrariumPI. This should make it work again.

G4KCM, 2024-05-15:
> If you add the following code `mqtt.CallbackAPIVersion.VERSION1` on line in the file `terrariumNotification.py` [...]
🤙 Yep that has fixed it.  Thank you!

closed 2024-05-15 by G4KCM

theyosh, 2024-05-15:
Nice! Fix is also pushed to Github. Thanks for testing!