theyosh/TerrariumPI #922: [BUG]:

Konrad84, 2024-06-05:
Hi Yosh,

thank you for your developement !

I got a problem. I tried to upgrade my TerrariumPI during the Upgrade Process there was some red signs about missing files or something. After Update and reboot i can´t start Terrarium PI. I can get into the Folder but no reaction if i try sudo service terrariumPI start oder sudo ./ run.sh etc. .

Have you an idea what i can do?

theyosh, 2024-06-05:
the ./runs.sh should not be run. Not sure what it will do on a Pi, but that is for Docker.

But you should be able to rerun the installer.

So what happens when you just run `./install.sh` again?

And than run it in debug mode: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

And did you made a backup of your database? https://theyosh.github.io/TerrariumPI/faq/which-files-to-backup/ So you could reinstall the software at all.

Konrad84, 2024-06-06:
ok that worked now i get this during debug

pi@raspberrypi:~/TerrariumPI $ source venv/bin/activate
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2024-06-06 22:05:11,746 - INFO    - terrariumEngine       - Starting up Terrariu                                                                                                                                                             mPI 4.9.0 on a Raspberry Pi 3 Model B Rev 1.2 ...
2024-06-06 22:05:11,788 - INFO    - terrariumEngine       - Loaded 32 settings i                                                                                                                                                             n 0.04 seconds.
2024-06-06 22:05:12,852 - INFO    - OpenWeatherMap        - Using Openweathermap                                                                                                                                                              Free API
2024-06-06 22:05:12,855 - INFO    - terrariumWeather      - Loaded weather data                                                                                                                                                              from source https://api.openweathermap.org/data/2.5/weather?q=Erding,DE&appid=de                                                                                                                                                             7a1fbd5f74afc83284ce53b8c9b504 in 1.02 seconds
/home/pi/TerrariumPI/terrariumNotification.py:2742: PTBUserWarning: If 'per_mess                                                                                                                                                             age=False', 'CallbackQueryHandler' will not be tracked for every message. Read t                                                                                                                                                             his FAQ entry to learn more about the per_* settings: https://github.com/python-                                                                                                                                                             telegram-bot/python-telegram-bot/wiki/Frequently-Asked-Questions#what-do-the-per                                                                                                                                                             _-settings-in-conversationhandler-do.
  ConversationHandler(
/home/pi/TerrariumPI/terrariumNotification.py:2749: PTBUserWarning: If 'per_mess                                                                                                                                                             age=False', 'CallbackQueryHandler' will not be tracked for every message. Read t                                                                                                                                                             his FAQ entry to learn more about the per_* settings: https://github.com/python-                                                                                                                                                             telegram-bot/python-telegram-bot/wiki/Frequently-Asked-Questions#what-do-the-per                                                                                                                                                             _-settings-in-conversationhandler-do.
  ConversationHandler(
^CTraceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 27, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 135, in __init__
    self.calendar = terrariumCalendar()
  File "/home/pi/TerrariumPI/terrariumCalendar.py", line 34, in __init__
    for event in self.get_events(
  File "/home/pi/TerrariumPI/terrariumCalendar.py", line 75, in get_events
    for event_data in events(string_content=self.__ical.to_ical(), start=start, end=end):
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/icalevents/icalevents.py", line 57, in events
    found_events += parse_events(
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/icalevents/icalparser.py", line 391, in parse_events
    if not event.recurrence_id and (event.uid, event.start) in [
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/icalevents/icalparser.py", line 392, in <listcomp>
    (f.uid, f.recurrence_id) for f in found
KeyboardInterrupt
2024-06-06 22:07:08,116 - WARNING - terrariumNotification - Error connecting to Telegram. Just retry once more: Timed out
2024-06-06 22:07:08,126 - ERROR   - terrariumNotification - Error connecting to Telegram: Unknown error in HTTP implementation: RuntimeError('cannot schedule new futures after interpreter shutdown')
^CKeyboardInterrupt
2024-06-06T20:08:00Z
Exception ignored in: <module 'threading' from '/usr/lib/python3.9/threading.py'>
Traceback (most recent call last):
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/monkey.py", line 868, in _shutdown
    orig_shutdown()
  File "/usr/lib/python3.9/threading.py", line 1428, in _shutdown
    lock.acquire()
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/gevent/thread.py", line 112, in acquire
    acquired = BoundedSemaphore.acquire(self, blocking, timeout)
  File "src/gevent/_semaphore.py", line 184, in gevent._gevent_c_semaphore.Semaphore.acquire
[... cut ...]
  File "src/gevent/_abstract_linkable.py", line 451, in gevent._gevent_c_abstract_linkable.AbstractLinkable._switch_to_hub
  File "src/gevent/_greenlet_primitives.py", line 61, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_greenlet_primitives.py", line 65, in gevent._gevent_c_greenlet_primitives.SwitchOutGreenletWithLoop.switch
  File "src/gevent/_gevent_c_greenlet_primitives.pxd", line 35, in gevent._gevent_c_greenlet_primitives._greenlet_switch
KeyboardInterrupt:

theyosh, 2024-06-06:
You stopped to early.. There was no error with telegram... just a message. Please keep it running until it says it is started and webgui available

Konrad84, 2024-06-06:
oh ok sorry i ll try again

Konrad84, 2024-06-06:
ok i waited about 30 min. and it stops at this point

pi@raspberrypi:~ $ cd TerrariumPI
pi@raspberrypi:~/TerrariumPI $ source venv/bin/activate
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2024-06-06 22:40:12,957 - INFO    - terrariumEngine       - Starting up Terrariu                                                                                                                                                             mPI 4.9.0 on a Raspberry Pi 3 Model B Rev 1.2 ...
2024-06-06 22:40:12,999 - INFO    - terrariumEngine       - Loaded 32 settings i                                                                                                                                                             n 0.04 seconds.
2024-06-06 22:40:13,923 - INFO    - OpenWeatherMap        - Using Openweathermap                                                                                                                                                              Free API
2024-06-06 22:40:13,926 - INFO    - terrariumWeather      - Loaded weather data                                                                                                                                                              from source https://api.openweathermap.org/data/2.5/weather?q=Erding,DE&appid=de                                                                                                                                                             7a1fbd5f74afc83284ce53b8c9b504 in 0.88 seconds
/home/pi/TerrariumPI/terrariumNotification.py:2742: PTBUserWarning: If 'per_mess                                                                                                                                                             age=False', 'CallbackQueryHandler' will not be tracked for every message. Read t                                                                                                                                                             his FAQ entry to learn more about the per_* settings: https://github.com/python-                                                                                                                                                             telegram-bot/python-telegram-bot/wiki/Frequently-Asked-Questions#what-do-the-per                                                                                                                                                             _-settings-in-conversationhandler-do.
  ConversationHandler(
/home/pi/TerrariumPI/terrariumNotification.py:2749: PTBUserWarning: If 'per_mess                                                                                                                                                             age=False', 'CallbackQueryHandler' will not be tracked for every message. Read t                                                                                                                                                             his FAQ entry to learn more about the per_* settings: https://github.com/python-                                                                                                                                                             telegram-bot/python-telegram-bot/wiki/Frequently-Asked-Questions#what-do-the-per                                                                                                                                                             _-settings-in-conversationhandler-do.
  ConversationHandler(

theyosh, 2024-06-06:
Hmm, looks like Telegram is causing this... That is not handy, as I cannot test it....

So we can disable Telegram for now with the following commands. Go to the database folder located in the `data` folder: `/home/pi/TerrariumPI/data`

1. `sqlite3 terrariumpi.db`
2. `update NotificationService set enabled = 0 where type = 'telegram';`
3. Press Ctrl+D

And now start in debug mode again. Make sure you change to the correct folder again.

theyosh, 2024-06-08:
Did the above help? Was you able to disable telegram and start up TerrariumPI?

Konrad84, 2024-06-08:
Ok now i get this 

pi@raspberrypi:~/TerrariumPI/data $ ls
calendar.ics  terrariumpi.db      terrariumpi.db-wal
terrarium.db  terrariumpi.db-shm
pi@raspberrypi:~/TerrariumPI/data $ sqlite3 terrariumpi.db
SQLite version 3.34.1 2021-01-20 14:10:07
Enter ".help" for usage hints.
sqlite> update NotificationService set enabled = 0 where type = 'telegram';                                          sqlite>

pi@raspberrypi:~/TerrariumPI/data $ cd..
-bash: cd..: Kommando nicht gefunden.
pi@raspberrypi:~/TerrariumPI/data $ cd
pi@raspberrypi:~ $ TerrariumPI
-bash: TerrariumPI: Kommando nicht gefunden.
pi@raspberrypi:~ $ cd TerrariumPI
pi@raspberrypi:~/TerrariumPI $ sudo service terrariumpi stop

cd
top
^C
pi@raspberrypi:~/TerrariumPI $ cd
pi@raspberrypi:~ $ sudo service terrariumpi stop

cd

^C
pi@raspberrypi:~ $ cd TerrariumPI
pi@raspberrypi:~/TerrariumPI $ source venv/bin/activate
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2024-06-08 12:01:58,671 - INFO    - terrariumEngine       - Starting up Terrariu                                     mPI 4.9.0 on a Raspberry Pi 3 Model B Rev 1.2 ...
2024-06-08 12:01:58,706 - INFO    - terrariumEngine       - Loaded 32 settings i                                     n 0.03 seconds.
2024-06-08 12:01:59,691 - INFO    - OpenWeatherMap        - Using Openweathermap                                      Free API
2024-06-08 12:01:59,694 - INFO    - terrariumWeather      - Loaded weather data                                      from source https://api.openweathermap.org/data/2.5/weather?q=Erding,DE&appid=de                                     7a1fbd5f74afc83284ce53b8c9b504 in 0.94 seconds
^CTraceback (most recent call last):
  File "/home/pi/TerrariumPI/terrariumPI.py", line 27, in <module>
    terrariumEngine = terrariumEngine(__version__)
  File "/home/pi/TerrariumPI/terrariumEngine.py", line 135, in __init__
    self.calendar = terrariumCalendar()
  File "/home/pi/TerrariumPI/terrariumCalendar.py", line 34, in __init__
    for event in self.get_events(
  File "/home/pi/TerrariumPI/terrariumCalendar.py", line 75, in get_events
    for event_data in events(string_content=self.__ical.to_ical(), start=start, end=end):
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/icalevents/icalevents.py", line 57, in events
    found_events += parse_events(
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/icalevents/icalparser.py", line 391, in parse_events
    if not event.recurrence_id and (event.uid, event.start) in [
  File "/home/pi/TerrariumPI/venv/lib/python3.9/site-packages/icalevents/icalparser.py", line 392, in <listcomp>
    (f.uid, f.recurrence_id) for f in found
KeyboardInterrupt

theyosh, 2024-06-08:
Hmm, this is strange... I get the feeling a broken SD card??

But first we can try the following. Rename the calender file so it will create a new one.

`mv /home/pi/TerrariumPI/data/calander.ics /home/pi/TerrariumPI/data/calander.ics.backup`

and restart in debug mode once again. Lets see what that does.

Konrad84, 2024-06-08:
Oh now it keeps running but with failure messages and error of the raspicam (but function was good before.. hmm

pi@raspberrypi:~ $ sudo service TerrariumPI stop
Failed to stop TerrariumPI.service: Unit TerrariumPI.service not loaded.
pi@raspberrypi:~ $ cd TerrariumPI
pi@raspberrypi:~/TerrariumPI $ source venv/bin/activate
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2024-06-08 20:32:49,793 - INFO    - terrariumEngine       - Starting up Terrariu                                     mPI 4.9.0 on a Raspberry Pi 3 Model B Rev 1.2 ...
2024-06-08 20:32:49,829 - INFO    - terrariumEngine       - Loaded 32 settings i                                     n 0.03 seconds.
2024-06-08 20:32:51,234 - INFO    - OpenWeatherMap        - Using Openweathermap                                      Free API
2024-06-08 20:32:51,238 - INFO    - terrariumWeather      - Loaded weather data                                      from source https://api.openweathermap.org/data/2.5/weather?q=Erding,DE&appid=de                                     7a1fbd5f74afc83284ce53b8c9b504 in 1.37 seconds
2024-06-08 20:32:52,341 - INFO    - terrariumEngine       - Start loading total                                      power and water usage
2024-06-08 20:32:54,529 - INFO    - terrariumEngine       - Loaded total power a                                     nd water usage in 2.19 seconds.
2024-06-08 20:32:54,531 - INFO    - terrariumEngine       - Loading existing sen                                     sors from database.
2024-06-08 20:32:55,558 - INFO    - terrariumSensor       - Loaded sensor 1-Wire                                      sensor temperature named 'Schildkrötenhaus' at address '28-3c01f09637c2' with v                                     alue 22.19C in 1.00 seconds.
2024-06-08 20:32:56,483 - INFO    - terrariumSensor       - Loaded sensor 1-Wire                                      sensor temperature named 'Außentemperatur' at address '28-3c01f096b371' with va                                     lue 18.31C in 0.92 seconds.
2024-06-08 20:32:57,436 - INFO    - terrariumSensor       - Loaded sensor 1-Wire                                      sensor temperature named 'Gewächshaus' at address '28-3c01f096fb7e' with value                                      20.81C in 0.95 seconds.
2024-06-08 20:32:58,397 - INFO    - terrariumSensor       - Loaded sensor 1-Wire                                      sensor temperature named 'Schildkröten Babys' at address '28-3de104572f5c' with                                      value 21.44C in 0.96 seconds.
2024-06-08 20:32:59,356 - INFO    - terrariumSensor       - Loaded sensor 1-Wire                                      sensor temperature named 'Schaltkasten (Reserve2)' at address '28-3de10457f6b5'                                      with value 20.06C in 0.96 seconds.
2024-06-08 20:32:59,364 - INFO    - terrariumEngine       - Scanning for new sen                                     sors ...
2024-06-08 20:33:13,413 - INFO    - terrariumEngine       - Loaded 5 sensors in                                      18.88 seconds.
2024-06-08 20:33:13,415 - INFO    - terrariumEngine       - Loading existing rel                                     ays from database.
2024-06-08 20:33:15,398 - INFO    - terrariumRelay        - Loaded relay gpio re                                     lay named 'DACH' at address '32' value 0.00 in 1.93 seconds.
2024-06-08 20:33:15,419 - INFO    - terrariumRelay        - Loaded relay gpio re                                     lay named 'TÜRE' at address '36' value 0.00 in 0.02 seconds.
2024-06-08 20:33:15,437 - INFO    - terrariumRelay        - Loaded relay gpio re                                     lay named 'HEIZEN' at address '38' value 0.00 in 0.02 seconds.
2024-06-08 20:33:15,455 - INFO    - terrariumRelay        - Loaded relay gpio re                                     lay named 'LICHT' at address '40' value 0.00 in 0.02 seconds.
2024-06-08 20:33:15,459 - INFO    - terrariumEngine       - Scanning for new rel                                     ays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2024-06-08 20:33:31,317 - INFO    - terrariumEngine       - Loaded 4 relays in 17.90 seconds.
2024-06-08 20:33:31,319 - INFO    - terrariumEngine       - Loading existing buttons from database.
2024-06-08 20:33:31,342 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.02 seconds.
2024-06-08 20:33:31,344 - INFO    - terrariumEngine       - Loading existing webcams from database.
2024-06-08 20:33:36,471 - WARNING - hardware.webcam       - Webcam Live Raspberry PI camera named 'Haus innen' at address 'rpicam_live' is crashed. Restarting the webcam.
2024-06-08 20:33:42,051 - WARNING - hardware.webcam       - Webcam Live Raspberry PI camera named 'Haus innen' at address 'rpicam_live' is crashed. Restarting the webcam.
2024-06-08 20:33:47,620 - WARNING - hardware.webcam       - Webcam Live Raspberry PI camera named 'Haus innen' at address 'rpicam_live' is crashed. Restarting the webcam.
2024-06-08 20:33:53,194 - ERROR   - hardware.webcam       - Webcam Live Raspberry PI camera named 'Haus innen' at address 'rpicam_live' has gone offline! Please check your webcam connections.
2024-06-08 20:33:53,908 - INFO    - terrariumWebcam       - Loaded rpicam-live webcam 'Haus innen' at address 'rpicam_live' in 22.54 seconds.
[... cut ...]
2024-06-08 20:38:25,098 - INFO    - terrariumEngine       - Stopped GPIO devices relay named 'LICHT' at address '40'
2024-06-08 20:38:25,100 - INFO    - terrariumEngine       - Stopped Live Raspberry PI camera named 'Haus innen' at address 'rpicam_live'
2024-06-08 20:38:25,100 - INFO    - terrariumEngine       - Stopped RTSP camera named 'Außengehege' at address 'rtsp://admin:admin@192.168.178.91:8554/Streaming/Channels/101'
2024-06-08 20:38:25,101 - INFO    - terrariumEngine       - Stopped TerrariumPI 4.9.0 after running for 0:05:36. Bye bye.

theyosh, 2024-06-08:
Hmm, ok, strange. But for now it will start.

For the webcam we can try to trouble shoot in a different issue.

Not sure why the telegram and the calendar file was causing issues.....

theyosh, 2024-06-09:
What you could do is a reboot of the PI. That could be the trick to start the webcam again. If that does not work for you, please open a new issue to trouble shout the webcam.

Konrad84, 2024-06-09:
Hi yes it works now but i decided to upgrade to 4.9.1. now .. maybe there are additional issues with the installation i get several hints like this .. 

[image]

Konrad84, 2024-06-09:
Maybe tr a new SD .. Hmm ..

Konrad84, 2024-06-09:
ok upgrade worked but now all relays are gone and scan says 0 found. No Cameras. Sensors are good i think.

Have you an idea what i can do or have i get a new setup SD and maybe Raspberry?

Konrad84, 2024-06-09:
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2024-06-09 12:44:01,506 - INFO    - terrariumEngine       - Starting up Terrariu                                     mPI 4.9.1 on a Raspberry Pi 3 Model B Rev 1.2 ...
2024-06-09 12:44:01,541 - INFO    - terrariumEngine       - Loaded 32 settings i                                     n 0.03 seconds.
2024-06-09 12:44:02,668 - INFO    - terrariumEngine       - Start loading total                                      power and water usage
2024-06-09 12:44:03,705 - INFO    - terrariumEngine       - Loaded total power a                                     nd water usage in 1.04 seconds.
2024-06-09 12:44:03,707 - INFO    - terrariumEngine       - Loading existing sen                                     sors from database.
2024-06-09 12:44:04,753 - INFO    - terrariumSensor       - Loaded sensor 1-Wire                                      sensor temperature named '1-Wire sensor measuring temperature' at address '28-3                                     c01f09637c2' with value 21.69C in 1.02 seconds.
2024-06-09 12:44:05,673 - INFO    - terrariumSensor       - Loaded sensor 1-Wire                                      sensor temperature named '1-Wire sensor measuring temperature' at address '28-3                                     c01f096b371' with value 21.38C in 0.92 seconds.
2024-06-09 12:44:06,631 - INFO    - terrariumSensor       - Loaded sensor 1-Wire                                      sensor temperature named '1-Wire sensor measuring temperature' at address '28-3                                     c01f096fb7e' with value 21.94C in 0.96 seconds.
2024-06-09 12:44:07,591 - INFO    - terrariumSensor       - Loaded sensor 1-Wire                                      sensor temperature named '1-Wire sensor measuring temperature' at address '28-3                                     de104572f5c' with value 30.88C in 0.96 seconds.
2024-06-09 12:44:08,551 - INFO    - terrariumSensor       - Loaded sensor 1-Wire                                      sensor temperature named '1-Wire sensor measuring temperature' at address '28-3                                     de10457f6b5' with value 27.19C in 0.96 seconds.
2024-06-09 12:44:08,556 - INFO    - terrariumEngine       - Scanning for new sen                                     sors ...
2024-06-09 12:44:22,643 - INFO    - terrariumEngine       - Loaded 5 sensors in 18.94 seconds.
2024-06-09 12:44:22,645 - INFO    - terrariumEngine       - Loading existing relays from database.
2024-06-09 12:44:22,663 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2024-06-09 12:44:39,614 - INFO    - terrariumEngine       - Loaded 0 relays in 16.97 seconds.
2024-06-09 12:44:39,616 - INFO    - terrariumEngine       - Loading existing buttons from database.
2024-06-09 12:44:39,640 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.02 seconds.
2024-06-09 12:44:39,642 - INFO    - terrariumEngine       - Loading existing webcams from database.
2024-06-09 12:44:39,672 - INFO    - terrariumEngine       - Loaded 0 webcams in 0.03 seconds.
2024-06-09 12:44:39,674 - INFO    - terrariumEngine       - Loading existing enclosures from database.
2024-06-09 12:44:39,687 - INFO    - terrariumEngine       - Loaded 0 enclosures in 0.01 seconds.
2024-06-09 12:44:39,849 - INFO    - terrariumEngine       - Message of the Day generated in 0.16 seconds
2024-06-09 12:44:39,857 - INFO    - terrariumEngine       - TerrariumPI 4.9.1 is up and running at address: http://0.0.0.0:8090 in 39.92 seconds.
2024-06-09 12:44:39,000 - INFO    - terrariumWebserver    - Running webserver at 0.0.0.0:8090

theyosh, 2024-06-09:
> Hi yes it works now but i decided to upgrade to 4.9.1. now .. maybe there are additional issues with the installation i get several hints like this .. [...]

Hmm, you have German language and locale in use... That could explain a bit.

Konrad84, 2024-06-10:
Hmm what do you mean about language. Should i Change Something ?

theyosh, 2024-06-10:
> Hmm what do you mean about language. Should i Change Something ?

Nope, but in German you use a comma for decimals, and I used a dot. So than it does not work. So I made the code now language aware. And now the strange error during installation should be fixed.

So you do not have to change anything. But I did not test with a different language which uses different decimal notation.

theyosh, 2024-06-12:
Everything is still running?

Konrad84, 2024-06-12:
Hi Just Sensors are running camera and Relais are offline /deleted. Maybe i Just need to set things Up new.. but i think WE can Close this issue and i create a new one for the cameras If i cant fix it ;) thank you

closed 2024-06-14 by theyosh