theyosh/TerrariumPI #924: [BUG]: Camera and Relays not worling after Upgrade and Titles of Sensors erased

Konrad84, 2024-06-20:
Hi there now after you solved my issue with telegram after the upgrade Camera isn´t working and there are no relais found anymore. Titles of sensores are erased. I upgraded again to 4.9.3 but still doesnt work. Have you got any solution or hint where to search ? Here ist the log ... 

                  |_   _|        | | | |     | ___ \_   _|
                     | |_   _ _ __| |_| | ___ | |_/ / | |
                     | | | | | '__| __| |/ _ \|  __/  | |
                     | | |_| | |  | |_| |  __/| |    _| |_
                     \_/\__,_|_|   \__|_|\___|\_|    \___/

                                                Version: 4.9.3

  Durchschnitt temperatur: 25.12 C  Betriebszeit:     11 days, 9:13:27
                                    Systemauslastung: 0.47, 0.53, 0.48
                                    CPU Temperatur:   60.148 C
                                    Speicher:         7.1 GiB(24.60%) used of total 29.0 GiB
                                    RAM:              247.5 MiB(28.42%) used of total 870.8 MiB

  Aktuelle Tätigkeit (0/0)  0/0.00 Watt, 0/0.00 L/m

      Letztes update: Thursday, 20-06-2024 22:02:43

Linux raspberrypi 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Jun 20 21:15:05 2024 from 192.168.178.24
pi@raspberrypi:~ $ sudo service terrariumpi stop
pi@raspberrypi:~ $ cd TerrariumPI
pi@raspberrypi:~/TerrariumPI $ source venv/bin/activate
(venv) pi@raspberrypi:~/TerrariumPI $ python terrariumPI.py
2024-06-20 22:04:41,737 - INFO    - terrariumEngine       - Starting up TerrariumPI 4.9.3 on a Raspberry Pi 3 Model B Rev 1.2 ...
2024-06-20 22:04:41,773 - INFO    - terrariumEngine       - Loaded 32 settings in 0.03 seconds.
2024-06-20 22:04:42,916 - INFO    - terrariumEngine       - Start loading total power and water usage
2024-06-20 22:04:43,954 - INFO    - terrariumEngine       - Loaded total power and water usage in 1.04 seconds.
2024-06-20 22:04:43,955 - INFO    - terrariumEngine       - Loading existing sensors from database.
2024-06-20 22:04:44,984 - INFO    - terrariumSensor       - Loaded sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3c01f09637c2' with value 25.50C in 1.00 seconds.
2024-06-20 22:04:45,910 - INFO    - terrariumSensor       - Loaded sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3c01f096b371' with value 24.69C in 0.92 seconds.
2024-06-20 22:04:46,867 - INFO    - terrariumSensor       - Loaded sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3c01f096fb7e' with value 24.88C in 0.96 seconds.
2024-06-20 22:04:47,827 - INFO    - terrariumSensor       - Loaded sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3de104572f5c' with value 24.81C in 0.96 seconds.
2024-06-20 22:04:48,787 - INFO    - terrariumSensor       - Loaded sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3de10457f6b5' with value 25.25C in 0.96 seconds.
2024-06-20 22:04:48,791 - INFO    - terrariumEngine       - Scanning for new sensors ...
2024-06-20 22:05:02,887 - INFO    - terrariumEngine       - Loaded 5 sensors in 18.93 seconds.
2024-06-20 22:05:02,889 - INFO    - terrariumEngine       - Loading existing relays from database.
2024-06-20 22:05:02,912 - INFO    - terrariumEngine       - Scanning for new relays ...
No GEMBIRD SiS-PM found. Check USB connections, please!
2024-06-20 22:05:19,886 - INFO    - terrariumEngine       - Loaded 0 relays in 17.00 seconds.
2024-06-20 22:05:19,888 - INFO    - terrariumEngine       - Loading existing buttons from database.
2024-06-20 22:05:19,910 - INFO    - terrariumEngine       - Loaded 0 buttons in 0.02 seconds.
2024-06-20 22:05:19,911 - INFO    - terrariumEngine       - Loading existing webcams from database.
2024-06-20 22:05:30,300 - INFO    - terrariumWebcam       - Loaded rpicam webcam 'Schildkröten Haus' at address 'rpicam' in 10.36 seconds.
2024-06-20 22:05:30,302 - INFO    - terrariumEngine       - Loaded 1 webcams in 10.39 seconds.
2024-06-20 22:05:30,303 - INFO    - terrariumEngine       - Loading existing enclosures from database.
2024-06-20 22:05:30,311 - INFO    - terrariumEngine       - Loaded 0 enclosures in 0.01 seconds.
2024-06-20 22:05:30,468 - INFO    - terrariumEngine       - Message of the Day generated in 0.16 seconds
2024-06-20 22:05:30,475 - INFO    - terrariumEngine       - TerrariumPI 4.9.3 is up and running at address: http://0.0.0.0:8090 in 50.31 seconds.
2024-06-20 22:05:30,000 - INFO    - terrariumWebserver    - Running webserver at 0.0.0.0:8090
2024-06-20 22:05:35,033 - WARNING - hardware.sensor       - Error getting new data from sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3de10457f6b5': no data, retrying in 0.5 seconds...
2024-06-20 22:06:35,056 - WARNING - hardware.sensor       - Error getting new data from sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3de10457f6b5': no data, retrying in 0.5 seconds...
2024-06-20 22:07:35,047 - WARNING - hardware.sensor       - Error getting new data from sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3de10457f6b5': no data, retrying in 0.5 seconds...
2024-06-20 22:08:35,053 - WARNING - hardware.sensor       - Error getting new data from sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3de10457f6b5': no data, retrying in 0.5 seconds...
2024-06-20 22:09:35,047 - WARNING - hardware.sensor       - Error getting new data from sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3de10457f6b5': no data, retrying in 0.5 seconds...
2024-06-20 22:10:35,056 - WARNING - hardware.sensor       - Error getting new data from sensor 1-Wire sensor temperature named '1-Wire sensor measuring temperature' at address '28-3de10457f6b5': no data, retrying in 0.5 seconds...

Now there is a Error from 1-Wire too but this worked and i changed nothing only updated ... hmm

greets Konrad 

Thank You

theyosh, 2024-06-24:
Hi sorry for the slow response. But this is strange, and hard to debug/fix. But it looks like only 1 sensor is complaining: '28-3de10457f6b5'

I have seen this also with my setup. That 1 1-wire sensor is broken. This is probably due to a bad connection. In my case. I know that I made a bad solder joint between two cables, and that joint is now laying on the bottom of my terrarium, probably in water. So for me, that means, 1 sensor just does not work, due to a broken cable. And I can't fix it, as I need to rebuild my terrarium.

About the 1-wire sensors, I am depending on OS support / integration. So I guess that there are some messages from the 1-wire system. You can check this by running: `grep w1 /var/log/kern.log`
This will probably produce some output. Paste it here.

And you should see some directories at '/sys/bus/w1/devices/' So when you list the contents of that folder what do you see?:
`ls -l /sys/bus/w1/devices/` What kind of list do you get?

It can also be, that somehow the Pi is overloaded, and therefore dropping some power resources. As I have seen that Bluetooth become a bit more unstable when the CPU is doing a lot of stuff.

And if the sensor is broken for what ever reason, you can put it on the ignore list in TerrariumPI, and that sensor will not be used anymore. You will lose a sensor, but the system will just continue without errors.

About the camera issues, I see nothing wrong in the log file. There it looks like the camera is working.

Konrad84, 2024-06-26:
Hi there thank you for the help. I set up a new Raspian Legacy and new SD Card now it all works :)

Can you just help with the rtsp stream? I got an Onvif camera, an rtsp camera and an usb camera.

i tried few links but no response like :

rtsp://[ipadress:port]/stream 
rtsp://[ipadress:port]/onvif/stream 
rtsp://[ipadress]/stream 

and how do i get the adress of de usb camera ?

Soon i will do some pictures of the turtles :)

theyosh, 2024-06-27:
I am not familiar with Onvif camera's. I see some information here about getting an RTSP url, but that is rather complicated when you are not that experienced on Linux: https://superuser.com/a/1711576
Also, I see this app: https://www.ipcent.com/mobile/onvifer Maybe that can show your camera in your network, and can give an RTSP url back. Not sure.
If you have found the RTSP stream, you can test it by using VLC https://www.thewindowsclub.com/how-to-play-rtsp-stream-in-vlc-media-player

For the USB camera, there should be something like `/dev/videoX` where X is a number. You can list them with `ls -l /dev/video*`
/dev/video0 is the Raspberry PI camera when added. Else I expect that you only have /dev/video1X devices, which are not camera devices TP can use. So the usb address should be something like: `/dev/video0` or `/dev/video1` I guess.

theyosh, 2024-07-10:
Any updates? Did you get the cameras working?

manolof, 2024-07-15:
> Did you get the cameras working?

Not sure if related, but my rpicam_live is crashing and not working. In fact, it just shows a live feed from the time that it last worked, which is around 8 days ago.
In the logs it shows:
```
2024-07-15 18:17:46,697 - WARNING - hardware.webcam       - Webcam Live Raspberry PI camera named 'Camera1' at address 'rpicam_live' is crashed. Restarting the webcam.
2024-07-15 18:17:40,983 - WARNING - hardware.webcam       - Webcam Live Raspberry PI camera named 'Camera1' at address 'rpicam_live' is crashed. Restarting the webcam.
2024-07-15 18:17:27,577 - INFO    - terrariumWebcam       - Updated webcam rpicam-live webcam 'Camera1' at address 'rpicam_live' in 16.63 seconds.
```

and here's the live feed:
[image]

I am using 4.10.0, Raspberry PI 4B

theyosh, 2024-07-15:
Hmm, could you run the following command through ssh on the console:

`ps fax` and show all the output here? I guess that somehow the camera is already being used..

manolof, 2024-07-15:
```
USER         PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
root           1  0.0  1.0 168644  9528 ?        Ss   Jul07   0:57 /sbin/init
root           2  0.0  0.0      0     0 ?        S    Jul07   0:01 [kthreadd]
root           3  0.0  0.0      0     0 ?        I<   Jul07   0:00 [rcu_gp]
root           4  0.0  0.0      0     0 ?        I<   Jul07   0:00 [rcu_par_gp]
root           5  0.0  0.0      0     0 ?        I<   Jul07   0:00 [slub_flushwq]
root           6  0.0  0.0      0     0 ?        I<   Jul07   0:00 [netns]
root          10  0.0  0.0      0     0 ?        I<   Jul07   0:00 [mm_percpu_wq]
root          11  0.0  0.0      0     0 ?        I    Jul07   0:00 [rcu_tasks_kthread]
root          12  0.0  0.0      0     0 ?        I    Jul07   0:00 [rcu_tasks_rude_kthread]
root          13  0.0  0.0      0     0 ?        I    Jul07   0:00 [rcu_tasks_trace_kthread]
root          14  0.0  0.0      0     0 ?        S    Jul07   0:28 [ksoftirqd/0]
root          15  0.0  0.0      0     0 ?        I    Jul07   1:47 [rcu_preempt]
root          16  0.0  0.0      0     0 ?        S    Jul07   0:00 [migration/0]
root          17  0.0  0.0      0     0 ?        S    Jul07   0:00 [cpuhp/0]
root          18  0.0  0.0      0     0 ?        S    Jul07   0:00 [cpuhp/1]
root          19  0.0  0.0      0     0 ?        S    Jul07   0:00 [migration/1]
root          20  0.0  0.0      0     0 ?        S    Jul07   0:30 [ksoftirqd/1]
root          23  0.0  0.0      0     0 ?        S    Jul07   0:00 [cpuhp/2]
root          24  0.0  0.0      0     0 ?        S    Jul07   0:00 [migration/2]
root          25  0.0  0.0      0     0 ?        S    Jul07   0:30 [ksoftirqd/2]
root          28  0.0  0.0      0     0 ?        S    Jul07   0:00 [cpuhp/3]
root          29  0.0  0.0      0     0 ?        S    Jul07   0:00 [migration/3]
root          30  0.0  0.0      0     0 ?        S    Jul07   0:25 [ksoftirqd/3]
root          33  0.0  0.0      0     0 ?        S    Jul07   0:00 [kdevtmpfs]
root          34  0.0  0.0      0     0 ?        I<   Jul07   0:00 [inet_frag_wq]
root          36  0.0  0.0      0     0 ?        S    Jul07   0:00 [kauditd]
root          38  0.0  0.0      0     0 ?        S    Jul07   0:00 [khungtaskd]
root          40  0.0  0.0      0     0 ?        S    Jul07   0:00 [oom_reaper]
root          41  0.0  0.0      0     0 ?        I<   Jul07   0:00 [writeback]
root          42  0.0  0.0      0     0 ?        S    Jul07   0:19 [kcompactd0]
root          43  0.0  0.0      0     0 ?        I<   Jul07   0:00 [kintegrityd]
root          44  0.0  0.0      0     0 ?        I<   Jul07   0:00 [kblockd]
root          45  0.0  0.0      0     0 ?        I<   Jul07   0:00 [blkcg_punt_bio]
root          46  0.0  0.0      0     0 ?        S    Jul07   0:00 [watchdogd]
root          50  0.0  0.0      0     0 ?        I<   Jul07   0:00 [rpciod]
root          51  0.0  0.0      0     0 ?        I<   Jul07   0:00 [xprtiod]
root          55  0.0  0.0      0     0 ?        S    Jul07   0:02 [kswapd0]
root          56  0.0  0.0      0     0 ?        I<   Jul07   0:00 [nfsiod]
root          57  0.0  0.0      0     0 ?        I<   Jul07   0:00 [kthrotld]
root          63  0.0  0.0      0     0 ?        S    Jul07   0:00 [irq/30-aerdrv]
root          65  0.0  0.0      0     0 ?        S    Jul07   0:00 [hwrng]
root          66  0.0  0.0      0     0 ?        I<   Jul07   0:00 [iscsi_conn_clea]
root          67  0.0  0.0      0     0 ?        I<   Jul07   0:00 [nvme-wq]
root          68  0.0  0.0      0     0 ?        I<   Jul07   0:00 [nvme-reset-wq]
root          69  0.0  0.0      0     0 ?        I<   Jul07   0:00 [nvme-delete-wq]
root          71  0.0  0.0      0     0 ?        I<   Jul07   0:00 [DWC Notificatio]
root          72  0.0  0.0      0     0 ?        I<   Jul07   0:00 [uas]
root          73  0.0  0.0      0     0 ?        S<   Jul07   0:32 [vchiq-slot/0]
root          74  0.0  0.0      0     0 ?        S<   Jul07   0:00 [vchiq-recy/0]
root          75  0.0  0.0      0     0 ?        S<   Jul07   0:00 [vchiq-sync/0]
root          82  0.0  0.0      0     0 ?        I<   Jul07   0:00 [zswap-shrink]
root         103  0.0  0.0      0     0 ?        I<   Jul07   0:00 [sdhci]
root         104  0.0  0.0      0     0 ?        S    Jul07   0:00 [irq/37-mmc0]
root         129  0.0  0.0      0     0 ?        I<   Jul07   0:00 [mmc_complete]
root         133  0.0  0.0      0     0 ?        I<   Jul07   0:07 [kworker/3:1H-kblockd]
root         146  0.0  0.0      0     0 ?        I<   Jul07   0:08 [kworker/1:2H-kblockd]
root         193  0.0  0.0      0     0 ?        S    Jul07   0:25 [jbd2/mmcblk0p2-8]
root         194  0.0  0.0      0     0 ?        I<   Jul07   0:00 [ext4-rsv-conver]
root         208  0.0  0.0      0     0 ?        I<   Jul07   0:00 [mld]
root         209  0.0  0.0      0     0 ?        I<   Jul07   0:00 [ipv6_addrconf]
root         248  0.0  8.2 115416 72356 ?        Ss   Jul07   0:50 /lib/systemd/systemd-journald
root         273  0.0  0.6  25848  5880 ?        Ss   Jul07   7:34 /lib/systemd/systemd-udevd
root         312  8.0  0.0      0     0 ?        S    Jul07 958:34 [w1_bus_master1]
root         315  0.0  0.0      0     0 ?        S    Jul07   0:00 [vchiq-keep/0]
root         319  0.0  0.0      0     0 ?        S<   Jul07   0:00 [SMIO]
root         338  0.0  0.0      0     0 ?        I<   Jul07   0:00 [mmal-vchiq]
[... cut ...]
manos    1809797  0.5  1.1  19448 10200 ?        Ss   19:38   0:00 /lib/systemd/systemd --user
root     1809798  0.0  0.0      0     0 ?        I    19:38   0:00 [kworker/1:3-events_freezable_power_]
manos    1809799  0.0  0.4 170404  4172 ?        S    19:38   0:00 (sd-pam)
root     1809807  0.0  0.0      0     0 ?        I<   19:38   0:00 [kworker/0:1H]
root     1809979  0.4  0.3  10640  3316 pts/0    S<s  19:38   0:00 login
manos    1809997  3.2  0.5   8104  4804 pts/0    S<   19:38   0:00 -bash
manos    1810016  0.0  0.4  11148  4072 pts/0    R<+  19:38   0:00 ps aux
```

theyosh, 2024-07-15:
@manolof you are running in Docker. For how long do you have this issue? And what was the last Docker image which worked? 

As the code for the live webcam is updated 5 months ago, and that should not produce the black box with name and timestamp anymore. So it is a bit unclear from which version you are upgrading from.

manolof, 2024-07-16:
@theyosh last working version was 4.8.4, and I have the issue since last time I upgraded due to https://github.com/theyosh/TerrariumPI/issues/911, which was around July 7 (same time the cameras last recorded).

Konrad84, 2024-07-17:
Hi there and sorry for the late answer. I now get a new SD Card with the latest Version of TerrariumPI installed.

Temperatur, Humidity and Relais are there. Rasperry Live Cam stays Blackscreen and Failure Message camera is offline. I think i have to check the connection first.

Till now i get my Telegram not working because bot says i have no rights if i want to test it . Maybe i have to set up it new.

For the Cam i check connection and then i can say if it is solved.

:)

theyosh, 2024-07-17:
The Live RPI camera software is broken. So leave you hardware. I am working on a fix.

theyosh, 2024-07-17:
@Konrad84 and @manolof There is a new release, which should fix the live webcam for docker and non docker setups.

So update your setup, and I guess that the webcam will start up again.

theyosh, 2024-07-17:
@Konrad84 About telegram, you need yo enter your account handle, I am not sure if that includes an '@' in the name.
https://theyosh.github.io/TerrariumPI/notifications/#telegram

Konrad84, 2024-07-18:
Hi there now everything works fine :-D

Tank You

Konrad84, 2024-07-18:
Telegram is name without @ was wrong 👀

theyosh, 2024-07-18:
Just to be sure: My documentation is wrong. You do **not** enter the `@` of your username

Konrad84, 2024-07-18:
Yes just the telegram handle without @

theyosh, 2024-07-26:
Any updates on this from either of you? Can I close this issue?

manolof, 2024-07-27:
@theyosh the camera issue is fixed; thanks!

closed 2024-07-29 by theyosh