theyosh/TerrariumPI #925: [BUG]: Clear the screen between messages for 20x4 LCD display

MakoShark2, 2024-07-21:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.10.2
 - Install: [Docker]
 - Raspberry PI: 4B
 - OS: Bookworm

**Describe the bug**
20x4 I2C LCD attached to the PI which receives notifications for sensor_update of 4 sensors (2 temperature + 2 humidity). Everything works fine except it seems the LCD screen is not cleared between messages. So for example if one message spans to the second line but the next one fits in one, the remaining second line of the previous message will still show.

**To Reproduce / Installation steps**
4 sensors connected (2 temperature + 2 humidity) connected to the PI 4B.
20x4 I2C LCD connected to the PI 4B.
20x4 LCD I2C display added to the notification service.
One notification message added,sensor_update, unlimited rate, message: ${name} ${value:.1f}${unit}

**Expected behavior**
LCD screen cleared before printing the notification message.
Or better, being this LCD 4 lines it would be nice if the message would scroll instead of printing every message always in the first line.

**Screenshots**
Don't have but I can get some if needed.

**Logfiles**
Don't have.

**Additional context**
None.

theyosh, 2024-07-23:
Hmm, as far I can judge, the screen should be cleared for every message. At the moment I do not have my display connected. So I can't test right now.

So it can take a bit more time to look into this.

MakoShark2, 2024-07-23:
OK, thanks.

I've record a video and made it available here: https://drive.google.com/file/d/1jQT8IiuSHqNTu3My4-diE4d0XW0J6hvp/view?usp=drive_link

At this time I have 4 sensors added named:
IN Temperature
IN Humidity
OUT Temperature
OUT Humidity

Hope this helps.

theyosh, 2024-07-26:
I think I have fixed this. If you try a development image, it should work now better. https://hub.docker.com/r/theyosh/terrariumpi/tags?page=&page_size=&ordering=&name=development

Could you try it out and see if it works better now.

MakoShark2, 2024-07-26:
Sure will do and let you know. Thanks!

MakoShark2, 2024-07-26:
Not sure if it's me or anything else but the image is not starting, stuck on:

ModuleNotFoundError: No module named 'cv2'
Traceback (most recent call last):
  File "/TerrariumPI/terrariumPI.py", line 24, in <module>
    from terrariumEngine import terrariumEngine
  File "/TerrariumPI/terrariumEngine.py", line 2, in <module>
    from hardware.webcam.rpilive_webcam import terrariumRPILiveWebcam
  File "/TerrariumPI/hardware/webcam/__init__.py", line 25, in <module>
    import cv2

theyosh, 2024-07-26:
which development image did you use?

MakoShark2, 2024-07-26:
development-bookworm

I'm running a RPI 4B with:
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian

theyosh, 2024-07-26:
But the 4.10.2 bookworm does work? That is strange...  there should not be that big changes that could cause this.

But for how long are you running on Bookworm? Because I did not fully test it... so I do not know how stable that is.

In the mean time I am thinking on how to test this. I suggest to go back to the latest working version for now.

MakoShark2, 2024-07-26:
I've been running 4.10.2 bookworm it since it's release with no issues.

The development branch just kept restarting and outputing that error on the log.

I've run the development branch just by replacing the version :**latest** with :**development-bookworm** in the yaml file, is that the correct way to do it?

Meanwhile I went back to 4.10.2.

theyosh, 2024-07-26:
Hmm, the `:latest` is not bookworm, but buster image. So than you can use `:development-buster` as image. 
I think that should do the trick.

MakoShark2, 2024-07-26:
Ah... of course you are right, my mistake... Up and running and it works beautifully, thanks :)

I'll record a video tomorrow and will post it here.

MakoShark2, 2024-07-27:
Here's the video. I've shortened the sensor names so it can fit on one line and it just looks perfect now, thanks.
https://drive.google.com/file/d/1udSrz46v_GnaDimJYdu4dqNIQbBVodGo/view?usp=drive_link

theyosh, 2024-07-27:
Looks better! Cool

theyosh, 2024-07-29:
You can now go back to the `:latest` tag, and the display should stay the same. I created a new release.

closed 2024-08-06 by theyosh