theyosh/TerrariumPI #926: [FEATURE]: Modify "mail_from" via GUI settings

manolof, 2024-07-21:
**Is your feature request related to a problem? Please describe.**

It would be very useful to be able to modify the email notifications sender's ("From") address.
Currently, it just takes the receiver address and adds the "+terrariumpi" to the first part of the URL.
https://github.com/theyosh/TerrariumPI/blob/b5d616285ea9302ce57432178468a30cd8a38f06/terrariumNotification.py#L556

The problem is that, some email providers like Gmail recognize this as a potential issue:

[image]

which makes services like Mailgun, or Smtp2Go fail sporadically.

**Describe the solution you'd like**

An option in the GUI to specify the "From" address.

theyosh, 2024-07-22:
I have updated the code. But you are using docker, so that is a bit harder to test. If you are able/wanting to test this, I can create a development version.

manolof, 2024-07-22:
> I have updated the code. But you are using docker, so that is a bit harder to test. If you are able/wanting to test this, I can create a development version.

I'd be up for it, towards end of the week or beginning of next one though.

Thanks for the quick work on this.

theyosh, 2024-07-23:
Image are being build: https://hub.docker.com/repository/docker/theyosh/terrariumpi/tags?page=&page_size=&ordering=&name=development

manolof, 2024-07-29:
@theyosh thanks, it works!
However, not sure if this is related but, the camera stopped working again.

theyosh, 2024-07-29:
More about the camera? I just created a new release, thought all was fixed

manolof, 2024-07-29:
I was using Bookworm and v4.10.2 on a Raspberry Pi 4 Model B Rev 1.5.
I used development-bookworm.
I will try to downgrade to the stable 4.10.2 and see what happens with the camera.

theyosh, 2024-07-29:
Ok, you could also try 4.11.0 in a few seconds. That is the latest release. 
I will look into this later today.

manolof, 2024-07-29:
I upgraded and all good, we are back online 😃 Thanks!