theyosh/TerrariumPI #927: [FEATURE]: support for Meross MS100FH

rossisbudda, 2024-07-23:
Hello.
You kindly added support for Meross MS100H thermo-hygrometer.
https://www.meross.com/en-gc/smart-sensor/wifi-temperature-sensor/51
Would you know if the same support will work for this variant, the MS100**F**H?
https://www.meross.com/en-gc/smart-sensor/smart-humidity-sensor/143
This one seems much nicer as it charges by light source, so no battery worries :) Also ms100H seems rare compared to ms100FH (at least in UK).
Many thanks.

theyosh, 2024-07-23:
Just did :)

If you are running from source, you can use it by doing a git pull.

For docker, wait a bit more when the development builds are ready

theyosh, 2024-07-26:
Are you able to test this? Do you already have this sensor at home? Would like to know for sure that it works. I cannot test it.

rossisbudda, 2024-07-26:
No, but I was wondering if you might like to leave this ticket open so that we can confirm it. I’ll buy one now as I wanted one anyway, especially now that you’ve confirmed it should work with terrariumPi.

rossisbudda, 2024-07-29:
Hello. It works! No problem whatsoever. Thanks!!!

closed 2024-07-29 by rossisbudda

theyosh, 2024-07-29:
Thanks for the feedback!