theyosh/TerrariumPI #928: [BUG]: Invalid configuration of calendar.ics

Awakawaka, 2024-07-26:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.10.2
 - Install: [Manual] 
 - Raspberry PI: Zero 2w
 - OS: Bullseye

**Describe the bug**
[BUG]: Invalid configuration of calendar.isc

**To Reproduce / Installation steps**
Today, I’ve noticed that Terraium Pi have stopped working. After several hours of investigation I’ve found out that calendar.ics file is not configured properly. It lacked ending directives. Was able to found out by running in debug mode.

**Expected behavior**
This is not happening:)

**Screenshots**
Please add screenshots of the [debug screens](https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/):

[image]

**Logfiles**
Please add the log file found in the log folder: `log/terrariumpi.log`. All sensitive information should be stripped, and save to upload.

**Additional context**
Add any other context about the problem here.

Awakawaka, 2024-07-26:
After fixing calendar.ics file, was able to start everything

theyosh, 2024-07-26:
What did you fix? I can't see what was wrong based on the screenshot.

Because, you can also just delete the calendar file if you do not use it. It will be recreated after a restart.

Awakawaka, 2024-07-26:
The file was missing the following lines at the very end of the file:
 END:VEVENT
 END:VCALENDAR

theyosh, 2024-07-26:
That is strange.... That should just be there. Did you by any change added an event to the calendar?

The only thing I can think of is that saving the calendar to disk did not work well when adding/updating an event. But I have not had this issue before.

Awakawaka, 2024-07-26:
No, I’ve done nothing:(
Maybe it is because of ungraceful shutdown. Will see. Also while trying to resolve the issue I updated the version from 4.6.2 to 4.10.2. And I no longer see “reversed GPIO Relay” but the behavior is still invalid from what I can say

theyosh, 2024-07-26:
Your GPIO relays are now normal GPIO relays, but you can use the calibration option now to inverse those. That should make it work again.

theyosh, 2024-07-26:
Make sure you clear the browser cache to when loading the gui after an upgrade

Awakawaka, 2024-07-26:
Where can I found calibration option?

theyosh, 2024-07-26:
[image]

Awakawaka, 2024-07-26:
Ooooh okay got it, thank you for the support!

Awakawaka, 2024-07-26:
Everything works as expected. Closing the ticket.
Thanks!

closed 2024-07-26 by Awakawaka