theyosh/TerrariumPI #931: [BUG]: Emails being sent twice

manolof, 2024-07-31:
**Setup:**
- TerrariumPI version: 4.11.0
- Install: Docker
- Raspberry PI: 4B Rev 1.5
- OS: Bookworm

**Describe the bug**
After the recent upgrade to 4.11.0, the email notifications service is sending the emails twice, the second time being 30 seconds after the first.
My email/SMTP provider that i am using for sending these emails, is receiving the same email twice.
Could it be that the email sending loop is not closing after sending the first?
https://github.com/theyosh/TerrariumPI/blob/212a9d7ed668a332afa40d61fbf09c6a8e739d0e/terrariumNotification.py#L600C17-L602C38

**Screenshots**

**1st email (normal)**

[image]

```
Received: from [10.208.238.136] (helo=terrariumpi)
	by smtpcorp.com with esmtpsa (TLS1.3:ECDHE_X25519__RSA_PSS_RSAE_SHA256__AES_256_GCM:256)
	(Exim 4.97.1-S2G)
	(envelope-from <XXX@XXX.com>)
	id 1sZ3Pu-4o5NDgrkLAD-m7Qe
	for XXX@XXX.com;
	Wed, 31 Jul 2024 07:06:34 +0000
Content-Type: multipart/mixed; boundary="===============0472279371=="
MIME-Version: 1.0
Date: Wed, 31 Jul 2024 09:06:33 +0200
X-Mailer: TerrariumPI version 4.11.0
Subject: Relay toggle
From: TerrariumPI <XXX@XXX.com>
To: "XXX@XXX.com" <XXX@XXX.com>
Message-Id: <E1sZ3Pu-4o5NDgrkLAD-m7Qe@message-id.smtpcorp.com>
```

**2nd email (duplicate)**

[image]
```
Received: from [10.208.238.136] (helo=terrariumpi)
	by smtpcorp.com with esmtpsa (TLS1.3:ECDHE_X25519__RSA_PSS_RSAE_SHA256__AES_256_GCM:256)
	(Exim 4.97.1-S2G)
	(envelope-from <XXX@XXX.com>)
	id 1sZ3QO-4o5NDgrkM1D-levE
	for XXX@XXX.com;
	Wed, 31 Jul 2024 07:07:04 +0000
Content-Type: multipart/mixed; boundary="===============1743467875=="
MIME-Version: 1.0
Date: Wed, 31 Jul 2024 09:07:03 +0200
X-Mailer: TerrariumPI version 4.11.0
Subject: Relay toggle
From: TerrariumPI <XXX@XXX.com>
To: "XXX@XXX.com" <XXX@XXX.com>
Message-Id: <E1sZ3QO-4o5NDgrkM1D-levE@message-id.smtpcorp.com>
```

**Logfiles**
I noticed that in the log files there is nothing being logged regarding notifications.

theyosh, 2024-07-31:
Hmm, I guess that there will be a notification of toggling the relay twice with a 30 sec interval? I think this has to do with the timer based on weather for the main lights?

manolof, 2024-08-06:
I think it works as intended after all. It's not a bug, it's a feature 😄 
I am toggling my relay on, wait 30 seconds, and toggle it off. The emails I get are for both on and off toggles.
Not sure why, but in the past I used to receive only 1 notification.

closed 2024-08-06 by manolof