theyosh/TerrariumPI #933: [BUG]: Missing Sensirion SHTxx Sensor

Mavajax, 2024-08-02:
**Setup:**
 - TerrariumPI version: 4.11.0
 - Install: Manual
 - Raspberry PI:  4B
 - Python 3.9.2
 - OS Bullseye

**Describe the bug**
Updated from 4.10.0 to 4.11.0 and it wont start. After deleting the SHTxx sensor it does start but I cant add the sensor again because it is missing in the list.

**To Reproduce / Installation steps**
Downgraded to 4.10.0, added the sensor, after that updated to 4.11.0 and it wont start.

**Expected behavior**
Being able to use the SHTxx sensor with version 4.11.0

**Screenshots**
[image]

theyosh, 2024-08-02:
I guess there should be a warning/error earlier in the log indicating not able to load a file. As I do have this sensor in the list of available sensors when I want to add it. So for me it works like expected.
Look at the logfile. There could be a warning message within the first 10 lines after startup.

What you also can try is to move the `venv` folder to `venv.old` and rerun the installer. This should install all python libraries again. That could fix the issue.

Else I can't think of anything. You can also rerun the installer multiple times to make sure all is installed.

You could also try to run in debug mode: https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/

I use the same OS and hardware for development, and it does work over here.

Mavajax, 2024-08-04:
Moving venv to venv.old and rerunning the installer solved the problem for me. Thank you !

theyosh, 2024-08-04:
You can than delete the venv.old folder when all is running again. Not need to save it any longer