theyosh/TerrariumPI #939: [BUG]: Blank Log Entries

cyber-shifter, 2024-08-18:
**Setup:**
 - TerrariumPI version: 4.11.1
 - Install: Manual
 - Raspberry PI: 4
 - OS: Bullseye

**Describe the bug**
When looking at logs on the web GUI, entries will appear with timestamps, but otherwise be completely blank. I should note that I recently updated, then reinstalled again because I was having issues with a sensor and thought that measure to be a prudent troubleshooting step.

**To Reproduce / Installation steps**
Login in first, then navigate to the "Logs" section.

**Expected behavior**
I expect for the logs to have content as they are entered into the log stream

**Screenshots**
So I followed the directions and started debugging. This gave me some output which I have attached here. I also included a screenshot of what the Logs section looks like in the GUI. 

[image]

However, when I interrupted the process as described in the instructions, more entries appeared. I haven't seen this sort of behavior before, so I wanted to note that.
[image]

Log GUI example
[image]

cyber-shifter, 2024-08-18:
I should also note, that it seems to be taking longer than normal to load the log entries in the web interface.

theyosh, 2024-08-18:
Found the issue. Do a git pull and restart. Now it should always work.

The case where it went wrong, was when you reload the GUI when you are already logged in. That should now be fixed.

cyber-shifter, 2024-08-18:
Thanks! Now I can get back to solving the rest of the issues I seemingly created by trying to create a new i2c bus.

theyosh, 2024-08-19:
So we can close this issue?

cyber-shifter, 2024-08-20:
Yes! I'll close it now. I've got another one coming your way though.

closed 2024-08-20 by cyber-shifter