theyosh/TerrariumPI #940: [BUG]:  After update all relays are reversed

worzel910, 2024-08-21:
**Setup:**
 - TerrariumPI version: 4.11.0
 - Install: Manual
 - Raspberry PI: 5
 - OS: Bullseye

**Describe the bug**
After updating to the latest version today upon restart all relays are working in reverse. 
Example. At the moment all relays should be off however they ar all activated. Including those not in use but state off in the UI.

I have rebooted the Pi several times too. 

**To Reproduce / Installation steps**
Install update

**Expected behavior**
When UI states the relays are off they should be off.

**Screenshots**
NA

**Logfiles**
Please add the log file found in the log folder: `log/terrariumpi.log`. All sensitive information should be stripped, and save to upload.

**Additional context**
Add any other context about the problem here.
[terrariumpi.log](https://github.com/user-attachments/files/16696510/terrariumpi.log)

theyosh, 2024-08-21:
Please search closed issues first: https://github.com/theyosh/TerrariumPI/issues/938

worzel910, 2024-08-21:
Odd, have no inverse option .

theyosh, 2024-08-21:
Clear browser cache, use a laptop/desktop. The calibration option is there when you edit the GPIO relay.

worzel910, 2024-08-21:
> Clear browser cache, use a laptop/desktop. The calibration option is there when you edit the GPIO relay.

That did it, for some reason all were inverted !

theyosh, 2024-08-21:
> > Clear browser cache, use a laptop/desktop. The calibration option is there when you edit the GPIO relay.
> 
> That did it, for some reason all were inverted !

https://github.com/theyosh/TerrariumPI/releases/tag/4.10.0

closed 2024-08-22 by theyosh