theyosh/TerrariumPI #942: 4.81 - 4.11.0 Upgrade Issue.

STGMavrick, 2024-08-23:
Steps taken to update: 

- Updated rpi via apt-get update upgrade
- Followed terrariumpi upgrade instructions via installation instructions of 4.11.0 post.

**After successful install and pi reboot Terrarium lights do not work, web portal does not work. SSH shows the following:**

[image]

**Trying to start the service gives the following error:**

_pi@raspberrypi:~/TerrariumPI $ sudo service terrariumpi start
Job for terrariumpi.service failed because the control process exited with error code.
See "systemctl status terrariumpi.service" and "journalctl -xe" for details._

**"systemctl status terrariumpi.service" and "journalctl -xe" command results:**

[image]

Any help would be great!

theyosh, 2024-08-23:
Did you use a git pull to update? Because this issue should be fixed with the latest code.

Also, what is the OS you are using?

Manual fix should be:

1. Login with ssh
2. `cd TerrariumPI`
3. `source venv/bin/activate`
4. `pip install yoyo-migrations===8.2.0`

And than it should start again.

STGMavrick, 2024-08-23:
Yessir, i did a git pull for the upgrade. 
I'm on buster (v10)

Ran the manual fix and everything is up and running! Thank you sir!

closed 2024-08-23 by theyosh