theyosh/TerrariumPI #943: [BUG]: Not able to recover from relay error

MakoShark2, 2024-08-24:
Setup:

TerrariumPI version: 4.10.2
Install: [Docker]
Raspberry PI: 4B
OS: Bookworm
Describe the bug
20x4 I2C LCD attached to the PI which receives notifications for sensor_update of 4 sensors (2 temperature + 2 humidity). Everything works fine except it seems the LCD screen is not cleared between messages. So for example if one message spans to the second line but the next one fits in one, the remaining second line of the previous message will still show.

To Reproduce / Installation steps
4 sensors connected (2 temperature + 2 humidity) connected to the PI 4B.
20x4 I2C LCD connected to the PI 4B.
20x4 LCD I2C display added to the notification service.
One notification message added,sensor_update, unlimited rate, message: ${name} ${value:.1f}${unit}

Expected behavior
LCD screen cleared before printing the notification message.
Or better, being this LCD 4 lines it would be nice if the message would scroll instead of printing every message always in the first line.

Screenshots
Don't have but I can get some if needed.

Logfiles
Don't have.

Additional context
None.

**Setup:**
 - TerrariumPI version: 4.11]0
 - Install: Docker
 - Raspberry PI: 4B
 - OS: Bookworm

**Describe the bug**
After getting a timeout for changing the state of a relay, it seems TP is not able to recover.

**To Reproduce / Installation steps**
Fan connected to a Shelly Smart Plug relay which is operated by the following area:

[image]

Main lights area:

[image]

From the logs after the timeout error, TP seems to be waiting for the settle time forever.

```
2024-08-24 15:39:19,776 - INFO    - terrariumRelay        - Updated relay shelly relay named 'Fan' at address 'http://192.168.1.16' with new value 100.00 in 0.11 seconds.
2024-08-24 15:39:19,912 - INFO    - terrariumArea         - Updated area Humidity named 'Fan' in 'sensors' mode at enclosure Gecko Loko in 0.00 seconds.
2024-08-24 15:39:50,107 - INFO    - terrariumRelay        - Updated relay shelly relay named 'Fan' at address 'http://192.168.1.16' with new value 100.00 in 0.27 seconds.
2024-08-24 15:40:03,261 - INFO    - terrariumArea         - Toggle the relays for area Humidity named 'Fan' part high to state off.
2024-08-24 15:40:03,268 - INFO    - terrariumArea         - Set the relay Shelly relay named 'Fan' at address 'http://192.168.1.16' to 0.0 with 0 seconds delay
2024-08-24 15:40:15,036 - ERROR   - hardware.relay        - Error changing state for relay Shelly relay named 'Fan' at address 'http://192.168.1.16' to 0.0 :Error changing relay Shelly relay named 'Fan' at address 'http://192.168.1.16' to state 0.0: Timed out after 10 seconds.
2024-08-24 15:40:15,210 - INFO    - terrariumArea         - Updated area Humidity named 'Fan' in 'sensors' mode at enclosure Gecko Loko in 0.00 seconds.
2024-08-24 15:40:19,889 - INFO    - terrariumRelay        - Updated relay shelly relay named 'Fan' at address 'http://192.168.1.16' with new value 0.00 in 0.21 seconds.
2024-08-24 15:40:20,025 - INFO    - terrariumArea         - Relays for Humidity named 'Fan' period high are not switched on because we have to wait for 30.0 more seconds of the total settle time of 30.0 seconds.
2024-08-24 15:40:20,025 - INFO    - terrariumArea         - Updated area Humidity named 'Fan' in 'sensors' mode at enclosure Gecko Loko in 0.00 seconds.
2024-08-24 15:40:49,996 - INFO    - terrariumRelay        - Updated relay shelly relay named 'Fan' at address 'http://192.168.1.16' with new value 0.00 in 0.21 seconds.
2024-08-24 15:41:03,674 - INFO    - terrariumArea         - Relays for Humidity named 'Fan' period high are not switched on because we have to wait for 30.0 more seconds of the total settle time of 30.0 seconds.
2024-08-24 15:41:03,675 - INFO    - terrariumArea         - Updated area Humidity named 'Fan' in 'sensors' mode at enclosure Gecko Loko in 0.01 seconds.
2024-08-24 15:41:24,803 - INFO    - terrariumRelay        - Updated relay shelly relay named 'Fan' at address 'http://192.168.1.16' with new value 0.00 in 0.10 seconds.
2024-08-24 15:41:24,939 - INFO    - terrariumArea         - Relays for Humidity named 'Fan' period high are not switched on because we have to wait for 30.0 more seconds of the total settle time of 30.0 seconds.
2024-08-24 15:41:24,940 - INFO    - terrariumArea         - Updated area Humidity named 'Fan' in 'sensors' mode at enclosure Gecko Loko in 0.00 seconds.
2024-08-24 15:41:50,048 - INFO    - terrariumRelay        - Updated relay shelly relay named 'Fan' at address 'http://192.168.1.16' with new value 0.00 in 0.21 seconds.
```

**Expected behavior**
Recover from the error on the next sucessful round.

**Screenshots**
None.

**Logfiles**
[terrariumpi_logfile.txt](https://github.com/user-attachments/files/16737836/terrariumpi_logfile.txt)

**Additional context**
None.

theyosh, 2024-08-24:
What has the LCD to do with your relay issue? What is the problem? It is not clear from your issue

MakoShark2, 2024-08-24:
Sorry, that was a copy paste, I've erased the LCD part.

theyosh, 2024-09-05:
Sorry, totally forgot this. But TerrariumPI should always continue. 

But your example looks like a hickup, which are hard to tackle. 

And on 2024-08-24 15:40:19,889 the Fan reports to be off. Which is only 4 seconds later.

And I guess it has to do with a slow bluetooth connection:
`2024-08-24 15:40:14,976 - INFO    - terrariumSensor       - Updated sensor LYWSD03MMC bluetooth sensor humidity named 'OUT Humid' at address 'a4:c1:38:1b:e1:b6' with new value 69.00% in 26.93 seconds.`
When this happens, mostly the Pi was a bit more busy than normal, and than sometimes it can cause issues.

But I think this is still a hickup, which should not happen that often.

theyosh, 2024-09-12:
I am not sure what else I can do about this. Does this still happen?

MakoShark2, 2024-09-12:
Didn't saw it happen again, but also no timeouts I think.

I understand this might be an unfrequent hickup and hard to trace.

closed 2024-09-14 by theyosh