theyosh/TerrariumPI #946: [BUG]: Lack of recovery from meross cloud loss

rossisbudda, 2024-09-07:
**Setup:**
 - TerrariumPI version: x4.10.2
 - Install: Manual
 - Raspberry PI: 3B+
 - OS: Raspbian

Last night at around 04:50 one of my heating relays remained off when it should have turned back on. We lost connection to meross cloud then some other errors occurred. Resulting in a cold period. I can see in the logs that once we regained connection to the cloud the relay did not ever come back to life as such.

I think I will switch to a wired device as I can't rely on any cloud service, but thought you wold like to know about this as it seems that we could recover better.

I might have preferred a more robust recovery and regular re-checking of relay states to prevent long periods of stuck inactivity.

[image]

`024-09-07 04:49:09,546 - WARNING - terrariumCloud        - Reconnecting to Meross cloud. Somehow the connection was lost ...
2024-09-07 04:49:09,558 - INFO    - terrariumCloud        - Stopping Meross cloud ... 
2024-09-07 04:49:10,173 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Setting temperature to 20.5 and humidity to 72.3
2024-09-07 04:49:10,174 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Setting temperature to 20.4 and humidity to 70.9
2024-09-07 04:49:10,175 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Relay state 2303179204369154060748e1e9c0360d [True, False, True, True, False, True]
2024-09-07 04:49:10,177 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Setting temperature to 20.5 and humidity to 72.3
2024-09-07 04:49:10,177 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Setting temperature to 20.4 and humidity to 70.9
2024-09-07 04:49:10,178 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Setting temperature to 20.5 and humidity to 72.3
2024-09-07 04:49:10,179 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Setting temperature to 20.4 and humidity to 70.9
2024-09-07 04:49:10,555 - WARNING - hardware.relay        - Error changing relay Meross power switches relay named '2. Heater' at address '2303179204369154060748e1e9c0360d,2' to state 0.0. Error: , retrying in 0.5 seconds...
2024-09-07 04:49:11,062 - WARNING - hardware.relay        - Error changing relay Meross power switches relay named '2. Heater' at address '2303179204369154060748e1e9c0360d,2' to state 0.0. Error: Error changing relay Meross power switches relay named '2. Heater' at address '2303179204369154060748e1e9c0360d,2' to state 0.0. Error: unknown, retrying in 0.5 seconds...
2024-09-07 04:49:11,568 - ERROR   - hardware.relay        - Error changing state for relay Meross power switches relay named '2. Heater' at address '2303179204369154060748e1e9c0360d,2' to 0.0 :Error changing relay Meross power switches relay named '2. Heater' at address '2303179204369154060748e1e9c0360d,2' to state 0.0. Error: Error changing relay Meross power switches relay named '2. Heater' at address '2303179204369154060748e1e9c0360d,2' to state 0.0. Error: unknown
2024-09-07 04:49:15,748 - INFO    - terrariumEngine       - Starting a new update round with 6 sensors, 5 relays, 0 buttons and 0 webcams.
2024-09-07 04:49:15,788 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT4X sensor humidity named 'Loft SHT40' at address '0x44' with new value 65.48% in 0.02 seconds.
2024-09-07 04:49:15,817 - INFO    - terrariumRelay        - Updated relay meross relay named '1. Light' at address '2303179204369154060748e1e9c0360d,1' with new value 0.00 in 0.01 seconds.
2024-09-07 04:49:15,902 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT4X sensor temperature named 'Loft SHT40' at address '0x44' with new value 24.98C in 0.01 seconds.
2024-09-07 04:49:15,930 - INFO    - terrariumRelay        - Updated relay meross relay named '2. Heater' at address '2303179204369154060748e1e9c0360d,2' with new value 100.00 in 0.01 seconds.
2024-09-07 04:49:16,017 - INFO    - terrariumSensor       - Updated sensor Meross MS100(F) humidity named 'Meross MS100(F) 1- ms100f measuring humidity' at address '39000DECBD1E' with new value 72.30% in 0.01 seconds.
2024-09-07 04:49:16,043 - INFO    - terrariumRelay        - Updated relay meross relay named '3. Humidifier' at address '2303179204369154060748e1e9c0360d,3' with new value 100.00 in 0.01 seconds.
2024-09-07 04:49:16,131 - INFO    - terrariumSensor       - Updated sensor Meross MS100(F) temperature named 'Meross MS100(F) 1- ms100f measuring temperature' at address '39000DECBD1E' with new value 20.50C in 0.01 seconds.
2024-09-07 04:49:16,155 - INFO    - terrariumRelay        - Updated relay meross relay named '4. Cooler' at address '2303179204369154060748e1e9c0360d,4' with new value 0.00 in 0.01 seconds.
2024-09-07 04:49:16,245 - INFO    - terrariumSensor       - Updated sensor Meross MS100(F) temperature named 'Meross MS100(F) 2- ms100f measuring temperature' at address '39000DECDA10' with new value 20.40C in 0.01 seconds.
2024-09-07 04:49:16,267 - INFO    - terrariumRelay        - Updated relay meross relay named 'Channel USB' at address '2303179204369154060748e1e9c0360d,5' with new value 100.00 in 0.01 seconds.
2024-09-07 04:49:16,359 - INFO    - terrariumSensor       - Updated sensor Meross MS100(F) humidity named 'Meross MS100(F) 2- ms100f measuring humidity' at address '39000DECDA10' with new value 70.90% in 0.01 seconds.
2024-09-07 04:49:16,475 - INFO    - terrariumArea         - Updated area Lighting named 'Lighting' in 'timer' mode at enclosure stuff in 0.00 seconds.
2024-09-07 04:49:16,481 - INFO    - terrariumArea         - Updated area Heating named 'Heating' in 'sensors' mode at enclosure stuff in 0.01 seconds.
2024-09-07 04:49:16,484 - INFO    - terrariumEnclosure    - Updated Enclosure stuff with 2 areas in 0.01 seconds.
2024-09-07 04:49:16,604 - INFO    - terrariumEngine       - Message of the Day generated in 0.12 seconds
2024-09-07 04:49:16,709 - INFO    - terrariumEngine       - Engine update done in 0.96 seconds. Waiting for 7.04 seconds for the next round.
2024-09-07 04:49:23,751 - INFO    - terrariumEngine       - Starting a new update round with 6 sensors, 5 relays, 0 buttons and 0 webcams.
2024-09-07 04:49:23,825 - INFO    - terrariumRelay        - Updated relay meross relay named '1. Light' at address '2303179204369154060748e1e9c0360d,1' with new value 0.00 in 0.01 seconds.
2024-09-07 04:49:23,849 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT4X sensor humidity named 'Loft SHT40' at address '0x44' with new value 65.50% in 0.07 seconds.
2024-09-07 04:49:23,937 - INFO    - terrariumRelay        - Updated relay meross relay named '2. Heater' at address '2303179204369154060748e1e9c0360d,2' with new value 100.00 in 0.01 seconds.
2024-09-07 04:49:23,963 - INFO    - terrariumSensor       - Updated sensor Sensirion SHT4X sensor temperature named 'Loft SHT40' at address '0x44' with new value 24.96C in 0.01 seconds.
2024-09-07 04:49:24,050 - INFO    - terrariumRelay        - Updated relay meross relay named '3. Humidifier' at address '2303179204369154060748e1e9c0360d,3' with new value 100.00 in 0.01 seconds.
2024-09-07 04:49:24,077 - INFO    - terrariumSensor       - Updated sensor Meross MS100(F) humidity named 'Meross MS100(F) 1- ms100f measuring humidity' at address '39000DECBD1E' with new value 72.30% in 0.01 seconds.
2024-09-07 04:49:24,163 - INFO    - terrariumRelay        - Updated relay meross relay named '4. Cooler' at address '2303179204369154060748e1e9c0360d,4' with new value 0.00 in 0.01 seconds.
2024-09-07 04:49:24,193 - INFO    - terrariumSensor       - Updated sensor Meross MS100(F) temperature named 'Meross MS100(F) 1- ms100f measuring temperature' at address '39000DECBD1E' with new value 20.50C in 0.01 seconds.
2024-09-07 04:49:24,275 - INFO    - terrariumRelay        - Updated relay meross relay named 'Channel USB' at address '2303179204369154060748e1e9c0360d,5' with new value 100.00 in 0.01 seconds.
2024-09-07 04:49:24,308 - INFO    - terrariumSensor       - Updated sensor Meross MS100(F) temperature named 'Meross MS100(F) 2- ms100f measuring temperature' at address '39000DECDA10' with new value 20.40C in 0.01 seconds.
2024-09-07 04:49:24,431 - INFO    - terrariumSensor       - Updated sensor Meross MS100(F) humidity named 'Meross MS100(F) 2- ms100f measuring humidity' at address '39000DECDA10' with new value 70.90% in 0.01 seconds.
[... cut ...]
2024-09-07 04:50:06,975 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Setting temperature to 20.5 and humidity to 72.2
2024-09-07 04:50:06,976 - INFO    - terrariumCloud        - Got an update from the Meross Cloud. Setting temperature to 20.4 and humidity to 70.8
2024-09-07 04:50:11,769 - INFO    - terrariumEngine       - Starting a new update round with 6 sensors, 5 relays, 0 buttons and 0 webcams.
`
 Thanks for looking. As I say I will switch to a wired relay and forget wifi products as I suppose they have an unreliability in their connectivity, or is there a more reliable wifi relay?

theyosh, 2024-09-08:
Hmm, this is meross cloud related. Not wifi. There was some disconnect, which could be Wifi. But the meross code is rather complex, and does not work easy with my software.

Also, how long are you using this meross hardware, and how often did it happen before. I do have the idea that the Meross system is pretty stable. I do not get much issues with it.

Because what I see, is that the temperature stays at 24C in the logs, while the graph goes down. Which I cannot explain.
So somehow, the state of sensors and relays looks frozen in the system/code. And that I cannot explain. This is so hard to debug due to the fact Meross does not officially allow us to do this. But as long as we play nice with their api, they allow us to use their api with our custom code.

My question is, how did the temperature got restored. You just restarted TerrariumPI?

I think I cannot do much here. This is so hard to debug. And when it only happens once a year, it can live with it.

rossisbudda, 2024-09-08:
Thanks so much. I will look out for it in the future and report if it is
too frequent.

I haven't been using this hardware for very long yet and I'm not certain
this has occurred before.

To "unstick" the relay I clicked the toggle button on the relays page, this
seemed to revive it. As this relay is frequently adjusting it is easy to
get stuck in a high or low state and I want to really avoid fluctuations in
the terrarium.  I've decided to try adding sudo service terrariumpi restart
to a crontab that runs every so often in the hope this would "refresh" the
engine.

Appreciate the advice 👍😎🙏

theyosh, 2024-09-12:
So we can close this issue?

rossisbudda, 2024-09-12:
Yes, please!

closed 2024-09-14 by theyosh