theyosh/TerrariumPI #948: [BUG]: day/night confusiojn

rossisbudda, 2024-09-07:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.[x].[y]
 - Install: [Manual] / [Docker]
 - Raspberry PI: [e.g. 3B or 4]
 - OS: [Buster, Bullseye, Bookworm]

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce / Installation steps**
Please provide the steps to reproduce or how you have installed the software.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Please add screenshots of the [debug screens](https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/):

**Logfiles**
Please add the log file found in the log folder: `log/terrariumpi.log`. All sensitive information should be stripped, and save to upload.

**Additional context**
Add any other context about the problem here.

theyosh, 2024-09-08:
Sorry, but the issue has no information. Could you please put your question in the issue?

rossisbudda, 2024-09-08:
Apologies, I misfired then forgot to submit the edit.

rossisbudda, 2024-09-08:
I include the day/night period settings screen also.
[image]

rossisbudda, 2024-09-08:
sorry about the messy ticket, all the original post has been cleaned up now 🧽

theyosh, 2024-09-09:
On the other hand, this is a nice test case. I think I fixed this issue in code. Do a `git pull` and restart TerrariumPI.

Keep the restarting crontab in place, and lets see what happens tonight. It should stay on after a reboot just after 00:00.

theyosh, 2024-09-12:
Did you had time to test the latest code with your timer setting? I think the issue should be fixed. A specially with you 30 min crontab restart.

rossisbudda, 2024-09-12:
Hi sorry for the delay.

I tried a gitpull but it said I had changed a bunch of files and I couldn’t
figure it out. I’m a bit of a noob and I didn’t want to upset the cycle of
my environment or lose my config as it’s all live and I don’t want to
affect the lighting cycle.

Completely accept my hack is awful and tbh my extra hack was to disable the
crontab during the “post midnight” daytime 12-3.30. Definitely sucks but it
worked this far.

If I can swap to manual analogue timer soon I would find time do
confidently make a fresh install and reconfigure identically, then test.

Any advice welcome. Thanks for your patience.

theyosh, 2024-09-12:
If you reinstall it, make some backups: https://theyosh.github.io/TerrariumPI/faq/which-files-to-backup/
And restore it like this: https://theyosh.github.io/TerrariumPI/faq/restoring-backup-files/

That should save some time setting it all up.

rossisbudda, 2024-09-12:
Ok thanks I did see this but I was a bit hesitant.

I’ll try to update better or install fresh tomorrow. Please save the ticket!

Thx

theyosh, 2024-09-12:
And you do not have to reset your SD card. Just move the existing `TerrariumPI` folder to a temporary name, and just start the TerrariumPI installation with the `git pull`.

You do not have to complete reinstall the OS. Just re-install TerrariumPI only. Saves a lot of time.

Will keep this issue open. No problem

rossisbudda, 2024-09-13:
I have done a manual backup, gitpull, reinstall (4.11.1)  and removed my bi hourly crontab. Will report back if the midnight glitch happens or not!

theyosh, 2024-09-13:
Ok, but without the crontab, the bug will not happen. But that is OK. As long it is stable for you. That is the most important.

The bug will only happen when you restart TerrariumPI after 00:00 and before 03:00 hours.

rossisbudda, 2024-09-13:
I will be watching, then try a manual restart after midnight to see if it can reproduce. You're the best!

rossisbudda, 2024-09-15:
Hi, I tried it. Restarting the service between 00:00 and 03:30 resumed the
day cycle as normal and the lights stayed on :)
Thanks again! I think you can close this.

theyosh, 2024-09-15:
Thanks for testing. And good to hear the bug is fixed!

closed 2024-09-15 by theyosh