theyosh/TerrariumPI #952: [BUG]: Relay state inverted

martynwheeler, 2024-09-21:
Hi,

Have just updated to 4.11.

My relay is controlled via GPIO 11.  The logic seems to be inverted after update (light is off when interface says on and on when interface says off).

 Raspberry Pi 3 Model B Rev 1.2/Raspbian GNU/Linux 10 (buster)"

Any ideas to fix?

Thank you

Martyn

**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.[x].[y]
 - Install: [Manual] / [Docker]
 - Raspberry PI: [e.g. 3B or 4]
 - OS: [Buster, Bullseye, Bookworm]

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce / Installation steps**
Please provide the steps to reproduce or how you have installed the software.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Please add screenshots of the [debug screens](https://theyosh.github.io/TerrariumPI/faq/how-to-debug-terrariumpi/):

**Logfiles**
Please add the log file found in the log folder: `log/terrariumpi.log`. All sensitive information should be stripped, and save to upload.

**Additional context**
Add any other context about the problem here.

theyosh, 2024-09-21:
Please look at old closed issues:

https://github.com/theyosh/TerrariumPI/issues/940
https://github.com/theyosh/TerrariumPI/issues/938

More info at:
https://github.com/theyosh/TerrariumPI/releases/tag/4.10.0

Clear browser cache, and you have a calibration option. Toggle inverse mode there

martynwheeler, 2024-09-21:
Thanks, sorry I did not see that.

closed 2024-09-21 by martynwheeler