theyosh/TerrariumPI #957: [BUG]:

MakoShark2, 2024-10-07:
**More info**
Please look first at the [FAQ](https://theyosh.github.io/TerrariumPI/faq/) or [Current Issues](https://github.com/theyosh/TerrariumPI/issues) for more information.
Or search in the [closed tickets](https://github.com/theyosh/TerrariumPI/issues?q=is%3Aissue+is%3Aclosed) to see if it should been fixed in the past

**Setup:**
 - TerrariumPI version: 4.11.1
 - Install: [Docker]
 - Raspberry PI: 4B
 - OS: Bookworm

**Describe the bug**
Trying to add to a Tapo C110 camera which supports live rtsp stream natively.

https://www.tp-link.com/en/home-networking/cloud-camera/tapo-c110/

Although TPI adds the camera successfully it's not able to connect and shows a black screen. The log reports continuously the following messages:

```
2024-10-07 21:31:26,074 - ERROR   - hardware.webcam       - Webcam Live RTSP camera named 'Tapo C110' at address 'rtsp://*********@192.168.0.218/stream1' could not be updated after 16.629 seconds! Please check your webcam connection.
2024-10-07 21:31:20,524 - WARNING - hardware.webcam       - Webcam Live RTSP camera named 'Tapo C110' at address 'rtsp://*********@192.168.0.218/stream1' is crashed. Restarting the webcam.
2024-10-07 21:31:14,981 - WARNING - hardware.webcam       - Webcam Live RTSP camera named 'Tapo C110' at address 'rtsp://*********@192.168.0.218/stream1' is crashed. Restarting the webcam
```.

**To Reproduce / Installation steps**
Camera setup like following:
![image](https://github.com/user-attachments/assets/b5c65d02-6953-4d5a-9c4e-1fc37e748e4a)

The address is correct and confirmed working throught VLC.

**Expected behavior**
Connect to the camera and show the live stream.

**Screenshots**
![image](https://github.com/user-attachments/assets/fab51b0a-55b6-437c-98d1-21d47c3f16dc)

**Logfiles**

**Additional context**

theyosh, 2024-10-08:
Could you watch the stream in VLC? https://www.tp-link.com/en/support/faq/2680/ (on 2/3 of the page)

MakoShark2, 2024-10-08:
> Could you watch the stream in VLC? https://www.tp-link.com/en/support/faq/2680/ (on 2/3 of the page)

Yes, I can watch the high quality stream (stream1) in VLC.
[image]

MakoShark2, 2024-10-08:
Just a note. To access the stream the camera requires credentials which should be passed this way:

rtsp://user:password@ip:554/stream1

VLC does not appreciate this and requires only the url rtsp://ip/stream1 and will prompt afterwards for the user and password.

theyosh, 2024-10-09:
Yeah, I found some issues with authentication in the url. But the hard part is that you run it in Docker. That is hard to test.

So let me think about this a bit more.... I think I have a way that you can test this easily.

MakoShark2, 2024-10-09:
OK.... But if it's just a couple of updated scripts I can copy them into my running container.

theyosh, 2024-10-09:
Ok than the following is maybe an option. I do not know all docker commands, so that is up to you to find.

In the container there is a helper file that needs to be changed: `/TerrariumPI/hardware/webcam/rtsplive_webcam.sh`

So that file should copied from the container to your home folder.
Than edit the file and change the following code part:

`-re -i "${DEVICE}" \` to `-re -i \""${DEVICE}"\" \` the change is two times `\"`  around `"${DEVICE}"`

We need to have some extra quotes due to the `user:pass@` part in the url.

When done, copy the file back. But I am not sure if the change will hold a restart. 
Keep a copy of that file in your home folder. That can be used for further testing.

I hope this helps...

Also, you could edit inside the container. You need to update apt first before you can install an editor.

MakoShark2, 2024-10-10:
OK... I've updated the script inside the container, but still it wasn't working.

So I tried to run the script from the command prompt which gave me: **[h264_omx @ 0x1eb20d0] OMX error 80001000**

After a quick search that seems to point to a insufficient resources encoding error:
https://forums.raspberrypi.com/viewtopic.php?t=188317

So I modified the address to point to the low quality camera stream (stream2) and it's working fine now.

I'll try to reduce the resolution of the high quality camera stream (which is at 2K now) and see if it works.

But anyway, not a software problem.

Issue can be closed, thanks.

MakoShark2, 2024-10-12:
Oh by the way... The \" around ${DEVICE} seems to be not working... I had to remove it back to what it was to get the camera working on the low quality stream.

theyosh, 2024-10-13:
Ok, that is nice to know. Thanks for testing as well.

closed 2024-10-13 by theyosh