theyosh/TerrariumPI #963: [BUG]: Webcam archive fills up disk completely

1ucket, 2024-10-27:
**Setup:**
 - TerrariumPI version: 4.7.0
 - Install: Manual
 - Raspberry PI: 4
 - OS: Buster

**Describe the bug**
The webcam makes regular screenshots. After 11 months (motion sensitive) it filled up the whole 60Gb disk. Because of this TerrariumPi was not working properly anymore (e.g. lamps not turning off).

**To Reproduce / Installation steps**
Install a webcam, let TerrariumPi run for a long enough time to fill up the disk.

**Expected behavior**
Some precautions would be taken to prevent filling up the disk
e.g.:
- setting a maximum number of saved webcam images
- setting a maximum age of kept photos
- or in my opinion best: maximum allowed archive space)
Older images get then deleted.

Of course one could manually setup a cronjob or similar script to delete the images periodically as described [here](https://github.com/theyosh/TerrariumPI/issues/329#issuecomment-537209871), but this isn't friendly to less technical users, and it took me a while to get from "why is the light not turning off?" to "oh, my disk is full!"

**Logfiles**
[terrariumpi.log](https://github.com/user-attachments/files/17535241/terrariumpi.log)

theyosh, 2024-10-31:
Ok, I updated to code to add an option to store X days of archive images. That is the best I can do within my software.
So, that means that you should enter a value lower than 6 * 30 = 180 days (half a year).

But the biggest issue is that you say you are on version 4.7.0. That means you need to upgrade to the latest version. That is a big jump. Not sure if that will work....

Also, this will cause probably issues with GPIO relays if you have. Those can be inverted (again). But in the GUI you can fix that now at the edit screen of the relay.

But that is what I can do. Personally I have connected a cloud drive to my Pi, and stored all my pictures there.

1ucket, 2024-10-31:
Thank you very much! That should definitely be enough.
I've since updated to the latest version. Had issues with my relays and some other things, so I did a clean reinstall (my terrarium is not too complex, only 3 relays and the camera, so it was setup again quickly).
Unfortunately I used Docker in the last install so I can't just pull the changes. But I'll see with the next version change. :)

closed 2024-10-31 by 1ucket